FreeCAD/FreeCAD #8341: [Issue] Partial part transparency makes the window transparent

nils-van-zuijlen, 2023-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * fasteners 0.4.52
  * freecad.gears 1.0.0
```
```

### Subproject(s) affected?

PartDesign

### Issue description

When making a part partially transparent (between 1 and 99% transparency), the part becomes transparent beyond the FreeCAD window and lets the windows under them be seen.

![image](https://user-images.githubusercontent.com/22444937/216735143-60b96732-1e05-443c-b07a-b0591998a85f.png)

### Anything else?

This may be related to https://github.com/FreeCAD/FreeCAD/issues/6177 

After restarting FreeCAD, without having made any updates, I cannot reproduce :sad

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-02-04:
Can you clarify your final comment there? Is this something that you observed once, but can't get to happen again? It seems like a graphics driver or window system bug to me: if even you can't get it to happen again I am not sure how anyone could track it down.

luzpaz, 2023-02-04:
This looks like the old 'Known Issue'  for old graphics cards.

nils-van-zuijlen, 2023-02-06:
> Can you clarify your final comment there? Is this something that you observed once, but can't get to happen again? It seems like a graphics driver or window system bug to me: if even you can't get it to happen again I am not sure how anyone could track it down.

Yes, it mean I observed it once only. I am creating an issue so that if anyone bumps into the same issue, they can continue searching that bug.

concretedog, 2023-04-15:
I have the same issue here on Manjaro Xfce with 0.20.2, photo's below with about info. 
[image]

If I make a datum plane or set an object to have some transparency then my manjaro desktop bleeds through in the preview pane!
[image]
[image]
[image]

luzpaz, 2023-04-15:
Found a related (dupe?) #6177

luzpaz, 2023-06-01:
@PLChris is this the known graphic card issue ?

Chacha75, 2023-08-19:
I have the same issue with both FreeCAD v 0.20 and 0.21 in Manjaro. However the issue disappears when I turn off Compositing in XFCE.

CPU: 6-core AMD Ryzen 5 4500U with Radeon Graphics (-MCP-)
speed/min/max: 1560/1400/2375 MHz Kernel: 6.1.44-1-MANJARO x86_64 Up: 8h 1m
Mem: 2.71/7.13 GiB (38.0%) Storage: 476.94 GiB (65.1% used) Procs: 283
Shell: Bash inxi: 3.3.29

OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: makepkg
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.11.3, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.59
  * sheetmetal 0.2.63

luzpaz, 2023-08-21:
> However the issue disappears when I turn off Compositing in XFCE.

@nils-van-zuijlen can you reproduce workaround ?

Chacha75, 2023-08-22:
@nils-van-zuijlen You can find it in **Window Manager Tweaks** >> **Compositor** tab >> **Enable display compositing**

concretedog, 2023-09-10:
Just to confirm I've just replicated this workaround on the same setup as I listed having the problem with on April 15th, sorry for the delay haven't used this Manjaro machine for a while.

zoopp, 2023-09-11:
Is this similar to what I reported here: https://github.com/swaywm/sway/issues/7739 (see attached videos)? SwayWM authors said it's likely an application specific bug.

zoopp, 2024-01-24:
```
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.5.0
  * fasteners 0.4.66
  * Curves 0.6.13
```

Seems to have gotten worse. Right click menu and toolbar menus have become completely transparent for me.

https://github.com/FreeCAD/FreeCAD/assets/1645023/89cb7c87-c5ae-46ef-ad62-b7367e339519

Chacha75, 2024-01-25:
> Seems to have gotten worse. Right click menu and toolbar menus have become completely transparent for me.

I'm not sure, if it's the same issue. Only the menus are transparent, and you can see the Freecad window beneath it. The FreeCad window itself is not transparent.
Maybe it's a theme issue. Try to switch back to Classic theme. It solves this problem for me.

zoopp, 2024-01-29:
Ah, you're right. I played a bit with the theming over the weekend and after re-applying the theme the menus become solid again for a while after which a Qt error (or warning, can't fully recall) is logged followed by them becoming transparent again.

Seems to be indeed a different issue. I'll take a better look at it some other time.

Chacha75, 2024-02-23:
I've found a much better workaround:

In **Kvantum Manager** under _'Configure Active Theme'_ on the _'Compositing & General Look'_ tab add freecad to the list of **Opaque apps**

pinkavaj, 2024-10-10:
I'm able to reproduce the issue pretty consistently with the following steps:

* Start some app (eg browser, or terminal with some text in)
* Start FreeCad
* Draw some (big) object and make it ag 90% transparent
* resize the FreeCad window to occupy only part of the screen
* click on the "maximize" or "restore" button in the window decoration to resize the window to full screen or back - bingo!

OS: Arch Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38942 (Git)
Python 3.12.6, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Platform: Wayland

Is anyone with similar/different configuration able to reproduce this?

benj5378, 2025-02-06:
@karliss and @captain0xff reports this as well on [Discord](https://discord.com/channels/870877411049357352/1336794683170488433), and can resolve it using `QT_QPA_PLATFORM=xcb` when running. This indicates an upstream bug.

karliss, 2025-02-06:
As a quick and simple workaround, adding:
```
    glPushAttrib(GL_COLOR_BUFFER_BIT);
    glColorMask(false, false, false, true);
    glClearColor(0, 0, 0, 1);
    glClear(GL_COLOR_BUFFER_BIT);
    glPopAttrib();
```
that is resetting alpha value for whole 3d view to 1.0 at the end of `renderScene` prevents the visual defect, like I guessed in the discord discussion.
It shouldn't be necessary, and more proper fix would probably be properly configuring options so that alpha values from 3d rendering isn't used when combining it with rest of the main window. Although I don't have high hopes that someone in near future will figure out whether that's fault of Qt OpenGL widget, Qt wayland backend, some other part of Qt, Quarter (glue layer between Qt and Coin3d) or FreeCad configuring and combining it all together.

luzpaz, 2025-07-28:
re-opened this because it seems it was auto-closed by the bot due to staleness but no confirmation if this was fixed legitmately.  
Is this still an issue on 1.1-dev ?

closed 2025-08-02 by luzpaz