FreeCAD/FreeCAD #8344: [Bug] App::PropertyColor erroneous behavior

FreeCAD-Tools, 2023-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=75287

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10
Word size of FreeCAD: 64-bit
Version: 0.21.0.31432 (Git)
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3

and

OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31599 (Git)
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.6.3
```

### Subproject(s) affected?

None

### Issue description

When setting the color, it is set not accurately. Let's say the user sets the color through the dialog: (r=50, g=100, b= 200). But FreeCAD will set the color as: (r=51, g=99, b=198).

Bug is confirmed in Windows and Linux operated systems.

More information on forum topic.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-02-04:
The Decimals preference ([Units](https://wiki.freecad.org/Preferences_Editor#Units)) is wrongly used to round the color values.

donovaly, 2023-02-05:
> The Decimals preference ([Units](https://wiki.freecad.org/Preferences_Editor#Units)) is wrongly used to round the color values.

@0penBrain , could you please have a look if this is a bug?

commit ba25e9a8 (2023-02-12): [Core] Add possibility to set color with tuple of integers, fixes #8344
commit ddc66c43 (2023-02-16): [Core] Add possibility to set color with tuple of integers, fixes #8344
closed 2023-02-12 by 0penBrain