FreeCAD/FreeCAD #8345: [Feature Request] Extend PD's python interface

Jee-Bee, 2023-02-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=75627

### Subproject(s) affected?

PartDesign

### Idea description

Currently the PartDesign Workbench is scriptable but not on a nice pythonic way.
There are only a few features directly accessible when import PD.

an example of what i have in mind(rough idea):
`import PartDesign as pd

my_body = pd.add_body()
new_sketch = my_body.add_sketch(Some_plane, ...)
new_pad = my_body.add_pad(new_sketch, ..., ...)`

Since it is some time ago that i checked the PD C++ code i'm not sure if my rough idea is the best way to go.
In my opinion every feature that is accessible from python is an improvement 

### Anything else?

Print screen of current in python available PD features

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-15:
@Jee-Bee can you make a list of requested python functions (that we can track in this ticket) ?

Jee-Bee, 2023-03-26:
I tried to create a list of features i want to acces. I don't care about the naming. I think that is not my role even if i created the FR.

- Create body
- Create Sketch
- Map sketch

- Create Datum Plane
- Create Datum Axis (line)
- Create Datum CS
- Create Datum point
  

- Create Shape Binder
- Create Sub object
- Create Clone

- Pad
- Revolution
- additive Loft
- additive Pipe
- additive Helix
 
- Pocket
- Hole
- Groove
- subtractive loft
- subtractive pipe
- subtractive helix
 
- Mirror
- Linear Pattern
- Polar Pattern
- Multi Transform

- Fillet
- Chamfer
- Draft
- Thickness

I'm not sure about the primitives. I personally don't care. But i can understand that others thing different about that

FlachyJoe, 2023-03-26:
NB : all listed objects can be make throw DocumentObject.addObject and BodyObject.newObject so only helpers are claimed.
```
>>> App.ActiveDocument.addObject('PartDesign::Body', 'Body')
<body object>
>> pad = App.ActiveDocument.Body.newObject('PartDesign::Pad','Pad')
>> mirror = App.ActiveDocument.Body.newObject('PartDesign::Mirrored','Mirrored')
>> mirror.Originals = [pad]
>> #etc.
```

luzpaz, 2023-03-26:
Maybe this is a usecase for @mnesarco's https://github.com/mnesarco/fcscript ?

mnesarco, 2023-03-26:
> Maybe this is a usecase for @mnesarco's https://github.com/mnesarco/fcscript ?

Yes, in the demo macro available after install, there are many examples creating sketches and padding in a PD Body.

Jee-Bee, 2023-03-27:
The request is not about what is already possible. Ofcourse i can record a macro and i am able to automate things...
The request is about having a dedicated (preferable )pythonic interface.
The mentioned commands work probably well when recording a macro but give no help when using with PD as imported Module.
see as example: https://forum.freecad.org/viewtopic.php?t=76312

mnesarco, 2023-03-27:
fcscript is a project  of a simplified and pythonic api for FreeCAD

Jee-Bee, 2023-03-27:
Thanks! Sounds like a good idea, assuming it ends in FC main code ;)

luzpaz, 2023-03-28:
@Jee-Bee it's currently available via the Addon Manager. It's super promising!

adrianinsaval, 2023-04-29:
I don't consider Part Design to be primarily intended for this kind of use, I consider this request out of scope for the Part Design project as maintaining an additional interface is quite the maintenance burden as it would need to be constantly updated and extended as Part Design evolves.

Not everything needs to be on the main repo, I would say mnesarco's addon satisfies this need. Also consider using the Part API instead when working through python.

closed 2023-04-29 by adrianinsaval