FreeCAD/FreeCAD #8346: [Issue]

mowley-bob, 2023-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Last working version...

[code]
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31556 (Git) AppImage
Build type: Release
Branch: master
Hash: 4091db47397f6e537a284f50c4f572545f3f30d0
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Help 1.0.3
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

Some characters <Shift>+B and lowercase t (possibly others) produce no output when renaming objects in the Attributes tree on the left of the application.  

Interestingly Caps-Lock+b does result in a 'B' however I have been unable to find a work around for getting a 't'.  

Possibly a keyboard shortcut related issue?

This occurs in all revisions since 31556.

This happens when creating new CAD Documents.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mowley-bob, 2023-02-08:
'w' 'e' 'a' & 't'  also don't work, to replicate create a new document and a new body.  Rename the  Body to test.

mowley-bob, 2023-02-08:
After further investigation this appears to only be in the weekly appimage.  I've just compiled from source and this version (at least when compiled from source) exhibits no problems.  

[code]
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31810 (Git)
Build type: None
Branch: master
Hash: fe1c09f841a1c858161bfee8b0959b6c879ebe17
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Help 1.0.3
  * fasteners 0.4.54
  * Assembly4 0.12.5
  * BIM 2021.12.0
[/code]

rklasen, 2023-02-10:
I can confirm the issue in the weekly appimage.

I think there is another issue too: when the focus is in an entry field (like dimensions of a cube or something), the numbers 1-6 cannot be entered (neither home row nor numpad. Instead, the view changes.

It seems the global shortcuts for view changes supersede the current input field, and I suspect the same issue occurs here. I've opened a related issue #8417

Strahlensauger, 2023-02-18:
This seems to be the same issue 
[#8305](https://github.com/FreeCAD/FreeCAD/issues/8305)

chennes, 2023-02-22:
@looooo do you have any idea why the AppImage would be doing this, when self-compiled versions do not?

looooo, 2023-02-23:
No, sry

herbkonik, 2023-02-23:
What we figured out at FC Forum: It sems somehow have to do with Qt 5.15.6 .
Qt 5.15.4 works at all systems which ere not working with Qt 5.15.6 and @mowley-bob uses Qt 5.15.8.
Wouldit be a big deal to switch the AppImage to one of this Qt versions?

mowley-bob, 2023-02-23:
I'm on Arch rolling release qt5-base is currently at 5.15.8.  I can re-test the appimage over the weekend and report back. :)

Strahlensauger, 2023-02-24:
The Appimage has to change to Qt 5.15.8, because the appimage delivers the qt libraries.
Using the appimage with QT5.16.8 installed on the computer doesn't change anything... 
My Arch is running Qt 5.15.8  and the issue in the appimage still persists...

KenwoodFox, 2023-03-15:
Thanks, coming from #8289 with the weekly appimage. Hope to see this fixed!

chennes, 2023-03-15:
I just reached out to the Qt Feedstock folks and they got 5.15.8 merged a few minutes ago. So hopefully the next Conda builds will have this fixed. When they come out, please give them a shot and report back about whether this solved the problem.

herbkonik, 2023-03-24:
Yesterday in installed
Version: 0.21.0.32457 (Git) AppImage
...
Python 3.10.9, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3

and the Problem seams to be solved. 
I didn't check each detail till now, but issues i hav had with numerical input and at Sketcher WB (also numerical input) are solved.

chennes, 2023-03-24:
Excellent, I'm going to mark this as closed then. If it recurs for anyone, please re-open.

closed 2023-03-24 by chennes