FreeCAD/FreeCAD #8347: [Problem] Not always second revolution around same axis possible

cnc50, 2023-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Version 0.20.2 (29177)
```

### Subproject(s) affected?

None

### Problem description

Wanted to revolve a second sketch around same axis.

[Iris Box Version.zip](https://github.com/FreeCAD/FreeCAD/files/10608907/Iris.Box.Version.zip)

Version 1 doesn`t work, version 2 works

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-05:
Full [About](https://wiki.freecad.org/About) info please

cnc50, 2023-02-05:
I revolved first sketch  and then another sketch around the same axis.
In the meanwhile I guess to know what is wrong. If there is no intersection
of these revolutions,
the second revolution isn`t shown. Otherwise no error is indicated.

cnc50, 2023-02-05:
Hi,

[image: image.png]

In Version 1 Revolution001  is not shown.
Thanks

luzpaz, 2023-02-05:
@cnc50 please log in to github directly instead of responding via email (because attachments don't save via email)

luzpaz, 2023-03-08:
@cnc50 ping

luzpaz, 2023-04-11:
bumping for visibility

Roy-043, 2023-07-15:
Version 1 violates the singe solid rule of the PartDesign Workbench. There should be a "does not intersect support warning". The missing warning is a bug.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

luzpaz, 2023-07-15:
@Roy-043 please rename the bug to accurately represent the issue, if you don't mind ?

closed 2025-01-13 by github-actions[bot]