FreeCAD/FreeCAD #8351: [Issue] GUI Fails to Render in 20.1 and 20.2

xloem, 2023-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
I tried both the 0.20.1 and 0.20.2 AppImage releases. The 0.20 AppImage release works.

[code]
OS: Red Hat Enterprise Linux (KDE/1-kde-plasma-standard)
Word size of FreeCAD: 64-bit
Build type: Release
Branch: (HEAD detached at 0.20)
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Core

### Issue description

The GUI does not render for me for versions after 0.20 . I am on RHEL 7.

### Anything else?

Screenshot from 0.20.1 . 0.20.2 looks just like this too:
[image]
Screenshot from 0.20:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-02-04:
Thanks for the report -- that has the smell of a hardware- or resolution- specific problem. Can you tell us more about what we are looking at here, hardware-wise?

xloem, 2023-02-04:
I resized the window to be small for the screenshots, but the hardware is relatively normal. It's a Thinkpad Anniversary Edition 25  .

The most likely culprit is that I have, in KDE System Settings -> Application Appearance -> Fonts, the fonts forced to a DPI of 48. This was to make better use of the screen real estate. I've changed other DPI settings elsewhere in the system, as well, and unfortunately don't remember all these changes.

The screen is natively 1920x1080 pixels, but I have the resolution scaled down to 960x540 and stretched to fill the display.
```
$ xrandr                                                                                                          
Screen 0: minimum 8 x 8, current 960 x 540, maximum 16384 x 16384                                                            
eDP-1-1 connected primary 960x540+0+0 (normal left inverted right x axis y axis) 309mm x 173mm                               
   1920x1080     60.01 +  59.97    59.96    59.93                                                                            
```
> ...
```
   960x540       59.96*   59.99    59.63    59.82                                                                            
```

xloem, 2023-02-04:
I disabled the forced font dpi and the gui now renders as in previous versions. :tada:

chennes, 2023-02-05:
Thanks for the further investigation, I'm glad you were able to narrow it all the way to that single setting. I guess we need to see if this is a known Qt bug, and if there is a workaround that doesn't involve you not using the setting 😃 .

adrianinsaval, 2023-02-12:
This is kind of a known bug in freecad, looks like it would be caused by qt+xorg but other qt apps work without issues (notably here all kde apps). Every time I've seen someone reporting this issue it was on xfce or other gtk based DEs though.

adrianinsaval, 2023-04-22:
I can't reproduce this at all, is this still a problem for you with the latest [dev version](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)? I think this has been solved some time ago

luzpaz, 2024-04-13:
bumping, what's left here ?

luzpaz, 2024-12-23:
ping @xloem