FreeCAD/FreeCAD #8357: [Feature Request]  Input alias or value in top bar of spreadsheet

Intuos5, 2023-02-05:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

-

### Subproject(s) affected?

Spreadsheet

### Idea description

In common spreadsheet editors like Excel or Google Sheets, you can use the top bars to input values into the cells. In FreeCAD, however, this is currently not possible.  To edit an alias, you currently have to either go to properties or use the alias button to go to the alias tab in the properties.

It would simplify things if you could type into the Alias or Content input boxes above spreadsheets directly.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-02-05:
Can you add your FreeCAD about info? This sounds like a bug to me, I can enter values in the editor boxes without issue.

Intuos5, 2023-02-05:
Yes, I am pressing enter. Also tried clicking off the box, none of those worked. 
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git)
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BIM 2021.12.0
  * Launcher
  * persistenttoolbars
  * SelectorToolbar
[/code]

adrianinsaval, 2023-05-11:
@Intuos5 are you still experiencing this problem on a recent build? I can't reproduce, pressing enter works correctly for both input fields

closed 2023-05-11 by Intuos5