FreeCAD/FreeCAD #8361: [Issue] Addon manager does not handle links in the WebTools addon description page properly

frodeseverin, 2023-02-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
Installed mods: 
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * fasteners 0.4.54
  * Help 1.0.3
  * MeshRemodel 1.8919.0
  * Pyramids-and-Polyhedrons
  * Reinforcement
  * retr3d
[/code]
```

### Subproject(s) affected?

Addon Manager

### Issue description

When browsing the WebTools addon in the Addon Manager, the links in the descriptions page triggers an error when clicked.
The linked documents open fine in my web-browser in the background, but the Addon Manager produces an error message, i.e.

> Could not load README data from URL https://github.com/yorikvanhavre/WebTools/blob/master/README.md

Please see attatched image and log file.

After such an error the Addon Manager will not display the addon description again, even after retruning to the list of addons.

There are two ways to make the Addon Manager display the addon description for the WebTools addon again

1. Use the "Refresh local cache" button to reload the cache
2. View the discription of a different addon in the Addon Manager, and then return to the WebTools addon description in the Addon Manager.

The links that triggers the error are e.g.

- [BimServer](https://www.freecadweb.org/wiki/Arch_BimServer)
- [BIM Server](http://www.bimserver.org/)
- [Scetchfab](https://www.freecadweb.org/wiki/Web_Sketchfab)
- [Sketcfab](http://www.sketchfab.com/)

### Anything else?

[image]
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/10611240/FreeCAD.log)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-02-05:
Thanks for the report: it's probably two errors interacting with one another. Obviously even if a link is bad, there should be some way of getting back to the previous page!

frodeseverin, 2023-02-07:
Please note that the issue I reported manifests itself even when I click on a valid link. At least for some types of links/targets.
The linked document opens a new tab in my default browser. When I return to my FreeCAD window, the Addon Manager dialog is still open, and the error 

> Could not load README data from URL https://github.com/yorikvanhavre/WebTools/blob/master/README.md

is displayed.

This is consistent on other addon description pages using a Readme.md on a Git-blob. I verified this problem to be present in the Readme.md page for 3D_Printing_Tools. Using the link indicated by a red arrow in the following image triggers the error.

[image]

Other links in the page, i.e. the one in the image below (1), does not trigger the error, and the linked document is rendered properly inside the Addon Manager. However, the descriprion page seems to be stuck on that page, and I am unable to navigate to the previous page, only forward using links not trigering the error. The Back-button (2) navigates back to the list of addons. 

[image]

Going directly back to the description of the same addon by clicking it in the list navigates to the page the view was stuck on previously.

[image]

It seems to me this error is in some way related to the render engine for HTML in the Addon Manager.

chennes, 2023-02-07:
I believe that the underlying bug is that when the Addon Manager has to open a link (good or bad) in the system's external browser (which will be most links, the AM deliberately refuses to load most links internally), I am not correctly handling what essentially needs to be a "refresh" in the AM's browser window.

frodeseverin, 2023-02-08:
The page for WebTools looks better now. Thanks.

Still, I think perhaps @chennes is on the right track here. The error 

> Could not load README data from URL (...Generic URL here...)

Appears every time the AM loads an URL in an external browser.

Also, there is no way to navigate back to the previous URL in the AM once a new URL is loaded internally.
The only way to revisit the original Readme.md for the Addon is to visist another Addon first, or Refreshing the local cache.
This might be best delalt with in a separate issue? Or perhaps I named this issue poorly in the first place?

chennes, 2023-02-08:
What you describe there is probably the real issue at work here -- I'm hoping to get to it later today.

luzpaz, 2023-03-01:
Hi @chennes, is this issue fixed ?

chennes, 2023-03-01:
Half of it is solved: there is still no "back" button for the browser. If you'd like to close this and open a more focused issue that's fine.

luzpaz, 2023-03-03:
Split the remaining issue in to #8731  
Closing.

closed 2023-03-03 by luzpaz