FreeCAD/FreeCAD #8365: [Issue] FreeCAD crashes after clicking adaptive path for 3rd time

JakeEhrlich, 2023-02-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 1397818c9664d158281339a84be59d3e9887f6ad
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
[/code]
If it helps this was from an official Ubuntu package
```

### Subproject(s) affected?

Path

### Issue description

I was just using the Path workbench and while I had the last adaptive path selected under my operations folder, I clicked adaptive path on the tool bar to add another adaptive path. FreeCAD instantly crashed leaving this backtrace

```
distance = 3.800000000000001mm    unit distance = 3.80 mm
solution count: 2
   solution 1:
      point1 -79   4.6   12.8
      point2 -79   4.6   9
      DeltaX 0
      DeltaY 0
      DeltaZ 3.800000000000001
      shape type on object1 is: Vertex
      shape type on object2 is: Vertex
   solution 2:
      point1 -79   93.65000000000001   12.8
      point2 -79   93.65000000000001   9
      DeltaX 0
      DeltaY 0
      DeltaZ 3.800000000000001
      shape type on object1 is: Vertex
      shape type on object2 is: Vertex

Failed to get shape

Failed to get shape

PathGeom.INFO: Zero working area to process. Check your selection and settings.
PathAdaptive.INFO: *** Adaptive toolpath processing started...

PathAdaptive.INFO: *** Done. Elapsed time: 0.006917 sec

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f838aedd090]
#1  0x7f8388ee1931 in QCheckBox::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x11
#2  /usr/lib/python3/dist-packages/PySide2/QtWidgets.cpython-38-x86_64-linux-gnu.so(+0x45dd2e) [0x7f8368760d2e]
#3  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a10bd) [0x7f83895ad0bd]
#4  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d) [0x7f8389380d6d]
#5  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xea8) [0x7f8389382018]
#6  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b) [0x7f838938c06b]
#7  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d) [0x7f8389380d6d]
[... 50 lines cut ...]
#58  0x7f8388ddba66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#59  0x7f8388de50f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#60  0x7f838a5e5218 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x88
#61  0x7f83883c180a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#62  0x7f83887aa7d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x6e3
#63  0x7f83887ac10b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x25b
#64  0x7f838878635b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#65  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x7f838226932e]
#66  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f838622e17d]
#67  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7f838622e400]
#68  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f838622e4a3]
#69  0x7f8388419435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#70  0x7f83883c03ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#71  0x7f83883c8116 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#72  0x7f838a65de0c in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x1b6c
#73  freecad(+0x52b4) [0x556822af32b4]
#74  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f838aebe083]
#75  freecad(+0x630e) [0x556822af430e]
```

I was not able to recover the CAM work that I had lost (it wasn't much luckily).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-05:
Official Ubuntu FreeCAD package is broken. Please try to recreate on AppImage, Flatpak, or Snap and report back.

sliptonic, 2023-02-22:
I'm unable to reproduce this.  Without a reproducible test case, I'm closing.  Will reopen if OP can reproduce it.

closed 2023-02-22 by sliptonic