FreeCAD/FreeCAD #8371: [Issue] Angle dimention not pointing to correct endpoint in version 0.20.0

Tcrawford49, 2023-02-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Issue description

The exercise "Gasket" has an angle component that when a dimention constraint is applied, the endpoint is not correct. When this file is loaded into version 19, it shows the correct endpoint.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-02-06:
Please do a thorough search before submitting issues, this is already logged under #8220

luzpaz, 2023-02-07:
Closing as dupe.

closed 2023-02-07 by luzpaz

Almisuifre, 2023-02-17:
Good morning,
Indeed, it is a duplication. But as a result, are there any advances?

luzpaz, 2023-02-17:
Please continue all discussion on #8220