FreeCAD/FreeCAD #8388: All parts disappear and get grayed out in the combo view [Problem]

TheRealDestroyer, 2023-02-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

i was attemptint to click on a surfase and all hell broke lose. the software tried to link things and when i finaly got that to stop everything just- poofed. in the combo view everything was marked for recompute and was greyed out and in the model view there was nothing. i tried to recompute and nothing changed.

### Anything else?

[Arduino Case.zip](https://github.com/FreeCAD/FreeCAD/files/10679461/Arduino.Case.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-09:
Is it possible for you to also reproduce this on the [development version](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) ?

Roy-043, 2023-07-15:
If you recompute the file and make the tip of the body visible all is OK. Closing this one...

closed 2023-07-15 by Roy-043