FreeCAD/FreeCAD #8394: [Problem] Instant Segmentation Fault when Opening Problematic File

florianehmann, 2023-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
  * Assembly4 0.12.5
[/code]
```

### Subproject(s) affected?

None

### Problem description

Attempting to open the problematic .FCStd file (attached below) instantly crashes FreeCAD with a segmentation fault reported in the console output. 

I modeled some geometry in a file and after FreeCAD failing to perform a Boolean fuse of the bodies in the Part Design workbench, I restarted FreeCAD and noticed that it crashes every time I try to load the file.

This is true for the provided stable version as well as the development version
```
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: (HEAD detached at d28d63b87b)
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

### Anything else?

The problematic file is available at [this link](https://file.io/I8xlUWdZmNlw)

and the while attempting to open the file the console reads
The console reports
```
FreeCAD 0.20.2, Libs: 0.20.2R29603 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench (v0.12.5) loaded
<Part> ViewProviderExt.cpp(1269): Cannot compute Inventor representation for the shape of Core_Post_100#Body002: Geom_BSplineCurve::Segment
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f5572168090]
#1  0x7f54c452c7e1 in BRepLib_MakeFace::Init(opencascade::handle<Geom_Surface> const&, double, double, double, double, double) from /tmp/.mount_FreeCALUIaQa/usr/lib/./libTKTopAlgo.so.7+0x131
#2  0x7f54c452fc6a in BRepLib_MakeFace::BRepLib_MakeFace(opencascade::handle<Geom_Surface> const&, double, double, double, double, double) from /tmp/.mount_FreeCALUIaQa/usr/lib/./libTKTopAlgo.so.7+0x6a
#3  0x7f54c457b8c2 in BRepBuilderAPI_MakeFace::BRepBuilderAPI_MakeFace(opencascade::handle<Geom_Surface> const&, double, double, double, double, double) from /tmp/.mount_FreeCALUIaQa/usr/lib/./libTKTopAlgo.so.7+0x72
#4  0x7f54c59e2134 in Part::Tools::triangulationOfFace(TopoDS_Face const&) from /tmp/.mount_FreeCALUIaQa/usr/lib/./Part.so+0x1c4
#5  0x7f54b9621554 in PartGui::ViewProviderPartExt::updateVisual() from /tmp/.mount_FreeCALUIaQa/usr/lib/./PartGui.so+0x6b4
#6  0x7f54b9623db3 in PartGui::ViewProviderPartExt::onChanged(App::Property const*) from /tmp/.mount_FreeCALUIaQa/usr/lib/./PartGui.so+0x173
#7  0x7f55738550e5 in App::Property::hasSetValue() from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0x135
#8  0x7f55738bf684 in App::PropertyBool::Restore(Base::XMLReader&) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0xb4
#9  0x7f55738602c0 in App::PropertyContainer::Restore(Base::XMLReader&) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0x1110
#10  0x7f557427f48f in Gui::Document::RestoreDocFile(Base::Reader&) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADGui.so+0x28f
#11  0x7f557346b435 in Base::XMLReader::readFiles(zipios::ZipInputStream&) const from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADBase.so+0x1a5
#12  0x7f55737407ae in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0x37e
#13  0x7f557390ad18 in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0x3d8
#14  0x7f557390bab5 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0x815
#15  0x7f557390d8a8 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0xc8
#16  0x7f557390da5f in App::Application::processFiles(std::__cxx11::list<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADApp.so+0x13f
#17  0x7f55746f2f21 in Gui::MainWindow::delayedStartup() from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADGui.so+0xb1
#18  0x7f55726f1f0e in QObject::event(QEvent*) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5+0xf0
#19  0x7f5572f6b0f7 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Widgets.so.5+0xd9
#20  0x7f55742b5da9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADGui.so+0x89
#21  0x7f55726db608 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5+0x8e
#22  0x7f55726db86f in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5+0x1fb
#23  /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5(+0x1f37dd) [0x7f557270c7dd]
#24  /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_dispatch+0x25c) [0x7f556eb361bc]
#25  /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/./libglib-2.0.so.0(+0x58489) [0x7f556eb36489]
#26  /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f556eb36551]
#27  0x7f557270c3f3 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5+0x5b
#28  0x7f55726d918a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5+0x186
#29  0x7f55726dd3b5 in QCoreApplication::exec() from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libQt5Core.so.5+0x107
#30  0x7f557423a77e in Gui::Application::runApplication() from /tmp/.mount_FreeCALUIaQa/usr/bin/../lib/libFreeCADGui.so+0x1b2e
#31  /tmp/.mount_FreeCALUIaQa/usr/bin/freecad(+0x68e5) [0x563945a698e5]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f5572149083]
#33  /tmp/.mount_FreeCALUIaQa/usr/bin/freecad(+0x6c29) [0x563945a69c29]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-02-24:
I cannot reproduce this bug with the latest stable release on Windows

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

When opening the file, FreeCAD outputs the following:

```
19:05:07  <Part> ViewProviderExt.cpp(1270): Cannot compute Inventor representation for the shape of Core_Post_100#Body002: Geom_BSplineCurve::Segment
19:05:07  Illegal storage access...
19:05:07  <Part> ViewProviderExt.cpp(1273): Cannot compute Inventor representation for the shape of Core_Post_100#AdditiveLoft
19:05:07  <Part> ViewProviderExt.cpp(1270): Cannot compute Inventor representation for the shape of Core_Post_100#Body002: Geom_BSplineCurve::Segment
```

racer-coder, 2023-02-24:
I get a similar error message (without the crashing) in #8640 when using Part, but when doing essentially the same thing in PartDesign I get "Resulting shape is not a solid,"  The bug may or may not be related to the same underlying cause.  #8640 does have a file that can be opened in FreeCAD without crashing, possibly making it easier to debug/diagnose.
[loftbug.zip](https://github.com/FreeCAD/FreeCAD/files/10829160/loftbug.zip)

bgbsww, 2023-10-31:
Some hints along the way here:

Unpack the Core Post 100.FCStd.gz file ( gunzip to get the freecad file, then unzip that file to get the constituent files )
Try importing the PartShape<N>.brp files.  The crash occurs trying to import either PartShape21.brp or PartShape22.brp.
Run a gdb session against a build with symbols, and you discover the crash at Mod/Part/App/Tools.cpp:643
Setting a breakpoint at Mod/Part/App/Tools.cpp:642 reveals that the call to make a surface from the face returns null.  Feeding that null on causes the segment fault.

Looking around the source code, there is some minor precedent for testing this return, IE `if ( !aSurface.IsNull() )` although it is rare.

Is adding a null defense here a correct fix?

PR #11329 (2023-11-20): Defend against surface not created from invalid face
Fix #8394 

I'm unsure about this, as there was no precedent for throwing an exception in this source module.  It does seem to operate correctly, but maybe there's a more appropriate solution.
closed 2023-11-20 by adrianinsaval