FreeCAD/FreeCAD #8413: Noob for 10 seconds and already have problem: Pulldown menus don't work following View > Fullscreen

yaxy, 2023-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

After switching to full screen (either with Alt-F11 or clicking View > Fullscreen), the pull down menus stop working. Mouse hover over File, Edit, View, etc. turn background light blue, but clicking does nothing. Same with keyboard shortcuts e.g. Alt-F. Same with workbench selector pulldown and Blender/CAD pullup menu.

Update: When changing app focus (using Alt-Tab or mouse), say, from browser to full screen FreeCAD (I have 2 monitors) and vice versa, FreeCAD flashes and for a split second, the items in the pull down menu appear then disappear.

Fortunately FreeCAD hasn't crashed. Alt-F11 to exit Fullscreen restores normal pulldown operation.

So much for my new career making training videos by screen capturing FreeCAD.

### Anything else?

* GitHub project search for "pulldown" and "pull down" yielded nothing related to this problem.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-10:
Can you recreate with https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

yaxy, 2023-02-10:
Problem goes away using Version: 0.21.0.31812. Thanks!

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31812 (Git)
Build type: Release
Branch: master
Hash: c074fe1c95e5aa4f21a11d1387c2ae38f5014a4d
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

luzpaz, 2023-02-10:
@donovaly heads up. Do you think this is Qt or Python related ?

donovaly, 2023-02-10:
Confirmed. However, when compiling using latest master and using the same library package than for the FreeCAD 0.20.2 build, the bug is not there.
So apparently it is already fixed.
Too bad that I don't know by what commit, to backport the fix for FreeCAD 0.20.3.

donovaly, 2023-02-10:
@luzpaz can you reproduce the issue with FreeCAD 0.20.2 using Linux or macOS?

luzpaz, 2023-02-10:
Can't test on macOS. My linux box is still not in full operational status.

adrianinsaval, 2023-02-23:
> So apparently it is already fixed. Too bad that I don't know by what commit, to backport the fix for FreeCAD 0.20.3.

I believe #8027, I'll check behavior on linux later

luzpaz, 2023-02-26:
Closing ticket since it's fixed.

luzpaz, 2023-02-26:
Sorry, it hasn't been tested on Linux yet.

adrianinsaval, 2023-02-26:
works for me 

> OS: Arch Linux (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.21.0.32070 (Git)
> Build type: None
> Branch: master
> Hash: 849f8ced135e6c068afef8580da67e599a043f37
> Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
> Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
> Installed mods:

closed 2023-02-26 by luzpaz

buccaneer-jak, 2023-06-16:
Menu doesn't work for me in V 0.20.2 although I see many reports of this.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]