FreeCAD/FreeCAD #8417: Global key binings supersede entry field / Cannot enter numbers 1-6

rklasen, 2023-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Freecad weekly appimage as of 2023-02-10 (sorry, Freecad is currently not installed, I'll update the full version info later)
```

### Subproject(s) affected?

None

### Issue description

When the focus is in an entry field (like dimensions of a cube or something), the numbers 1-6 cannot be entered (neither home row nor numpad. Instead, the view changes.

It seems the global shortcuts for view changes supersede the current input field, and I suspect the same issue occurs here

Possibly related issue #8346 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rklasen, 2023-02-10:
I've just compiled from source (using Archs freecad-git) and the issue is not there. So it seems to be in the appimage only.

luzpaz, 2023-02-10:
Should we transfer to FreeCAD-bundle ? (@looooo ?)

adrianinsaval, 2023-02-23:
this is a very serious issue but I don't know how to track it down, @0penBrain could the dot substitution mechanism have anything to do with this?

0penBrain, 2023-02-23:
No, I hardly can help. This is an issue with AppImage, that is almost appearing only on Arch.
It probably has to do with the compose table, both embedded in AppImage and provided with Arch.
When I test here, I see a "failed lo load compose table" message in the report view, but I can't reproduce the more severe issues that Arch users have.
Just for French keyboard the dead keys aren't working anymore.
This would be hard to debug.
In the forum (there are multiple such reports), I asked @looooo if he knows if something changed in the compose tables of the AppImage but got no answer.

chennes, 2023-02-28:
https://github.com/conda-forge/qt-feedstock/pull/239

chennes, 2023-02-28:
I'm going to close this as a duplicate of #8346 -- the problem appears to be with the specific version of Qt used in the feedstock. There is a PR in to update to Qt5.15.8, which should resolve the issue.

closed 2023-02-28 by chennes