FreeCAD/FreeCAD #8426: Part Cube property editor is broken and changes set values when reentering position values for x, y, and z

pcoe1959, 2023-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20.2
29177 +426 (Git)
2022/12/07 02:35:37
Ubuntu Core 20
64-bit
tag: 0.20.2
930dd9a
```

### Subproject(s) affected?

Part

### Issue description

My units are set to Building US (ft-in/sqft/cft). Create a cube in part and try to position it via the cubes properties->Placement->Position, x y and z are set to 0 change any one of them and set its  value to a negative to move it below its axis, leave the field and try to adjust it, it will change to something other than what it was when you first edited it forcing you to renter the entire number even though no edit has occurred yet, it appears that the fields On_Enter handler has gone amok. It is really frustrating when trying to fine tune positioning. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-13:
Please add an example file

adrianinsaval, 2023-04-26:
To reproduce set unit schema to Building US and on almost any input field (say a Part Cube's Position x) input for example -1.5" and hit enter this will then be displayed as -1+1/2" when clicking again to edit this field this will be evaluated by the parser and change to -1/2" instead

adrianinsaval, 2023-04-26:
@ConnorWorrell may you have a look into this one too? I'm not sure if this is possible to solve on the parser since it is correct to interpret -1+1/2 as -1/2, maybe we can make the Building US scheme use parenthesis when dealing with negative numbers?

ConnorWorrell, 2023-04-27:
I see a few solutions,

1. Handle the division in the expression parser -1'1/2", this seems not great because there is no easy way to get a dimension on the bottom of the fraction, ex: 1in*1/2in -> .5in^2 instead of .5 unitless the current expected result.
2. Parentheses -(1+1/2) , this is straight forward, but since this change would happen in the scheme these parentheses would show up everywhere, and it annoys me because they look just like reference dimensions. This might also lead to bad inputs, I often edit a box and type +1" to add a bit of length to something, but here the result would be -(1+1+1/2) which is the opposite of what was intended.
3. Carry the negative through -1-1/2, this seems like the cleanest.

adrianinsaval, 2023-05-31:
fixed by #9408

closed 2023-05-31 by adrianinsaval