FreeCAD/FreeCAD #8429: [Issue] freecad issue to export compound (or other object) in stl format

hook101, 2023-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

freecad: "can't determin mesh format..."

### Version

0.20 (Release)

### Full version info

```shell
Can't determine mesh format from file name: '/home/rap etc.
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

when trying to export a file in STL, freecad answer:"can't determin mesh format for..."

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-02-11:
Hi,
you have to specify the filename extension.

luzpaz, 2023-02-11:
@FlachyJoe perhaps this merits updating the error message with a hint?

luzpaz, 2023-02-11:
Location in source:  
https://github.com/FreeCAD/FreeCAD/blob/c437b79adabb6a91e8fb6dc46729cb720ba16a5f/src/Mod/Mesh/App/AppMeshPy.cpp#L270

luzpaz, 2023-02-11:
Probably adding something like `\nPlease specify mesh format file extension.`

hook101, 2023-02-11:
Hello there, thank you gents for your help.
FlashyJoe is right, the extension .stl was missing when I put on the target file.
I thought freecad was doing it by itself, or is it the new version, I don't know. Anyway, the solution is to put the extension format.
Thank you FlashyJoe.

closed 2023-02-11 by hook101