FreeCAD/FreeCAD #8432: [Problem] User Models break when topological entities change name (Topologocial Naming Issue)

sliptonic, 2023-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31608 +3 (Git)
Build type: Unknown
Branch: (HEAD detached from upstream/pr/8186)
Hash: c92ddbd6a4d21a7b13eb011c432d64828948bc00
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.5
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.59
  * Nodes 0.1.36
  * FeedsAndSpeeds 0.5.0
  * ExtremeProDark 2.6.4
  * FeedsAndSpeeds.bak
  * Curves 0.6.5
  * Dracula 0.0.2
[/code]
```

### Subproject(s) affected?

Core

### Problem description

The FreeCAD Topological Naming issue is well known.  It has been a concern for years but no issue exists for it.
The resolution of this issue will involve many individual Pull Requests and probably create numerous secondary issues.

This issue is being created to centralize information, coordinate discussion, and track related topics.

Realthunder has demonstrated a viable solution to the problem in his Linskstage3 branch.  Merging this into master has proven difficult for a variety of reasons.  

### Anything else?

[This video](https://www.youtube.com/watch?v=QSsVFu929jo) gives a reasonable layman's introduction to the problem.
[
This paper ](https://link.springer.com/content/pdf/10.1007/978-0-387-35187-2_34.pdf)gives a more scholarly description of topological naming in general

Realthunder has [described how his algorithm works](https://github.com/realthunder/FreeCAD_assembly3/wiki/Topological-Naming-Algorithm)

```[tasklist]
### Related Issues 
- [ ] https://github.com/FreeCAD/FreeCAD/issues/7657
- [ ] https://github.com/FreeCAD/FreeCAD/issues/6789
- [ ] https://github.com/FreeCAD/FreeCAD/issues/5897
- [ ] https://github.com/FreeCAD/FreeCAD/issues/5755
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-11:
JFTR, original Toponaming issue was #6317 which was #[922](https://tracker.freecad.org/view.php?id=922) on Mantis by Jurgen in 2012.

JohnAD, 2023-02-22:
An almost on-topic question: any recommendations for a good way to do breakpoint-style debugging the C++ code in FreeCAD. I figured I'd ask before I started diving in head-first. Up to this point I've just been analyzing code and mapping out connections.

My normal flow for cpp work is JetBrains CLion for my editor and a newish 16-core laptop running PopOS! (an Ubuntu variant). But I've got lots of tools available including Visual Studio and a 4-core Win10 laptop; and I am happy to switch if there is an advantage.

berniev, 2023-02-23:
> any recommendations for a good way to do breakpoint-style debugging the C++ code

I'm concerned.

As someone who has been working with C++ for less than a year I was hoping to learn something from the methodology of the paid professional. I was surprised then to find him needing help to do something as trivial as setting a breakpoint. I looked at his YouTube channel, and nothing there is C++ related, and his only C++ github repository is FreeCAD.

sliptonic, 2023-02-23:
> > any recommendations for a good way to do breakpoint-style debugging the C++ code
> 
> I'm concerned.
> 

You just questioned the credentials of a new contributor to the project by misquoting him.  You left off the _in FreeCAD_ part of his question.  This is a person you've never met and your first comment is meant to undermine him.  You should be ashamed.  The Github issue discussions should have a higher standard of professional discourse and courtesy.  In the future, please try to keep your comments constructive.

chennes, 2023-02-23:
Just so it's recorded someplace: the trick to getting CLion to work with FreeCAD is you need to explicitly "Build All" -- for some reason it doesn't get the "ALL_BUILD" target that you would have in Visual Studio.

berniev, 2023-02-24:
I too use CLion and there was no trick when I did this a couple of weeks ago. Just do debug build and away you go. Breakpoints work immediately.

JohnAD, 2023-02-27:
You all have a preferred place for a Gantt chart? If not, I'll throw something into a public repo using https://www.diagrams.net/ in the next few days.

JohnAD, 2023-03-01:
@realthunder 

Does the following table and interpretation from the code look correct to you:

| elementMap key | meaning       |
|----------------|---------------|
| `:H`           | tag (local)   |
| `:C`           | child element |
| `:X`           | external tag  |
| `:I`           | index         |
| `D`            | duplicate     |
| `:T`           | decrement? tag |
| `:M`           | mod           |
| `G`            | modgen        |
| `:G`           | gen           |
| `:U`           | upper         |
| `:L`           | lower         |

And anything else is most likely an operation.

(for ref: look through code in `ComplexGeoData.cpp`, `TopoShapeEx.cpp`, and `src/Mod/Part/App/TopoShapeOpCode.h`)

And so, the interpretation of the simple `Unnamed#Fusion.;Face6;:M2;FUS;:H1ab:8,F.Face10` (asuming Hashing turned off) would be

`Unnamed` = the documents name

`#` = doc / object separator

`Fusion` = object name

`.` = object / elementMap name separator

**MappedName portion begins**

`;` = elementMap entry prefix

`Face6` = the item's historic parent element name

`;` = elementMap entry prefix

`:M` = marker of modification
`2` = the object at index 2 (the third object)

`;` = elementMap entry prefix

`FUS` = modification was a fusion operation; see `src/Mod/Part/App/TopoShapeOpCode.h`

`;` = elementMap entry prefix

`:H` = This is a TAG
`1ab` = the tag's id in hex (would be '-' if it didn't have one)
`:8` = marker indicating length of _previous_prefix, in decimal
`,` = comma separator for next parameter for ':H'
`F` = This is a "Face"; literally the first letter of the actual name

**MappedName portion stops**

`.` = MappedName / element's name separator

`Face10` = the actual name for this element

JohnAD, 2023-03-01:
@berniev 

I apologize for not responding sooner. I am a generalist. While it is true that I've been using C++ since version 2.0 in the late 90s; that isn't enough. I've learned that most projects; especially ones written in non-idiomatic languages like  C++, C, and assembler, are never in a predictable pattern. Each project has a team that has learned behaviors, formatting, tools, and institutional knowledge. So, I'm happy to ask basic questions like "what is the recommended way of doing such and such". Not just because it can help (it can), but the questions usually bring in more insight.

Any more, I also like to pair program for a bit with a new client, but I'm not sure I can pull that off with an open source group :-)

As to the need to do ALL_BUILD at least once. On my machine, it seems to be a requirement as CLion is not detecting that a lot of the supporting elements have not been generated/moved/etc (such as SVG files, etc.) until it is run at least once. After that, I can simply use the 'Debug FreeCadMain` icon on the toolbar and away it goes.

realthunder, 2023-03-02:
@JohnAD In my code, I call those `elementMap key` that you listed as `postfix`, which includes the `elementMapPrefix()`, i.e. `;`.

Correction: the `modgen` postfix is `;:MG`, because in the code, it is defined as `modPostfix() + "G"`.

`;:T` is an deprecated version of the `tag` postfix, serving the same purpose of `;:H`. `dec` in the code means decimal because it encodes object ID using decimal digits instead of hex. The postfix is only used in one of my old branch. It is included for backward compatibility.

Just some quick notes. I'll add more later.

Zolko-123, 2023-03-02:
Wouldn't the first step of investigating this TopoNaming issue be to trace the usage and whereabouts of the 2 functions `setElementName()` and `getElementName()` in **realthunder**'s branch ?

https://github.com/realthunder/FreeCAD/blob/LinkDaily/src/App/ComplexGeoData.cpp#L1587
https://github.com/realthunder/FreeCAD/blob/LinkDaily/src/App/ComplexGeoData.cpp#L1456

at least that's where I would begin the journey

JohnAD, 2023-03-02:
Breaking this into to planning parts. First, the general pattern:

```mermaid
flowchart TD
    id1([Prepare])
    id2([Implement])
    id3([Integrate])
    id4>Enable]
    id5(Optimize)
    id1 --> id2
    id2 --> id3
    id3 --> id4
    id4 --> id5
    id5 --> id5
```
1.  (Prepare)
Create the supporting functions needed by element naming and the algorithm.
These are simply libraries with good unit testing to ensure they are accurate when used later.
**FreeCAD behavior is unchanged for users**
2. (Implement)
Create the naming-of-elements algorithm.
Create the ElementMap class.
Using the previous libraries, import the general algorithm that will _eventually_ be called by ComplexGeoData.cpp to have the more advanced naming.
Using unit tests, verify that the naming is very predictable and reliable.
**FreeCAD behavior is unchanged for users**
3. (Integrate)
Add the naming, but do not use it yet.
Add the members to the ComplexGeoData.cpp class.
While the member values are set, they are not actually used.
Add support for visibility in the Selection View.
   **This stage will slow down FreeCAD and will make the naming visible on an element when selected.**
4. (Enable)
Use the toponaming so that the model is more resilient to upstream object changes.
**Users will start seeing increased stability.  Performance is still impacted**
5. (Optimize)
Selectively add optimizations to speed FreeCAD up again.
This could include: MappedName-hashing and IndexedName-int:string-caching.
**Repeat**
```

JohnAD, 2023-03-02:
Individual steps so far. I'm sure there will be more. Any and all insight/feedback is great as you all have much experience with the structure of FreeCAD.

Much of this is mild reorganization since the basics are already written by RealThunder. But adding test cases and slower migration will help with organizing this stuff in our collective heads.

1. (Prepare)

   * IndexedName

     see `indexedname.md`

   * MappedName

     see `mappedname.md`

   * ElementMap.h

     see `elementmap-h.md`

   Are there any other clean libraries we could write to help with the following steps?

2. (Implement)

   * ElementMap.cpp

     see `elementmap-cpp.md`

3. (Integrate)

   * Basics and members added to ComplexGeoData.cpp/.h

   * Adjustments to Part workbench

   * Adjustments to Sketcher workbench

   * Adjustments to ComplexGeoDataPy

4. (Enable)

   * Adjustments to Save / Restore processes

   * Adjustments to Part workbench

   * Adjustments to Sketcher workbench

5. (Optimize)

   * consider MappedName hashing

   * consider IndexedName int:string map cache per-project or system-wide.

JohnAD, 2023-03-02:
The steps are not in a gantt chart yet as I'm still untangling order in my head for steps 3 and 4; but I'm not far.

sliptonic, 2023-03-02:
Here's the start of a Gantt chart.  

- Start date arbitrarily set to 3/15 just to work on the chart.
- All task dates are currently set to 10d because better estimates aren't available yet.
- Task concurrency is strictly by phase now.  That will likely change.  
- This chart will undergo heavy editing in the future.  

**In other words, assuming anything about 'when' by reading this chart is foolish.  It's guaranteed to be wrong.**

```mermaid
gantt
    title Toponaming Fix Plan
    dateFormat  YYYY-MM-DD

    section Prepare
    start            : milestone, m1, 2023-03-15,2min
    IndexedName      :p1, after m1, 10d
    MappedName       :p2, after m1  , 10d
    ElementMap.h     :p3, after m1  , 10d
    end            : milestone, m2, after p3,2min

    section Implement
    ElementMap.cpp   :i1, after m2  , 10d
    end           : milestone, m3, after i1, 2min

    section Integrate
    Basics and members added to ComplexGeoData.cpp/.h  :ii1, after i1  , 10d
    Adjustments to Part workbench       :ii2, after i1  , 10d
    Adjustments to Sketcher workbench   :ii3, after i1  , 10d
    Adjustments to ComplexGeoDataPy     :ii4, after i1  , 10d
    end            : milestone, m4, after ii4 ,2min

    section Enable
    Adjustments to Save / Restore processes    :e1, after ii4  , 10d
    Adjustments to Part workbench              :e2, after ii4  , 10d
    Adjustments to Sketcher workbench          :e3, after ii4  , 10d
    end            : milestone, m5, after e3, 2min

    section Optimize
    MappedName hashing?                                             :o1, after e3  , 10d
    IndexedName? int:string map cache per-project or system-wide.   :o2, after e3  , 10d
    end            : milestone, m6, after o2, 2min
```

adrianinsaval, 2023-03-03:
> * Adjustments to Part workbench
> * Adjustments to Sketcher workbench

Considering that sketcher is the one that will get the strongest stability and already provides a los of functionality outside of it, and in light of many previous comments from zolko, is it possible to do sketcher first and then Part? Or is the Part side an unavoidable prerequisite for doing it in sketcher?

adrianinsaval, 2023-03-03:
> 3\. Add support for visibility in the Selection View.
>     **This stage will slow down FreeCAD and will make the naming visible on an element when selected.**

This is very nice and desired (and I think not currently provided by rt's solution), but the mapped names seem very long and not too user friendly, is this what will be displayed? Can we find something else? Like unique numbers or something.

JohnAD, 2023-03-03:
> This is very nice and desired (and I think not currently provided by rt's solution), but the mapped names seem very
>  long and not too user friendly, is this what will be displayed? Can we find [...]

The long mapped names would only be seen in the Selection View window. Elsewhere only the short name is seen. And, certainly, we can even change that behavior later during the optimization phase. There is no real need for the end user to see the mapped names; it is more of a developer convenience.

Perhaps during the optimization dev phase, we could have a global flag settable in the Python Console that lets developers see the mapped name, otherwise the simpler name is shown?

JohnAD, 2023-03-03:
I've made edits to the charts above based on changes made in a meeting with other devs this morning.

JohnAD, 2023-03-07:
I've added the first 4 cards as Issues and edited the above chart. I'll start generating Phase 2 cards next. Those get somewhat more complicated but shouldn't be too bad.

pinpox, 2023-03-07:
I'm subscribed to this issue  because I stumbled across the topo naming problem a few times and am excited for upcoming fixes for it. I read through this issue, but have trouble understanding if it will be fixed by this and what the fix will actually be. 

Is the problem, current state and solution documented somewhere? As a newish user of freecad it would be helpful to know how likely this will solve the problem from a users perpective and what the connection to @realthunder 's branch is. Should I be using that? Should I just wait for the next official release? 

I assume I'm not the only one stumbling over this issue, maybe someone can link an issue to read up on the plans

macdroid53, 2023-03-07:
> I'm subscribed to this issue because I stumbled across the topo naming problem a few times and am excited for upcoming fixes for it. I read through this issue, but have trouble understanding if it [...]

This covers t pretty well: https://ondsel.com/blog/freecad-topological-naming/?fbclid=IwAR0cS7C-WK42n_1lwwqIS_Le8ULOwfJk6aMRT5updGgcGVxxgntPgNMmUGA

adrianinsaval, 2023-03-07:
> The long mapped names would only be seen in the Selection View window. Elsewhere only the short name is seen. And, certainly, we can even change that behavior later during the optimization phase. There is no real need for the end user to see the mapped names; it is more of a developer convenience.
 
I agree the user doesn't need to see the mapped name but a stable name would be appreciated, could we make the short name stable too? currently it's constructed from a dumb indexing of all subshapes in an object, but there is no real reason why we should keep following this pattern for the user facing name as it will never be stable, on the other hand the mapped name is too long for direct human usage. Can we maybe have a table with 3 identifiers for each element?
The current index id which can change on recompute
realthunder's mapped name which is used to identify the correct element in the index for a recomputed shape
and a third user friendly name that will be associated with the mapped name and won't change on recompute, this third name then can be referenced by the user safely and it would be what is shown in the UI everywhere.

see also #7657, for sketcher the mapped name itself could be used as it is short but for complex shapes with a long history it's too long, perhaps we can use the hash as a stable user facing name?

JohnAD, 2023-03-25:
Done with Phase One!

I wanted to express my gratitude for all the hard work everyone put into phase 1 of this project. It was a massive undertaking, and you all went above and beyond to import/test/refactor RealThunder’s working code!

Now it is time for phase 2 and I can't wait to see what we can accomplish together.

While I have been researching what should be in this phase, I’m not the FreeCAD expert you all are; so I’m asking for help! I’m hoping to have all of the next steps laid out by Friday, and getting comments and suggestions from the experts is invaluable!

JohnAD, 2023-03-25:
The general idea for phase 2 is adding the algorithm but not actually using it yet. The big goal is good consistent testing; and then updating/modifying that testing in phase 3 as we inevitably learn more.

Specifically, it will be adding the naming algorithm. Some of that is found in the ElementMap class. I suspect that will be step 1 of phase 2.

And, I was thinking this will also mean that we go ahead and add the `ElementMapPtr _ElementMap` item to the ComplexGeoData class with its supporting methods (and possibly TopoShape). However, those methods would not yet be called in the live FreeCAD code. They are only called by the testing files.

We might also want to add indirect support into the Sketcher workbench, but I’ve not really started looking at it to know what that would involve. **Any Sketcher enthusiasts around?**

chennes, 2023-03-26:
Thanks, @JohnAD, your organization of the project has been invaluable, this is an excellent team effort. And aren't we **all** Sketcher enthusiasts?! 😁 Of course, @abdullahtahiriyo is really The Dude there.

abdullahtahiriyo, 2023-03-26:
@JohnAD 

Feel free to ask any questions you want about the existing code of the Sketcher. Hopefully, I should be able to answer most.

One issue that was "synchronised" in the past between RT's branch and master relates to the storage of IDs for geometry. The Sketcher uses ```Part::Geometry``` for storing geometry related data. 

Here I set strong boundary that any Sketcher related data structures should not pollute Part. 

From the many solutions possible, the one chosen was to provide support for what is called ```Part::GeometryExtensions``` (not to be mixed up with App:Extension, it has nothing to do). A Part::Geometry has the ability to store a number of extensions. This enables every module using ```Part::Geometry``` to see just a ```Part::Geometry``, and then every module will know about its extensions, which can be used to retrieve module specific data. This framework is also used to separate concerns between App (```Sketcher```) and Gui (```SketcherGui```), as some extensions carry pure visualisation data (scaling factors, visual layer Ids) while others carry core App data (identifiers, construction/nonconstruction, type of internal geometry, ...). 

All that is just to provide some basic background to where the geometry data is stored and why. 

So coming back to storage of geometry identifiers, ```Sketcher::SketchGeometryExtension``` is the extension where all the identifiers for regular geometry are stored. This includes the Id (getId) that is only used in RT's branch. In fact, while it exists in the class ```Sketcher::SketchGeometryExtension``` it is not even serialised during persistence (it is serialised in RT's branch only). This was done so that data is not inconsistent between RT's branch format and master, and to facilitate future migrations. So the piece of information here is: that extension is where regular geometry identifiers are to be stored.

To enable a transparent use of Geometry and ```SketchGeometryExtension``` at Sketcher level,, ```Sketcher::GeometryFacade```  was created, which is nothing but a facade to be able to access any relevant information about regular geometry directly, no matter if it is stored in the ```Part::Geometry``` or in the sketcher extension.

One meaningful difference between RT's branch and master that has some bearing in toponaming is external geometry. External Geometry comprises two aspects: (1) geometry aspect (Part::Geometry) and (2) link aspects (where outside the sketcher they connect to). This is stored as an internal array at SketchObject level for master, but it is (or perhaps was) integrated into geometry in RT's branch.

In master there is a ```Sketcher::ExternalGeometryExtension``` that was created to store the data necessary for RT's branch. It is unused in master. It was put there to limit divergence of RT's branch, so that in the future it would be easier to merge any development.

I think that is about what the Sketcher has as in-place regarding interfaces with toponaming code. Feel free to ask questions.

Unrelated to this interface, but related to code changes activity:
I am currently thinking about how to refactor the Sketcher to make use of smartpointers for geometry, so basically going from ```Part::Geometry *``` to ```std::shared_ptr<Part::Geometry>``` and its weak pointer counterpart. This means there are likely to be changes to ```Part::PropertyGeometryList``` and ```Sketcher::GeometryFacade```, which will call for quite some code changes all around (mostly within the Sketcher WB, other than the list itself), as that is the most used data structure in the Sketcher.

Those changes should not impact toponaming support, but depending on what exactly the current Sketcher toponaming support entails, they might lead to (easy to resolve but numerous) prospective merge conflicts.

JohnAD, 2023-03-29:
I've created two WIP issues for the next phase (#9115 and #9116). Feel free to comment on them but please don't start any work on them as a they might still be wrong.

JohnAD, 2023-03-29:
Just added WIP issue #9119 . This is easily the card I'm least confident in as I've done the least research.

@abdullahtahiriyo does this card seem too open-ended? Should I break it up into smaller cards? Are there better ways to say things?

Actually, now that I re-read your earlier comment, I'm going to look further into the `GeometryFacade` part next.

JohnAD, 2023-03-30:
Eventually, it would be nice if some of the unit tests ran the CPP equivalent of this python script:

```py
# Arrange

box = App.ActiveDocument.addObject("Part::Box", name="box")
cylinder = App.ActiveDocument.addObject("Part::Cylinder", name="cylinder")
newShape = box.Shape.fuse(cylinder.Shape)
newObject = App.ActiveDocument.addObject("Part::Feature","newobject")
newObject.Shape = newShape
Gui.Selection.clearSelection()
Gui.Selection.addSelection('Unnamed','box')
Gui.Selection.addSelection('Unnamed','cylinder')
Gui.runCommand('Std_ToggleVisibility',0)
Gui.Selection.clearSelection()

# Act - get the Element Names

App.ActiveDocument.recompute()
em = newObject.Shape.ElementMap

# Assert - verify the names are correct

tc = unittest.TestCase()
tc.assertEqual('Edge1', em['Edge2;:M;FUS;:H1003:7,E;:H,E'])
tc.assertEqual('Edge2', em['Edge3;:M;FUS;:H1004:7,E;:H,E'])
```

For the python to work, we would need to be near the end of Phase 4.

But, in theory, we could run the fairly limited (history-less) equivalent of this test in CPP at Phase 2. And a history-enabled version of this test in Phase 3.

How, precisely, to write those tests is making my head hurt a bit. But it should be very possible.

Thoughts?

chennes, 2023-03-30:
I know how to create the objects in C++, but not how to do the fuse. Is that where you are stuck?

JohnAD, 2023-03-30:
I was thinking of just the simpler case, without the fuse yet.

Without FreeCAD running as a whole, how would a unit test create those objects in context. But, now that I think about it, I could simply trace back how the python code does things as a first step.

So ... creating a new Document() then calling `addObject` on it with a new DocumentObject(). Would that create enough context? I'll have to try some things tomorrow.

chennes, 2023-03-30:
Actually, it's fairly idiomatic in FreeCAD to actually just call the Python functions from the C++. That said, this will create your objects:
```
#include <App/Application.h>
#include <App/Document.h>
#include "Mod/Part/App/FeaturePartBox.h"

void makeObjects() {
    auto *activeDoc = App::GetApplication().getActiveDocument();
    auto *box = activeDoc->addObject("Part::Box", "box");
    auto *cylinder = activeDoc->addObject("Part::Cylinder", "cylinder");
}
```

chennes, 2023-04-03:
Regarding formatting: so far all toponaming code that we've merged is run through clang-format (including that which is embedded in another file which has *not* been clang-formatted). I think we should just press ahead with reformatting as the various WB maintainers are ready to do it in their own development cycle, and those of us merging toponaming code can just deal with it. It's not like we're using git to do the merge anyway, it's all done by hand right now.

JohnAD, 2023-06-20:
Card 2 (https://github.com/FreeCAD/FreeCAD/issues/9115) of phase 3 is done! Thanks @Pesc0 for all the great work!

Next are cards 3 and 4; which is where the actual name maps/arrays are added (but NOT given names yet; that is phase 3 and will might slow down FreeCAD somewhat at first.)

Card3 is about adding names to `ComplexGeoData` is the parent of object of `TopoShape` and possibly other classes.

Card4 is about adding names as needed in the Sketcher workbench. This *could* be done by having objects in Sketcher inherit ComplexGeoData, but I suspect that is overkill. @AjinkyaDahale has volunteered to help this card.

Is anyone wanting to do Card 3? I'm tempted to pick this one up myself, but I'm happy let someone else do it. Or, I'm happy to pair with someone and plan it out jointly.

AjinkyaDahale, 2023-06-20:
> Card 2 (#9115) of phase 2 is done! Thanks @Pesc0 for all the great work!

> Card4 is about adding names as needed in the Sketcher workbench. This _could_ be done by having objects in Sketcher inherit ComplexGeoData, but I suspect that is overkill. @AjinkyaDahale has volunteered to help this card.

I think something like this is already happening. Let's discuss this in the PR (#9696) itself.

> Is anyone wanting to do Card 3? I'm tempted to pick this one up myself, but I'm happy let someone else do it. Or, I'm happy to pair with someone and plan it out jointly.

I believe @chennes already started with this in #9780.

JohnAD, 2023-06-20:
Thanks @AjinkyaDahale ! I somehow missed that the new PRs were already created. :grin:

chennes, 2023-06-20:
@JohnAD if you'd like to work on card 3 feel free to -- I'm quite busy right now and could certainly use the assistance!

JohnAD, 2023-07-21:
An update to TN issue. Ajinkya is close to having completed the final task of Phase 2! (PR #9696)!

After that is merged, I'll take about a week to start refining phase 3's tasks. A lot was learned in phase 2. I'll then post the new list here for feedback and changes.

masterxq, 2023-08-03:
Hello sir, thanks sir, any update on that?
We are waiting with anticipation :)

JohnAD, 2023-08-07:
Update to the community about Phase 3. (Phase 2 is complete.) I'm still working on a rough draft of tasks for phase 3, but I've refrained from posting anything yet.

FreeCAD is having a hackathon next week in Vancouver. Details at https://blog.freecad.org/2023/05/24/freecad-hackathon-11-13-august-2023/ . I'll be there. And a few other coders with knowledge of FreeCAD's C++ system will be there also. This is a **great** opportunity to refine those tasks, an perhaps even start working on some of them a bit. I'll be showing that draft to them then. With our laptops out, and copies of both the master branch and RealThunders fork in front of us, and perhaps a whiteboard to jot stuff on, I suspect we could run through the possibilities and create a better set of task cards than I could just on my own. We will also post notes and questions here and in the Telegram group. So folks not in person can help also.

JohnAD, 2023-08-13:
Proposed cards for Phase 3:

# Phase 3: Integrate

> Add the naming, but do not use it yet. So, while the member values are set, they are not actually used. This stage will slow down FreeCAD and will make the naming visible on an element when selected.

> You will see some items marked with a question mark (?). These are where I'm not sure the changes are TN related and might be skipped or done later.

- **Add History Support** Naming is based on history. Add the RT changes to history to expand it's support.
  - Part Workbench
    - [ ] PropertyTopoShape (do before the next set of cards)
    - [ ] PartFeature (just the History parts like 'buildHistory' and 'joinHistory')
    - [ ] FeaturePartCommon
    - [ ] FeaturePartExtension?
    - [ ] FeatureChamfer
    - [ ] FeatureFillet
    - [ ] FeaturePartBoolean
    - [ ] FeaturePartFuse
    - [ ] FeatureExtrusion
    - [ ] FeatureRevolution
  - Part Design Workbench
    - [ ] ShapeBinder
    - [ ] FeatureDressUp (just internal identifier naming?)
    - [ ] FeatureSketchBased
    - [ ] FeatureTransformed (just internal identifier naming?)
  - Sketcher Workbench
    - [ ] SketchObject (see GeoHistory and updateGeoHistory)
- **Add Name Assignment** Based on the advanced history, generate the names and assign to Doc Objects
  - Part Workbench. This includes saving and restoring.
    > not all of these can be worked independently, lessons will be learned :smile:
    - [ ] TopoShape and TopoShapeEx (BIG card)
    - [ ] PartFeature
    - [ ] Attacher
    - [ ] AttachExtension
    - [ ] Geometry and GeometryExtension and GeometryMigrationExtension?
    - [ ] CrossSection?
    - [ ] FaceMaker
    - [ ] FeatureExtrusion
  - Part Design Workbench
    - [ ] ShapeBinder
  - Sketcher Workbench
    - [ ] SketchObject ('buildShape' change and related) (what is toggleFreeze used for?)
    - [ ] ExternalGeometryExtension, ExternalGeometryFacade, GeometryFacade (set/get for saving/restore)
    - [ ] SketchGeometryExtension (set/get for saving/restore)
- **Add Python Support** Add in the Py and PyImp changes for viewing the new names
  - Part Workbench
    - [ ] Command
    - [ ] AppPartPy
    - [ ] AttachEnginePyImp?
    - [ ] PartFeaturePy and PartFeaturePyImp
    - [ ] TopoShape * Py *
  - Part Design Workbench
    - [ ] Command
  - Sketcher Workbench
    - [ ] CommandAlterGeometry, CommandCreateGeo, CommandSketcherTools
    - [ ] SketchObjectPy, SketchObjectPyImp

JohnAD, 2023-08-13:
Please make comments and suggestions. I know more about the Part workbench than the Sketcher workbench; so all Sketcher help is extra appreciated :smile: .

JohnAD, 2023-08-24:
Over the course of the next 24 hours, I will be creating 32-or-so Issues named in the form of:

`[Issue] TopoNaming Phase 3 (card 1 of 32) - Add History Support/Part Workbench/PropertyTopoShape`

They will be linked into the opening comment/description at the top of this issue; and they will be linked to the  [TopoNaming project](https://github.com/orgs/FreeCAD/projects/2).

JohnAD, 2023-10-23:
After starting up the cards, it appears that the "TopoShape and TopoShapeEx (BIG card)" mentioned in Part 2 the list above must be done before the remaining parts of Part 1. So, I'm adding moving that card after card 1 and before card 2. Essentially:

AlexDaniel, 2023-12-12:
Any updates? Phase 3 was planned some time ago, but the work wasn't started.

chennes, 2023-12-12:
We are working on the TopoShape class -- it is massive (94 added methods in 10,000 lines of code). Right now we are waiting for some additional documentation about a caching mechanism from @realthunder.

furgo16, 2024-02-19:
Many thanks for the work everyone is doing towards solving this issue. For those of us who are following the progress, would it be possible for a maintainer to add the "Phase 3" issue list (in https://github.com/FreeCAD/FreeCAD/issues/8432#issuecomment-1676433915 or by the list of issue references to the main issue description?

This is just a suggestion to make it easier to potential contributors to see what's being done and one left from a SSoT (Single Source of Truth) position without having to navigate through all the comments.

If it makes it easier, here's the markdown to copy and paste the current list of issues

```
#### Phase 3

1. https://github.com/FreeCAD/FreeCAD/issues/10353
1. https://github.com/FreeCAD/FreeCAD/issues/10354
1. https://github.com/FreeCAD/FreeCAD/issues/10355
1. https://github.com/FreeCAD/FreeCAD/issues/10356
1. https://github.com/FreeCAD/FreeCAD/issues/10357
1. https://github.com/FreeCAD/FreeCAD/issues/10358
1. https://github.com/FreeCAD/FreeCAD/issues/10359
1. https://github.com/FreeCAD/FreeCAD/issues/10360
1. https://github.com/FreeCAD/FreeCAD/issues/10361
1. https://github.com/FreeCAD/FreeCAD/issues/10362
1. https://github.com/FreeCAD/FreeCAD/issues/10398
1. https://github.com/FreeCAD/FreeCAD/issues/10399
1. https://github.com/FreeCAD/FreeCAD/issues/10400
1. https://github.com/FreeCAD/FreeCAD/issues/10401
1. https://github.com/FreeCAD/FreeCAD/issues/10402
```

Thanks!

Pesc0, 2024-02-19:
this should be what you're looking for https://github.com/orgs/FreeCAD/projects/2, let me know if that helps

furgo16, 2024-02-19:
> this should be what you're looking for https://github.com/orgs/FreeCAD/projects/2, let me know if that helps

I'm aware of the GitHub project, which is also mentioned in the issue description. But thanks for pointing it out in any case!

AlexDaniel, 2024-04-13:
Wow, it looks like there has been a lot of progress lately! Keep it up! ❤️

closed 2024-05-20 by chennes

chennes, 2024-05-20:
We've just activated the TNP mitigation code in main: it's a lot of code and of course we expect there to be some bugs, as well as probably some edge cases that the team has missed. If you find them, please open new GitHub issues for each (after checking to see if someone else has not already done so, of course!). Thanks!

PR #13705 (2024-05-20): Toponaming: Enable FC_USE_TNP_FIX code
Enable the Topological Naming Problem mitigation code. This is an interim step: eventually the flag will be removed, along with the old code.

Fixes #8432.