FreeCAD/FreeCAD #8440: [Issue] Empty Qwidget above Qmenubar

MisterMakerNL, 2023-02-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31812 (Git)
Build type: Release
Branch: master
Hash: c074fe1c95e5aa4f21a11d1387c2ae38f5014a4d
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.11.4
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0
  * BIM.backup1668883108.234155 2021.12.0 (Disabled)
  * Dracula 0.0.2
  * ExtremeProDark 2.6.4
  * ExtremeProDark.backup1665346471.659064 2.3.1 (Disabled)
  * ExtremeProLight 1.4.3
  * ExtremeProLight.backup1665559197.7999926
  * FeedsAndSpeeds 0.5.0
  * FEM_FrontISTR 0.1.0
  * IconThemes
  * lattice2 1.0.0
  * Nodes 0.1.9
  * PieMenu 1.2.3
  * PieMenu.backup1664822562.2342157
  * ProDarkThemePreferencePack 1.0.0
  * scanObjects
  * SelectorToolbar
  * sheetmetal 0.2.59
  * SP_Startup
  * ToolbarStyle
[/code]
```

### Subproject(s) affected?

None

### Issue description

My ExtremeProDark and ProDark broke by something in 0.21 branch.
It is caused by a empty Qwidget above the menu.
Something I assume shouldn't be there.
I can only fix this in my theme bu removing the Qwidget definition and specify it for every QWidget separate, which clutters the theme file by a lot. (just like the not broken theme's do it)
You can see the empy floating Qwidget by adding this to the qss file:
```
QWidget {
    background-color: #1bb004;
}
```
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2023-02-11:
Whoops did some more research and it is caused by an addon.

chennes, 2023-02-11:
> Feel like this is then still an issue in Freecad, as it breaks the ui.

What sort of action could we take on our end? Addons can directly manipulate the user interface.

MisterMakerNL, 2023-02-12:
> 

I might need to test more if this Qwidget is created by an addon. But it's not there in 0.2 with same addons.
We should remove the Qwidget behind the menubar. If we need a Qwidget holder or some sorts it should be 1px or 0px.
This one is 200px on 4k and 100px on 1080p.

Syres916, 2023-02-12:
While investigating another regression that gets triggered by PieMenu I found this https://github.com/FreeCAD/FreeCAD/commit/7458912b3340a428cd5fcb763859659ebb8b76ce to be the culprit, hope it's of use.

chennes, 2023-02-13:
Ah, nice detective work @Syres916 -- maybe @realthunder or @wwmayer have an easy solution. It sounds like that widget is there on purpose, anyway, and that it should be "zero size".

realthunder, 2023-02-14:
Can some one please test [this](https://github.com/realthunder/FreeCAD/commit/850c3219849) patch on Windows? I only got this similar problem on MacOS (which I seldom use actually), and cannot reproduce on Windows. Maybe we also need to explicitly set a zero size dimension for some reason.

luzpaz, 2023-02-27:
@MisterMakerNL by chance can you build and test FC with Syres916 port of realthunders patch in #8608 ?

MisterMakerNL, 2023-02-28:
> 

Would like, but haven't set the build process up.

luzpaz, 2023-03-14:
Should be fixed by #8608

closed 2023-03-14 by luzpaz