FreeCAD/FreeCAD #8451: TechDraw panning in the Page with Shift-LMB stopped working

g3grau, 2023-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
Worked as expected in FreeCAD_weekly-builds-0-2022-12-15-conda-Linux-x86_64-py310.AppImage, doesn't work in FreeCAD_weekly-builds-31812-2023-02-09-conda-Linux-x86_64-py310.AppImage

[code]
OS: openSUSE Tumbleweed (XFCE//usr/share/xsessions/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31812 (Git) AppImage
Build type: Release
Branch: master
Hash: c074fe1c95e5aa4f21a11d1387c2ae38f5014a4d
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Silk 1.0.0
  * sheetmetal 0.2.61
  * kicadStepUpMod 10.17.1
  * CfdOF 1.21.4
  * Assembly4 0.12.5
  * fasteners 0.4.54
  * offline-documentation 1.0.0-alpha
  * MeshRemodel 1.8919.0
  * LCInterlocking
  * 3D_Printing_Tools
  * A2plus 0.4.60i
  * 3DfindIT 1.2.0
  * EM
  * ExplodedAssembly
  * ThreadProfile 1.84.0
  * workfeature
  * Lithophane
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * Defeaturing 1.2.0
  * Assembly3 0.11.4
  * freecad.gears 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.6
  * frame 0.1.0
  * FEM_FrontISTR 0.1.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * OpticsWorkbench 1.0.9
  * CommandPanel
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

I use the "Touchpad"  mouse style which gives me panning for Shift-LMB and rotating for Alt-LMB (single finger drag on a touchpad). This still works in the CAD drawing, but not in the Page. Using scrollbars for panning requires a lot of extra mouse movements. If that change was on purpose, please change it back :)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-02-12:
I use Touchpad navigation almost exclusively.  Shift + LMB is not the panning combination.  Shift + mouse movement does panning.  Shift + LMB + mouse movement is Rotate View, which does not make sense in TechDraw.   https://wiki.freecad.org/Mouse_navigation

There have not been any recent changes to this are of the code.

g3grau, 2023-02-12:
Sorry to cause confusion! Now, having a mouse next to the touchpad, I notice the difference :)
The error is still there, but as you pointed out, it's Shift+mouse movement that stopped working in the AppImage from Feb. 9th. In fact it neither works with the touchpad nor with the mouse, while both works in the AppImage from Dec. 15 (and older versions). It is only broken in the Page view (TechDraw), for the CAD drawing it still works.
I think there are no other major differences in this time (Qt, Python, ..)?
I just wondered if this may be also a side effect of the strange keyboard issue (8453/8417)? However, the only Shift-related hotkey I found still works (Ctrl-Tab and Ctrl-Shift-Tab), so I guess that is not related. Do you also use the AppImage version?

WandererFan, 2023-02-13:
I don't use the Appimage on a regular basis, but I did try this one, and panning with Shift + mouse move works.  TechDraw uses a different mechanism for pan & zoom compared to the 3d window.

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31850 (Git) AppImage
Build type: Release
Branch: master
Hash: d8d3bb8120ad139cf1eb2acf0be0ae6c297ac7f3
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * sheetmetal 0.2.59
  * FreeCAD_Electric
  * Assembly4 0.12.4
  * A2plus 0.4.56a
[/code]

luzpaz, 2023-04-11:
@g3grau can you still reproduce ?

g3grau, 2023-04-13:
Hi @luzpaz,

Thanks for the heads up. I just installed the Appimage 
[code]
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32821 (Git) AppImage
Build type: Release
Branch: master
Hash: eaf9d43f37238c33dd1c47b859a6fdf5b7f629fc
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Silk 1.0.0
  * sheetmetal 0.2.61
  * kicadStepUpMod 10.17.1
  * CfdOF 1.21.6
  * Assembly4 0.12.6
  * fasteners 0.4.54
  * offline-documentation 1.0.0-alpha
  * MeshRemodel 1.8919.0
  * LCInterlocking
  * 3D_Printing_Tools
  * A2plus 0.4.60k
  * 3DfindIT 1.2.0
  * EM
  * ExplodedAssembly
  * ThreadProfile 1.84.0
  * workfeature
  * Lithophane
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * Defeaturing 1.2.0
  * Assembly3 0.11.4
  * InventorLoader 1.3.0
  * FEMbyGEN 1.2.0
  * freecad.gears 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * frame 0.1.0
  * FEM_FrontISTR 0.1.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * pyrate
  * OpticsWorkbench 1.0.9
  * CommandPanel
[/code]
and made a short test .. I can rename items and edit numbers! Problem solved :)

thanks & best regards
  GG

luzpaz, 2023-04-15:
Closing per @g3grau feedback

closed 2023-04-15 by luzpaz