FreeCAD/FreeCAD #8453: cannot rename element in Combo View, drawing area capturing the keys

g3grau, 2023-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
No problem in FreeCAD_weekly-builds-0-2022-12-15-conda-Linux-x86_64-py310.AppImage, bug in FreeCAD_weekly-builds-31812-2023-02-09-conda-Linux-x86_64-py310.AppImage

[code]
OS: openSUSE Tumbleweed (XFCE//usr/share/xsessions/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31812 (Git) AppImage
Build type: Release
Branch: master
Hash: c074fe1c95e5aa4f21a11d1387c2ae38f5014a4d
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Silk 1.0.0
  * sheetmetal 0.2.61
  * kicadStepUpMod 10.17.1
  * CfdOF 1.21.4
  * Assembly4 0.12.5
  * fasteners 0.4.54
  * offline-documentation 1.0.0-alpha
  * MeshRemodel 1.8919.0
  * LCInterlocking
  * 3D_Printing_Tools
  * A2plus 0.4.60i
  * 3DfindIT 1.2.0
  * EM
  * ExplodedAssembly
  * ThreadProfile 1.84.0
  * workfeature
  * Lithophane
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * Defeaturing 1.2.0
  * Assembly3 0.11.4
  * freecad.gears 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.6
  * frame 0.1.0
  * FEM_FrontISTR 0.1.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * OpticsWorkbench 1.0.9
  * CommandPanel
[/code]
```

### Subproject(s) affected?

None

### Issue description

Trying to rename an element in the Combo View, the text is not edited but the CAD drawing reacts to the key strokes, e.g. backspace works, but delete will not delete the character behind the cursor but the object (and it cannot be canceled, need to ^Z. Entering numbers will switch the view etc.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-02-12:
might be a duplicate of #8417

g3grau, 2023-02-12:
Indeed, the issue looks very similar, haven't seen that.
Not sure if this is due to "global" hotkeys or not, at least I couldn't see "Delete" in the menu (where can I find a list of hotkeys?). 
There is a difference between editing names in the model tree (e.g. 0..6 don't work) vs. editing names in a TechDraw tree (works). According to the hotkeys listed in the menu, 0..6 are free in the TechDraw, but Space is not. Indeed, I can rename a View with _ and 1..6, but I cannot add a space in the name because it will hide the view which I want to rename. So yes, the 8417 seems to be right about hotkeys!

g3grau, 2023-02-12:
I just wanted to find out if I can write a comment in the python console to paste it to the name field. Nope...those keys are also captured before they enter the console :-D
What is a possible workaround: enter your new text elsewhere outside FC, e.g. in this window. Say part_12, copy it with ^C and paste it with ^V into the name field of the part ... that works!

adrianinsaval, 2023-02-28:
@g3grau are you still experiencing this? I cannot reproduce anymore

g3grau, 2023-02-28:
I'm afraid the bug is still there :(
I just tried FreeCAD_weekly-builds-31917-2023-02-26-conda-Linux-x86_64-py310.AppImage and immediately found this issue while editing the diameter of a sphere in the Part Design WB. Entering 1 or 2 changes the viewpoint instead of adding it to the number. I can copy and paste an old "1" in the same edit field to get a second one.
Similar, trying to rename an object, a and b don't work, c and d do. Same as before, sorry...

g3grau, 2023-02-28:
That's weird! I managed to install and start (that was the complicated part) the flatpack version FreeCAD:
```
Name                            Application ID                                       Version             Branch            Origin             Installation
org.freecadweb.FreeCAD                               0.21.0.31979        beta              flathub-beta       user

```
The bug is apparently not there! But unfortunately the flatpak doesn't care about my settings and somehow looks very different, maybe like KDE vs. Gnome style?! Left side is AppImage with the issue, right side is Flatpak w/o this issue. I rearranged the Part Design WB and put them side by side:

[image]

I think others reported similar discrepancies between the different packages for the same version? 
I can't really imagine how, both AppImage and Flatpak are kind of containers and should have been built 
with the same stuff inside? At least a major difference exists in the GUI components which are used.
Can I somehow list all the "ingredients" of these two build variants?

If I now had a clue where this flatpak thing has been installed to, and how to change this location. Google help!

adrianinsaval, 2023-02-28:
This is a clear duplicate of #8417 so I think it should be closed in favor of that one. You may have missed but we have already identified that this is an appimage specific issue. No other package is presenting this. The flatpak is indeed completely independent from the appimage version, it is compiled with a different set of libraries. The different look is to be expected, the flatpak version has better integration with system themes while the appimage version only includes the ugly "Fusion" style provided by qt.

chennes, 2023-02-28:
I'm going to close this as a duplicate of #8346 -- the problem appears to be with the specific version of Qt used in the feedstock. There is a PR in to update to Qt5.15.8, which should resolve the issue.

closed 2023-02-28 by chennes

g3grau, 2023-03-02:
ok, thanks for this info! I hope the AppImage will soon work again.
I don't really like the concept of the flatpak (or snap) and the flatpak version consumes way more screen area for the same GUI items.