FreeCAD/FreeCAD #8454: TechDraw observations

g3grau, 2023-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
FreeCAD_weekly-builds-0-2022-12-15-conda-Linux-x86_64-py310.AppImage and FreeCAD_weekly-builds-31812-2023-02-09-conda-Linux-x86_64-py310.AppImage

[code]
OS: openSUSE Tumbleweed (XFCE//usr/share/xsessions/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31812 (Git) AppImage
Build type: Release
Branch: master
Hash: c074fe1c95e5aa4f21a11d1387c2ae38f5014a4d
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Silk 1.0.0
  * sheetmetal 0.2.61
  * kicadStepUpMod 10.17.1
  * CfdOF 1.21.4
  * Assembly4 0.12.5
  * fasteners 0.4.54
  * offline-documentation 1.0.0-alpha
  * MeshRemodel 1.8919.0
  * LCInterlocking
  * 3D_Printing_Tools
  * A2plus 0.4.60i
  * 3DfindIT 1.2.0
  * EM
  * ExplodedAssembly
  * ThreadProfile 1.84.0
  * workfeature
  * Lithophane
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * Defeaturing 1.2.0
  * Assembly3 0.11.4
  * freecad.gears 1.0.0
  * CurvedShapes 1.0.4
  * Curves 0.6.6
  * frame 0.1.0
  * FEM_FrontISTR 0.1.0
  * fcgear 1.0.0
  * Pyramids-and-Polyhedrons
  * OpticsWorkbench 1.0.9
  * CommandPanel
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

My original problem was that TechDraw was frequently crashing FC. However, it appears that this has been a consequence of a bad model with circular reference. After fixing that, I did not see another crash. 

However, having cirular and half-sphere features, cross-sections, and a number of measurements, I find two specific types of messages in the log:
a) Error: 
"asCircle - OCC error - gp_Dir() - input vector has zero norm - while approx spline as circle
Geom::baseFactory - unknown error occurred while making spline"

b) Warning: 
"QXcbConnection: XCB error: 3 (BadWindow), sequence: 36619, resource id: 6014731, major code: 40 (TranslateCoords), minor code: 0"

I don't see a malfunction. Any hint what this means? I can't attach this particular model. In a test case I couldn't repeat it yet, will try again..

Trying to repeat the issue with the original bad model (circular reference), I can reproduce a crash when first deleting one measurement on the Page and then one of the views from a Projection Group:

First comes a bunch of error messages (probably induced by the cyclic reference):

```
...
Err: asCircle - OCC error - gp_Dir() - input vector has zero norm - while approx spline as circle
Err: Geom::baseFactory - unknown error occurred while making spline
Err: 133.993 <App> Document.cpp(3201): The graph must be a DAG.
Wrn: Part::Part2DObjectPython: Link(s) to object(s) 'Body004' go out of the allowed scope 'Placment'. Instead, the l
inked object(s) reside within 'N/A'.
Wrn: PartDesign::Body: Link(s) to object(s) 'Fusion' go out of the allowed scope 'Body005'. Instead, the linked obje
ct(s) reside within 'Part002'.
Wrn: Part::Part2DObjectPython: Link(s) to object(s) 'Body004' go out of the allowed scope 'Placment'. Instead, the l
inked object(s) reside within 'N/A'.
Wrn: PartDesign::Body: Link(s) to object(s) 'Fusion' go out of the allowed scope 'Body005'. Instead, the linked obje
ct(s) reside within 'Part002'.
Err: 142.008 <App> Document.cpp(3556): Multiprobenhalter4b#Placment still touched after recompute
Err: asCircle - OCC error - gp_Dir() - input vector has zero norm - while approx spline as circle
Err: Geom::baseFactory - unknown error occurred while making spline
...
```

Then the final message before the crash (not included in the log file):

```
corrupted double-linked list
/tmp/.mount_FreeCADqKWCn/AppRun: line 43: 12129 Aborted                 (core dumped) ${MAIN} "$@"

```

The "bad" model looks ok, the drawings look ok, the measurements look ok. The cyclic reference seems to stem from two Parts which were somehow sharing a ShapeBinder. After merging these Parts, the cycle seems to be gone and the crashes also.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-02-12:
a) Error:
"asCircle - OCC error - gp_Dir() - input vector has zero norm - while approx spline as circle
Geom::baseFactory - unknown error occurred while making spline"

This one happens when trying to treat a BSpline as a circle when making a dimension.  There would (should?) have been a message saying that the edge was not a circle and that the dimension would be approximate.    "input vector has zero norm" comes from OCC, and I can't tell you more without a sample.

WandererFan, 2023-02-12:
b) Warning:
"QXcbConnection: XCB error: 3 (BadWindow), sequence: 36619, resource id: 6014731, major code: 40 (TranslateCoords), minor code: 0"

I see this one sometimes, and I have no idea what is means.  AFAICT it is not specific to TechDraw.

WandererFan, 2023-02-12:
Trying to repeat the issue with the original bad model (circular reference), I can reproduce a crash when first deleting one measurement on the Page and then one of the views from a Projection Group:

I'd need a sample file to figure this one out.

g3grau, 2023-02-12:
Thanks for these hints. I was not aware of any splines in the model. But indeed there are some, resulting from fillets on a tapered pad. The model been created with a previous build. In the current I indeed get an error and a dialog box if I try to measure the radius of the fillet. Maybe an existing measurement can break if the fillet is added at a later time? I'll try to figure it out

WandererFan, 2023-02-13:
Maybe an existing measurement can break if the fillet is added at a later time?

Yes this is the little brother of the big Topological Naming Problem.   If the dimension is linked to Edge5 as a circle, and changes to the model make Edge5 now a BSpline, that could be the issue.

WandererFan, 2023-03-15:
I think a) is resolved by #8803.

the little TNP is #8878.

luzpaz, 2023-03-17:
Thanks WF. Closing this ticket then.

closed 2023-03-17 by luzpaz