FreeCAD/FreeCAD #8476: [Issue] Incorrect export of hollow part to step

xypron, 2023-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31862 (Git) Snap 603
Build type: Release
Branch: master
Hash: 4382bd199a060fe4c8c284b80fe3ab91ece299f7
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Issue description

I add this OpenSCAD part:

```
difference() {
	cylinder(r=5, h =30,center = true);
	translate([0,0, 1])
		cylinder(r=4, h= 30,center=true);
}
translate([0,0, -17])
	cylinder(r=1.6, h =4,center = true);
translate([0,0, -20])
	cylinder(r2=1.6, r1=1, h =2,center = true);
```

The part is shown correctly with a hollow cylinder.

I export as step file with default settings.

I import the step file.

The part is not hollow anymore. Instead of a difference of cylinders I get the union of the cylinders.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xypron, 2023-02-13:
[huelse.tar.gz](https://github.com/FreeCAD/FreeCAD/files/10726413/huelse.tar.gz)

luzpaz, 2023-03-07:
@KeithSloan would you mind trying to reproduce and weigh-in?

KeithSloan, 2023-03-07:
Thought I was able to reproduce the problem when I initially tried with Add Elements but then I tried importing as a scad file and now when I try I am not seeing the problem. Looking at the first step file test.step the problem is there, so whatever I did, the problem is there, after that all the other step files 2-8 I am not seeing the problem.

Maybe a problem with selection?
Could @xypron try with importing a scad file, Alternate Importer, all there import options.
Maybe experiment with selection order?

Tonight is Winchester Photography Society meeting, so I will leave now and look again tomorrow.

[Archive.zip](https://github.com/FreeCAD/FreeCAD/files/10913141/Archive.zip)
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.46.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * toSketch 1.0.1
  * Curves 0.6.7
[/code]

Also with Alternative importer not seeing the problem
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.46.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Curves 0.6.7
[/code]

grubertechnic, 2023-06-30:
Works also flawless here (step export and import).

[image]

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

@luzpaz or @adrianinsaval : candidate for closing?

adrianinsaval, 2023-06-30:
Sounds like this was just user error, selecting but the object used for the cut and the cut itself is expected to behave like that. When exporting only the result should be selected.

closed 2023-06-30 by adrianinsaval