FreeCAD/FreeCAD #8477: Clipping Plane Dialog Box

DrawQuick, 2023-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31790 (Git)
Build type: Release
Branch: master
Hash: 23e240e017e6712e027d39d0ee764fd7a963e718
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * BIM 2021.12.0
  * fasteners 0.4.54
  * frame 0.1.0
  * parts_library
  * ThreadProfile 1.84.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

I work with Freecad on a 14" laptop and whenever I call up the View Clipping Plane dialog box it pushes my property windows off the screen. This dialog box appears to be rather large for the info contained within it. Can this dialog box be reduced to half its size, narrow and provide a preferences option to open in the centre of the screen so that the user can decide where to locate it and once placed to persist when opening Freecad. I've been using Freecad for some time and everyone associated with the development of this project have done an amazing job.
Thank you

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-02-23:
You can place the panel wherever you want by dragging from it's title, you can detach from the dock, dock to the other side, have it tabbed with the other panels, etc. You can also resize the panel.

I suggest reformulating the issue as "Clipping plane dialog gives bad UX on smaller screens, takes too much space", that is something that we may act upon, it's position is already your choice.

DrawQuick, 2023-02-23:
Hi Adrian,
The problem with the dialog box is, it has a minimum resize because the
controls are spaced quite wide. If you allow the dialog box to have scroll
bars(similar to the Combo View), then it would fit better in another dock
position. As you can see in the images I posted, when the clipping plane is
launched it pushes my other docked dialogs off the screen. I have to move
the clipping dialog somewhere else, minimize the main window, resize it and
then the other docked dialogs come back, then maximize the main window. I
think if you can make the clipping dialog narrower by compressing the
controls together it would dock better in other locations.
Thanks for your help.

MiqCG, 2024-06-21:
Hi,
I can confirm this behaviour with my 15'' screen laptop and 0.22 rev 37819. When Clipping plane is called and it get to the bottom left of the side panel, it pushes the FreeCAD window out and beyond its maximized size. Note: I have undecorated windows enable.

When 2 screens are connected, you can see FreeCAD "popping out" the small one (at the bottom). See video attached.

On the other hand I think it's not obvious that the Clipping panel can be detached from the side zone. If you don't know that, you will not be able to access the "Close" button on the bottom.

@DrawQuick , do you mind editing the title and changing to "Clipping plane dialog gives bad UX on smaller screens, takes too much space" as @adrianinsaval suggested?

`
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
Build type: Release
Branch: main
Hash: 2ca9b6ef8036d5fd808f01205b66eb0424059e21
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.18
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.24

`

https://github.com/FreeCAD/FreeCAD/assets/18265355/e42f1387-2e8a-4cec-83e3-f8903074fc34

MiqCG, 2024-06-22:
> When 2 screens are connected, you can see FreeCAD "popping out" the small one (at the bottom).

In the last video, when Clipping plane is called, you can see that the inferior bar or zone of FreeCAD "disappears" too .