FreeCAD/FreeCAD #8486: I can't seem to edit values in profile operation.

LibertyWarriorMusic, 2023-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

None

### Issue description

I am new to free cad, it seems a little buggy at times. I have created a profile, but the program doesn't seem to allow me to edit some values. I am trying to change the Final Depth, I hight light the value and trying to type numbers and it doesn't work. This seems like some that should be really easy to do.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LibertyWarriorMusic, 2023-02-14:
If I like on the formula editor, I had to remove the auto generated formula text and replace it with a value. There has to be a better way to do this, it is very confusing. Why does it default to using a formula editor? That should be an option, not the default. If I click on a value in property, I expect to be able to change that value, not stuff around with a formula editor.

luzpaz, 2023-02-14:
Please open a [Path subforum](https://forum.freecad.org/viewforum.php?f=15) thread to get assistance.

closed 2023-02-14 by luzpaz