FreeCAD/FreeCAD #8494: [Bug] Two unequal placements reported equal.

wwmayer, 2023-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=75448

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.1 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31906 (Git)
Build type: debug
Branch: master
Hash: 99a7262e60f1b5d79cf169b95be764b045499cdd
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

Core

### Issue description

1. Load movecube.FCStd attached to the forum posting
2. Select (bottom) cube under Part001
3. Run this code snippet 
```
sel = Gui.Selection.getSelectionEx('',0)[0]
path = sel.SubElementNames[0]
shp = sel.Object.getSubObject(path)
gpl1 = sel.Object.getSubObject(path, retType = 3)
gpl = App.Placement()
for obj in sel.Object.getSubObjectList(path):
    gpl = gpl*obj.Placement

print(f'gpl1 = {gpl1}')
print(f'gpl = {gpl}')
print(f'Same? {gpl.isSame(gpl1, 1E-7)}')
print(gpl.Rotation.isSame(gpl1.Rotation, 1e-7))
print(gpl.Base.isEqual(gpl1.Base, 1e-7))
```
4. Output is
```
>>> print(f'gpl1 = {gpl1}')
gpl1 = Placement [Pos=(0,0,20), Yaw-Pitch-Roll=(106.26,0,0)]
>>> print(f'gpl = {gpl}')
gpl = Placement [Pos=(0,0,20), Yaw-Pitch-Roll=(90,0,0)]
>>> print(f'Same? {gpl.isSame(gpl1, 1E-7)}')
Same? True
>>> print(gpl.Rotation.isSame(gpl1.Rotation, 1e-7))
True
>>> print(gpl.Base.isEqual(gpl1.Base, 1e-7))
True
```
6. gpl and gpl1 shouldn't be considered equal

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jolbas, 2023-02-15:
Here is a minimal snippet including creation of objects. To run in a new document. Also showing what happens if gpl1 is copied back into a new object. The property view shows 60° bud the 3D view implies 106.26°
```
doc = App.ActiveDocument
Cube = doc.addObject('Part::Box', 'Cube')
CubeLink = doc.addObject('App::Link', 'CubeLink')
CubeLink.LinkedObject = Cube
CubeLink.Scale = 2
CubeLink.Placement.Base.z = 20
CubeLink.Placement.Rotation.Yaw = 90
gpl1 = CubeLink.getSubObject('', retType = 5)
gpl = CubeLink.Placement
CubeLink2 = doc.addObject('App::Link', 'CubeLink2')
CubeLink2.LinkedObject = Cube
CubeLink2.Scale = 2
CubeLink2.Placement = gpl1
doc.recompute()

print(f'gpl1 = {gpl1}')
print(f'gpl = {gpl}')
print(f'gpl2 = {CubeLink2.Placement}')
print(f'Same? {gpl.isSame(gpl1, 1E-7)}')
print(gpl.Rotation.isSame(gpl1.Rotation, 1e-7))
print(gpl.Base.isEqual(gpl1.Base, 1e-7))
```
I agree that those placements are not equal. But i would say that they should be.

Jolbas, 2023-02-15:
Could it be that `quat[0]` etc. is not prefixed with `this->`?

https://github.com/FreeCAD/FreeCAD/blob/master/src/Base/Rotation.cpp#L757

Jolbas, 2023-02-15:
It is the `isSame()` method linked to in previous comment that cant handle non normalized quaternions. The algorithm first mentioned in the comment works. But the simplification that then is used presumes a normalized quaternion.  Here is a very simple python code snippet showing the problem.

```
matrix = App.Placement().Matrix
matrix.scale(0.5)
rot = App.Rotation(matrix)
rot.isSame(rot, 1e-7)
# False!!
```

I think the main problem is that `void Rotation::setValue(const Matrix4D & m)` does not handle scaled and skew matrices. It wouldn't be very difficult to support uniform scale at least. Unfortunately I not very familiar with C++ so here is more of an idea of how it could be fixed:

https://github.com/FreeCAD/FreeCAD/blob/ffd9ca6e69fc7fa8c3d631176760aebdc7f9d73e/src/Base/Rotation.cpp#L221

```
void Rotation::setValue(const Matrix4D & m)
{
   // get column vectors
   ScaleType scale = dMtrx4D.hasScale(double tol=0.0)
   if (scale != NoScaling and scale != Uniform) {
        this->quat[0] = 0.0;
        this->quat[1] = 0.0;
        this->quat[2] = 0.0;
        this->quat[3] = 1.0;
        _axis.Set(0.0, 0.0, 1.0);
        _angle = 0.0;
        return;
    }
    double dx = copysign(m.getCol(0).Length(), m[0][0]);
    double dy = copysign(m.getCol(0).Length(), m[1][1]);
    double dz = copysign(m.getCol(0).Length(), m[2][2]);
    // don't know how to copy or if it's necessary before modifying
    m = m.copy();
    m.scale(1/dx, 1/dy, 1/dz);
    double trace = (m[0][0] + m[1][1] + m[2][2]);
    if (trace > 0.0) {
        double s = sqrt(1.0+trace);
        this->quat[3] = 0.5 * s;
        s = 0.5 / s;
        this->quat[0] = ((m[2][1] - m[1][2]) * s);
        this->quat[1] = ((m[0][2] - m[2][0]) * s);
        this->quat[2] = ((m[1][0] - m[0][1]) * s);
    }
    else {
        // Described in RotationIssues.pdf from <http://www.geometrictools.com>
        //
        // Get the max. element of the trace
        unsigned short i = 0;
        if (m[1][1] > m[0][0]) i = 1;
        if (m[2][2] > m[i][i]) i = 2;

        unsigned short j = (i+1)%3;
        unsigned short k = (i+2)%3;

        double s = sqrt((m[i][i] - (m[j][j] + m[k][k])) + 1.0);
        this->quat[i] = s * 0.5;
        s = 0.5 / s;
        this->quat[3] = ((m[k][j] - m[j][k]) * s);
        this->quat[j] = ((m[j][i] + m[i][j]) * s);
        this->quat[k] = ((m[k][i] + m[i][k]) * s);
    }

    this->evaluateVector();
}
```

wwmayer, 2023-02-16:
> Could it be that quat[0] etc. is not prefixed with this->?

No, this doesn't matter.

Jolbas, 2023-02-17:
Made a [PR](/FreeCAD/FreeCAD/pull/8522)

PR #8522 (2023-02-25): Fix issue #8494
When created a Rotation from a scaled matrix, the quaternion become malicious. This filters out the scale from the matrix if there is one. Fixes #8494
closed 2023-02-25 by chennes