FreeCAD/FreeCAD #8501: [Issue] FreeCAD crashes when constraining a sketch with an expression containing '.' (dot)

rrrossi, 2023-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31914 (Git)
Build type: Debug
Branch: master
Hash: e566daa1af241b5944d22a2f638e79bf9c5e3c45
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Issue description

Start FreeCAD
Create new
select sketcher workbench
create a new sketch, XY-Plane
create a circle centered in the Origin
constrain diameter
select function or type =
try to type 20 * 25.4 mm
When the '.' is typed, FreeCAD crashes.

ExpressionCompleter.cpp line 782,
char last = path[path.size() - 1];
I think that path.size() is 0

Problem did NOT occur for
[code]
OS: Manjaro Linux (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31594 (Git)
Build type: Debug
Branch: (HEAD detached at a9ed294765)
Hash: a9ed294765136438f26513d9ea01e046e502fbfa
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rrrossi, 2023-02-17:
It is possible I found the bug.

Here a possible patch:

`diff --git a/src/Gui/ExpressionCompleter.cpp b/src/Gui/ExpressionCompleter.cpp
index 3a1bb3492b..8dac1a8eaf 100644
--- a/src/Gui/ExpressionCompleter.cpp
+++ b/src/Gui/ExpressionCompleter.cpp
@@ -774,7 +774,7 @@ QStringList ExpressionCompleter::splitPath(const QString& input) const
             }
             else if (retry == 1) {
                 // restore path from retry 0
-                if (lastElem.size() > 1) {
+                if (lastElem.size() >= 1) {
                     path = path + lastElem;
                     lastElem = "";
                 }
`
It would be nice if somebody can check if it is correct.

luzpaz, 2023-02-17:
@0penBrain can you weigh-in ?

0penBrain, 2023-02-23:
Sorry. I thought I reported back on this a week ago. I was unable to reproduce at this time.
A good thing may be to test again after #8565 is merged.

0penBrain, 2023-02-23:
#8565 is merged. @rrrossi may you please try to reproduce when next weekly is available (on Sunday IIRC) ? Thx

rrrossi, 2023-02-23:
Before reporting the bug, I compiled from sources.
I pulled the last version (with your fix for #8590), and recompiled.
The error is still present.

My analysis: 
When I type the dot in the expression, spitPath() is called with as argument a QString with just a dot.
If this is correct (and that I don't know), the first cycle fails, and retry is incremented. 
In the second attempt, we enter in if (retry == 1)
with lastElem="." and path = ""
the condition (lastElem.size() > 1) is false,
hence char last = path[path.size() - 1];
is last = path[0-1] --> abort.
If the condition  had been >=, then path is ".", and the error not occurs. 

I did a test, and everything seems to work, but I can't exclude there are side effects.

0penBrain, 2023-02-23:
My problem is that my build absolutely not reproduce the issue ...

0penBrain, 2023-02-23:
OK, with debugger I can see the problem :)
I think your patch contradicts what is in the comment just after:
https://github.com/FreeCAD/FreeCAD/blob/5bf522ebcfe86684b592756adaf8b7cdd9291e5e/src/Gui/ExpressionCompleter.cpp#L781

So if I had to suggest a patch, I'd rather change the block as this:
```
else if (retry == 1) {
                // restore path from retry 0
                if (lastElem.size() > 1) {
                    path = path + lastElem;
                    lastElem = "";
                }
                // else... we don't reset lastElem if it's a '.' or '#' to allow chaining completions
               if (!path.isEmpty()) {
                    char last = path[path.size() - 1];
                    if (last != '#' && last != '.' && path.find('#') != std::string::npos) {
                        path += "._self";
                        ++retry;
                        continue;
                    }
                }
            }
```
Notice the same problem exists in the `retry==2` block just after.

I see this code has been introduced by @realthunder. May you help ?

rrrossi, 2023-02-23:
Probably you are right.
I implemented your proposal (except that path is std::string, and isEmpty() is named empty()), both for retry = 1and 2,
and FreeCAD does not crash anymore at the point. Expression is computed correctly. A number terminating with a dot is considered illegal, but I think this is perfectly acceptable (also 1. mm is illegal).
I don't know which additional tests I could make.

rrrossi, 2023-02-23:
What is a PR?

0penBrain, 2023-02-23:
Github Pull Request :)

rrrossi, 2023-02-23:
Never did before....

0penBrain, 2023-02-23:
> Never did before....

OK. Your first will probably some efforts. Up to you. If you then want to contribute the code more frequently, it may be a good investment to setup everything. If you think it's a one-time contributing, I (or someone else) will do the PR.
If you want to go for it, I know this pretty comprehensive guide (in French, but automatic translator should do a good job on this): https://blog.zenika.com/2017/01/24/pull-request-demystifie/

wwmayer, 2023-02-26:
@0penBrain 

> My problem is that my build absolutely not reproduce the issue ...

Were you able to reproduce the crash? I failed.

@rrrossi 
It looks like you have a self-compiled FreeCAD version. Can you add a line to splitPath() that prints the argument `input`? 
As far as I can see this function only uses this class member `currentObj`. So, printing the result of `currentObj.getPropertyPython()` would be helpful, too.

Then it might be easier to find the reason.

0penBrain, 2023-02-26:
> Were you able to reproduce the crash? I failed.

No I wasn't. But using debugger I can see that indeed 'path' is empty, so accessing 'path[path.size()-1]' is clearly illegal.
But it doesn't create a crash on my build too.

rrrossi, 2023-02-26:
@wwmayer 
when I type numbers, input is empty ("").
When I type the dot, input is just the dot (".").
currentObj.getPropertyPython() is FreeeCad.getDocument('Unnamed').getObject('Sketch')

It looks like Manjaro linux delivers a stdlib with a high "FORTIFY" level.   
FreeCAD aborts on an asset in include/bits/basic_string.h

By the way, how do you make a printout in FreeCAD? I used the Qtcreator debugger to get the information.
std::cout and qDebug() did not produce any output (either in the terminal, in QtCreator Application output, FreeCAD report view).
I saw the FC_TRACE macro, but I don't know how to activate it!

wwmayer, 2023-02-26:
Inside the function you can do a printf.
`
printf("%s\n", input.toUtf8().constData());
printf("%s\n", currentObj.getPropertyPython().c_str());
`
When starting from the console you will see them there.

Thorhian, 2023-03-08:
I've been trying to work with the latest commits, including to do some work on the Path workbench. This bug makes FreeCAD compiled on Arch Linux almost completely unusable, especially the path workbench to define tool sizes and various depth and height parameters. What can I do to help out?

rrrossi, 2023-03-08:
@openbrain
This makes the issue more urgent.
This is the patch to apply.
[issue8501.patch.txt](https://github.com/FreeCAD/FreeCAD/files/10917786/issue8501.patch.txt)
Last Sunday I was scared from the too many MUST, SHALL, MUST NOT in the PR guidelines of FreeCAD, and I did not proceed.
May you do the PR?

In the commit long description, you may write that the bug emerges only when the stdlib is compiled in debug or fortify mode. It seems this is the case of actual arch Linux and derived (my Manjaro).

@Thorhian 
since the expression evaluator is not completely broken, a simple workaround until the fix is not implemented, is to avoid the dot,

Expression like 10 * 2.54 must be translated in 10 *254/100. 
Ugly, but I hope it helps!

0penBrain, 2023-03-08:
> Last Sunday I was scared from the too many MUST, SHALL, MUST NOT in the PR guidelines of FreeCAD, and I did not proceed.
You shouldn't have been scared. We are pretty tolerant and guiding. :)

> May you do the PR?
Yes I can. @wwmayer are you OK for a hotfix PR for now, then keep trace of this to conduct further analysis ?

wwmayer, 2023-03-08:
Although I still don't get a crash I can confirm that the line `char last = path[path.size() - 1];` is wrong if path is empty.
The input of the function splitPath() is "." and it does two iteration steps in the while loop. After the first step path becomes an empty string and is supposed to crash in the second step.

However, the patch doesn't fully fix the problem because there is still the line `path.resize(path.size() - 6);` which results in allocating a huge amount of memory. In case path.size() is less than 6 the expression causes an underflow and produces a very high number that is passed to resize(). Depending on the system this may freeze your machine or if you are lucky throws an exception.

0penBrain, 2023-03-08:
> However, the patch doesn't fully fix the problem because there is still the line `path.resize(path.size() - 6);`

Correct. I guess this is originally present to remove the '._self' previously added, but because of the patch it may not be actually added.

wwmayer, 2023-03-08:
https://github.com/FreeCAD/FreeCAD/pull/8786
This PR fixes the pure segmentation fault.

rrrossi, 2023-03-08:
what is the meaning of the # sign in this context?

rrrossi, 2023-03-08:
> > However, the patch doesn't fully fix the problem because there is still the line `path.resize(path.size() - 6);`
> 
> Correct. I guess this is originally present to remove the '._self' previously added, but because of the patch it may not be actually added.
I think that retry == 2 can never happen if path is empty or pat terminates with # or . or there is a # in path

commit b7b81138 (2023-03-09): Gui: fixes #8501: FreeCAD crashes when constraining a sketch with an expression containing . (dot)
closed 2023-03-09 by wwmayer