FreeCAD/FreeCAD #8502: [Issue]

xixxi, 2023-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Issue description

Sketcher:
Create circle and triangle around it.
each line with tangent constrain with circle.
define constant length for 1 line.
then make circle bigger and bigger, until two triangle lines become parallel => FreeCad crash.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-15:
Would you consider trying to recreate this on the dev version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

chennes, 2023-02-15:
I can't reproduce in the latest dev version: when the circle hits the maximum 50mm diameter, it stops getting bigger. 
```
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.0.31821 (Git)
Build type: Release
Branch: refactorDataSource
Hash: e52f7aac17685e7543200afd46079c1997743a08
Python 3.10.9, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)

```

AjinkyaDahale, 2023-02-17:
Cannot reproduce with
```
OS: Ubuntu Core 20 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31924 (Git) Snap 611
Build type: Release
Branch: master
Hash: 29cebe45c05850961290ae323e589528c0956c77
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

xixxi, 2023-02-17:
> @AjinkyaDahale <https://github.com/AjinkyaDahale> Do you have a moment to
> try to reproduce ?

Yes, i'll do it today. I can record video.

luzpaz, 2023-03-01:
@xixxi bump

xixxi, 2023-03-01:
[bug_line_vert.zip](https://github.com/FreeCAD/FreeCAD/files/10861406/bug_line_vert.zip)

Bug works only if triangle created by lines, not by triangle tool.

AjinkyaDahale, 2023-03-01:
Might be windows specific. Doesn't seem to happen in the aforementioned system.

luzpaz, 2023-11-26:
@xixxi can you still reproduce on latest windows development version https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

maxwxyz, 2024-03-30:
Will close this as there was no further feedback and I cannot reproduce this with the provided file or a new one on
<details><summary>Windows 11 0.22dev</summary>
<p>

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36487 (Git)
Build type: Release
Branch: main
Hash: f5baef888c25937026ee50001a4fe4293a9f11ee
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.4
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.15
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.9

``` 

</p>
</details>

closed 2024-03-30 by maxwxyz