FreeCAD/FreeCAD #8509: Sketcher Solver Failures

louisrousseau, 2023-02-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31847 (Git)
Build type: Release
Branch: master
Hash: 502ef8c9ba619671b5137833851a7ebbbab24bf0
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/Canada (fr_CA)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

Apologies, this bug report is going to be messy on account of ongoing trial and error when these messages occurred and not noticing them immediately.

Got two different messages saying to report the issue while trying to add Spreadsheet and Expression based scaling to my model:

`Important: the LevenbergMarquardt solver succeeded where the DogLeg solver had failed.`
and
`Important: the SQP solver succeeded where all single subsystem solvers have failed.`

Initially had failed recomputes with little else to go on - the "Report Window" is unfortunately quiet on these things - started suspecting the way my constraints are set might break on a diagonal once it starts using scaled values.

I removed one length constraint on diagonal `18-Line` and replaced it with a vertical one using expression `sqrt(pow(<<Dimensions>>.DiagonaleEntreNiveaux; 2) / 2)` instead.

I had inserted rows and renamed aliases along the way and it looked like the rename did not propagate correctly, which broke some constraint references whom were no longer following changes in my scaling factor, instead trying to get the original unscaled value in column `C`.

I tried scaling factors of 0.5 and 2, going back to one with a successful solve in between.

To see my sketch scale I would sometimes need to double-click on a constraint and cancel out of that dialog.

See attached for a version of the file taken while writing this report.

### Anything else?

[T-RV-7.zip](https://github.com/FreeCAD/FreeCAD/files/10752477/T-RV-7.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

louisrousseau, 2023-02-16:
Quick update, there were all kinds of problems I hadn't noticed with my constraints earlier: coincidents not on the correct vertices, overlapping lines, etc.

I think I've managed to clean all of that up and the sketch now **looks** like it scales correctly both up and down.
[T-RV-7.zip](https://github.com/FreeCAD/FreeCAD/files/10752982/T-RV-7.zip)

I still get the errors though, to reproduce:

1. Open the attached file 
[T-RV-7.zip](https://github.com/FreeCAD/FreeCAD/files/10753054/T-RV-7.zip)

2. In the "Dimensions" spreadsheet modify the factor in C23 from 1 to:

- 2 to get `Important: the SQP solver succeeded where all single subsystem solvers have failed.`
- 0.5 to get `Important: the LevenbergMarquardt solver succeeded where the DogLeg solver had failed.`

After repeated edits in that cell the program became unresponsive and I had to end its task.

luzpaz, 2023-03-26:
@AjinkyaDahale can you weigh-in when you have a moment?

louisrousseau, 2023-03-30:
Won't be in a position to test for several days, the unresponsiveness happened after multiple edits rather than a specific value that I could recall.

bgbsww, 2023-10-31:
I was unable to duplicate using the instructions above, and 30 values between 0.1 and 5.  Changing the scale in the spreadsheet does result in about a 1 sec delay, but the sketch solves and updates.

I do see
```
19:08:40  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
19:08:40  Invalid solution from LevenbergMarquardt solver.
```

luzpaz, 2023-10-31:
@bgbsww please always paste your full [About](https://wiki.freecad.org/About) info. Thanks!

AjinkyaDahale, 2023-11-02:
So are we closing this as "not reproducible"/"need more information"?

luzpaz, 2023-11-02:
> I do see
> 
> ```
> 19:08:40  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
> 19:08:40  Invalid solution from LevenbergMarquardt solver.
> ```

@AjinkyaDahale looks like there is still some issue as mentioned by @bgbsww

AjinkyaDahale, 2023-12-01:
So what I get is:
1. Move value to 0.5: nothing
2. Move value to 2.0: report as per @bgbsww 
3. Move value to 1.1 (say) and gradually increase: a new report, same everytime, even when changing to 2.0
   ```
   Revolution: Revolve axis intersects the sketch
   ```

version:
```
OS: Ubuntu Core 22 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35260 (Git) Snap 850
Build type: Release
Branch: main
Hash: 0e9d5914cb5590df599a0cd07e417648fb46a92b
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

bgbsww, 2023-12-02:
Toggling back and forth between 1 and 2 seems to quickly start to repeatedly throw warnings.  I have not been able to get it to be unresponsive, but the calculation time is quite variable between sub-second and several seconds.
```
14:25:07  Invalid solution from LevenbergMarquardt solver.
14:25:07  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
14:25:07  Invalid solution from BFGS solver.
14:25:12  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
14:25:12  Invalid solution from DogLeg solver.
14:25:22  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
14:25:22  Invalid solution from LevenbergMarquardt solver.
14:25:22  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
14:25:22  Invalid solution from BFGS solver.
14:25:27  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
14:25:27  Invalid solution from DogLeg solver.
14:25:31  Updating geometry: Error build geometry(5): gp_Circ::SetRadius() - radius should be positive number
14:25:31  Invalid solution from LevenbergMarquardt solver.
```
```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35267 (Git)
Build type: Debug
Branch: bgbsww-patch-16
Hash: d580731b1eea0edf547880e0e5d9c995ca330d0e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * fcgear 1.0.0
  * Plot 2022.4.17
  * DynamicData 2.53.0
  * BIM 2021.12.0
  * parts_library
```

luzpaz, 2025-01-29:
@AjinkyaDahale can you check the status of this ticket? TIA

AjinkyaDahale, 2025-01-29:
I can confirm this bug still persists. As described in the previous comments, this is more of a "limitation" of the current solver than a bug, in the sense that the solver is not designed to handle such huge jumps.

AjinkyaDahale, 2025-05-28:
What I said in the previous cokmenr remains valid.

This case is probably not fixed, except accidentally as a side effect of some PR. If that IS they case though, the larger issue still remains an issue.

Also related: https://github.com/FreeCAD/FreeCAD/issues/21553. A different jump likely happens there with the same effect.