FreeCAD/FreeCAD #8523: In Sketch:Attachment::Placement, when entering a valut to move in y-axis, it actually moves the sketch in the z direction

Kulamigator, 2023-02-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * Curves 0.6.5
  * Manipulator
[/code]
```

### Subproject(s) affected?

None

### Issue description

I want to move the sketch in y direction.
I highlight the sketch and under properties, I go to attachment --> ... --> placement

I enter 30mm under "Verschiebung" (displacement).

Then I doubleclick the sketch. Now, the sketch has been moved by 30mm in the z-direction, instead in the y-direction

Then I do the same, but this time I enter 30mm in z-direction.

Now, the sketch has been moved in z-direction instead of the y-direction.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-02-23:
I suggest closing if there isn't any response within the week, this is most likely just user cofnussion

Kulamigator, 2023-02-23:
Yes, please close it.

Kulamigator, 2023-02-23:
> Hmmh, I don't have permission to close, you as creator of the issue can close or a maintainer would have to chime in

I closed it.

closed 2023-02-23 by Kulamigator