FreeCAD/FreeCAD #8526: Path tools is not loading and saving the right file path for shape file and definition

CNC2030, 2023-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
Installed mods: 
  * FCGear
[/code]
```

### Subproject(s) affected?

Path

### Issue description

When you press the tool function the first time, Freecad asks you if you would like to create folders and copy example files. This step works. However, on windows the file paths are not used correct. 
[image]

This is the current file path, C:\FreeCAD_config\Shape
In the screen you can see that the file selector chooses the right path
[image]

When you try to save Freecad throws different types of errors and the common denominator is that it cannot find the files:
[image]

Please note that the file path is not compatible with windows: 
[image]

If you edit the definition file with the correct file path you can have the tool definiton (JSON) working, but saving breaks it again

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-18:
Looks like it's using the windows backslash format instead of foward slash

Here's the code location:  
https://github.com/FreeCAD/FreeCAD/blob/de95a22eae2b8c2bb5e440fab208912704d02800/src/Mod/Path/Path/Tool/Gui/BitLibrary.py#L714

luzpaz, 2023-02-20:
Looks like `os.path.sep` isn't being used correctly for windows. @mlampert can you take a look ?

JohnOCFII, 2025-01-14:
@maxwxyz - Not sure should be auto-closed as Sliptonic had moved it to his list for CAM activities.

PR #21425 (2025-06-02): CAM: Integration of the tool and shape editors from BTL (Better Tool Library)
This very large PR essentially replaces CAM Tool Management completely, as part of the [Better Tool Library](https://github.com/knipknap/better-tool-library/) integration.

## Relevant Issues

Discussion: #9466

I also went through all open CAM issues to find any that may be resolved or otherwise affected:

- Resolves: #8201
- Resolves: #8526
- Resolves: #9625
- Resolves: #11522
- Resolves: #12712
- Likely resolves: #12823, but should be tested to be sure
- Possibly resolves #13603, to be tested
- Resolves: #17083
-  [... cut]
closed 2025-06-02 by sliptonic