FreeCAD/FreeCAD #8534: [Feature Request] Add the ability to specify the path to the OpenFileDialog folder that opens when the user changes the App::PropertyFileIncluded property.

FreeCAD-Tools, 2023-02-18:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

...

### Subproject(s) affected?

Core

### Idea description

When we add a property like 'App::PropertyFileIncluded' we have ability to specify a filter for the OpenFileDialog:

https://github.com/FreeCAD/FreeCAD/blob ... #L613-L629

It would be nice if could specify a path to folder so that when user try to change the 'App::PropertyFileIncluded' property, OpenFileDialog opens immediately in a specific folder.

Let's say I have a property that stores a sheet template. But if I want to change the template, OpenFileDialog opens in

C:\Users\....\AppData\Local\Temp\FreeCAD\Cache\FreeCAD_Doc_d98298d3-068e-420b-a18f-e67b3a9598d8_da39a3_9188

And there is nothing useful there.

It would be better if could specify the path to the folder in which all the templates are located. Then if the user changes the parameter, OpenFileDialog opens in the templates folder, which is very convenient.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-18:
@FreeCAD-Tools please fix broken link

FreeCAD-Tools, 2023-02-18:
Fixed.

luzpaz, 2023-02-18:
FTFY

Roy-043, 2023-02-20:
I don't think this FR makes sense. To change the template of a TD page you need to change its Template property. Not its Page Result property. If you edit the Template property you will land in the template directory. Unpacking an included template file in that directory would be undesirable, and there would be a significant risk of overwriting an existing file.

FreeCAD-Tools, 2023-03-06:
I don't think you understand what I'm talking about.
Imagine that the user wants to change the template, he click on filed an OpenDialog showed. He does not need Temp folder that does not contain anything, he needs a folder with templates. And looking for a folder with templates and spending a few minutes on it every time is not convenient.

Therefore, I suggest adding the ability to specify a path. This is an option.

FreeCAD-Tools, 2023-03-06:
@wwmayer a time ago is add an option to specify file types in Open Dialog. But right now he's probably busy with something more important.

FreeCAD-Tools, 2023-03-06:
In principle, this is a trifle. It just makes the interface more convenient.