FreeCAD/FreeCAD #8548: [Problem] requirements.txt is out of date

chennes, 2023-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.0.31927 (Git)
Build type: Release
Branch: cMakelibfmt
Hash: 29660e5a61f0ad586866cad675ed9665666ccca4
Python 3.10.9, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
[/code]
```

### Subproject(s) affected?

None

### Problem description

https://github.com/FreeCAD/FreeCAD/blob/master/requirements.txt

lists dependencies on versions of PySide and Shiboken that we don't support (as far as I am aware). We should also be in the process of removing any need for six (though I don't know if that's complete). I also don't know where markdown is getting used.

This document should be refreshed to correspond to our current actual requirements, so that packagers could use it if they wanted to.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-03-02:
An automated scan using pyreqs shows the following direct imports in each Mod:
```
./src/Gui/requirements.txt
----------
pivy==0.3.0
ptvsd==4.3.2
rpdb2==2.0.0.1.2

./src/Mod/AddonManager/requirements.txt
----------
requests==2.27.1
vermin==1.5.1

./src/Mod/Arch/requirements.txt
----------
ifcopenshell==0.7.0.230218
ladybug==0.0.2
numpy==1.24.2
pivy==0.3.0
pyshp==2.3.1
pysolar==0.10

./src/Mod/Draft/requirements.txt
----------
lazy_loader==0.1
pivy==0.3.0

./src/Mod/Fem/requirements.txt
----------
BOPTools==0.12
matplotlib==3.7.0
numpy==1.24.2
pivy==0.3.0
ply==3.11
pyNastran==1.3.4
PyYAML==6.0
six==1.16.0

./src/Mod/Image/requirements.txt
----------
pivy==0.3.0

./src/Mod/Import/requirements.txt
----------
Assembly==1.3.0
six==1.16.0

./src/Mod/Material/requirements.txt
----------
ConfigParser==5.3.0
PyYAML==6.0

./src/Mod/Mesh/requirements.txt
----------
pivy==0.3.0

./src/Mod/MeshPart/requirements.txt
----------
numpy==1.24.2

./src/Mod/OpenSCAD/requirements.txt
----------
ply==3.11

./src/Mod/PartDesign/requirements.txt
----------
numpy==1.24.2

./src/Mod/Path/requirements.txt
----------
area==1.1.1
lazy_loader==0.1
numpy==1.24.2
ocl==0.12
opencamlib==2023.1.11
packaging==23.0
pivy==0.3.0
six==1.16.0

./src/Mod/Plot/requirements.txt
----------
matplotlib==3.7.0

./src/Mod/Show/requirements.txt
----------
pivy==0.3.0

./src/Tools/requirements.txt
----------
bzr_pqm==1.4.0
cog==0.6.1
pivy==0.3.0
requests==2.27.1
```

Pesc0, 2023-03-25:
@chennes [pyreqs](https://github.com/rmad17/pyreqs) or [pipreqs](https://github.com/bndr/pipreqs)?

chennes, 2023-03-25:
pipreqs, sorry