FreeCAD/FreeCAD #8549: Fillet is bumpy on edge of cylinder

redfaern, 2023-02-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Current Released Version

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]

AND 

Weekly Build Version

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31847 (Git)
Build type: Release
Branch: master
Hash: 502ef8c9ba619671b5137833851a7ebbbab24bf0
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

It is very easy to reproduce. Create a cylinder with radius 25 mm and default height (10 mm). Pick any edge or face and create a fillet with default radius (1 mm). The filleted edge will show the bumps. I have experimented with different radii and the bumps are present with a 10 mm radius cylinder (barely visible) up to 64 mm radius. Using a 65 mm radius cylinder, the bumps disappear and the fillet is fine.

This issue also happens when the fillet is reduced to 0.5 mm radius or increased to 2 mm radius, or set anywhere inbetween.

The problem is reproducible in the Part Design workbench and the Part workbench.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-02-20:
I think the bumps are just because of how the items are drawn. These should, in principle, not show up in 3D printing, for example.

redfaern, 2023-02-20:
I thought that originally, but they are present in the STL file when the item is exported and they are present in the 3D print.

redfaern, 2023-02-20:
This is the exported STL opened in Cura.
[image]

AjinkyaDahale, 2023-02-21:
Interesting. This may actually be two issues in a trenchcoat: Some sort of discretization is done to display the fillets and the STL uses this discretization instead of doing this. 

Sidenote: STL is always going to be approximate. Could you try IGES or STEP export?

AjinkyaDahale, 2023-02-21:
Also, please try to add an example file in the future. Github won't accept FCStd, but you can always rename them to zip.

redfaern, 2023-02-21:
[Token.zip](https://github.com/FreeCAD/FreeCAD/files/10798213/Token.zip)
Okay, I will upload a zipped FCStd file in the future. I have just uploaded it for this bug now.

I exported to both IGES and STEP. Cura does not support either so if they do fix the problem, it won't fix it for the purposes of 3D printing in my case. I have not explored any other slicer software to see if they support IGES or STEP. The only method I have of opening IGES or STEP on my PC is with FreeCAD.

When exported to IGES and opened in FreeCAD, the fillet doesn't have the bumps, but I wouldn't call it smooth either. It looks very slightly bumpy around the entire 360 degrees of the fillet. This also happens when I have a much larger radius cylinder (>=65 mm) when the bumpy fillet problem disappears. I guess this is just how IGES does things.

When exported to STEP and opened in FreeCAD, the problem is still present. The fillet looks identical to the one in the original FreeCAD file with the bumps present around half of the fillet.

The thing that is weird is that half the fillet is fine and half is not. IMO, the bumps are not related to how FreeCAD renders the view on screen. The bumps seem to be a part of the shape of the body. When I rotate the body, the bumps rotate with the body. Note, I know basically nothing about the software side of FreeCAD or CAD in general.

redfaern, 2023-02-21:
This is a close up view of the bumps and from a different angle.

[image]

AjinkyaDahale, 2023-02-22:
I tried this a bit. There is no visible artifact when the fillet is 2 mm on my end.

sliptonic, 2023-02-22:
I also tried this.  I see the bumps in the 3D window and agree with @AjinkyaDahale that it look like a rendering issue.  I exported as both .stl and .amf and imported both into cura.  I don't get the bumps there, just the normal facets from conversion to a mesh format.

As far as I know, none of the 3D printing slicers will read .stp or .iges.

AjinkyaDahale, 2023-02-23:
@sliptonic I think there is something to investigate here. When exporting as STL and re-importing, there is a clear distinction of the fillet meshing at places.
[image]

[Token-Blank TokenFillet.zip](https://github.com/FreeCAD/FreeCAD/files/10809624/Token-Blank.TokenFillet.zip)

I suspect this is yet another OCCT issue, but to confirm that we may have to re-create the object in OCCT's DRAW.

redfaern, 2023-02-23:
@AjinkyaDahale I wasn't clear in my initial post, but at 2 mm it is still visible, but barely. I was just trying to explain the range of cylinder and fillet radii when the issue occurred. You can just make it out in this image where the fillet radius is 2 mm. It is visible roughly between 2 and 4 pm when viewed from above (the view you get by pressing '2').

[image]

In my testing, the bumps appear when exported as either STL or AMF and opened in Cura. The bumps also appear on the 3D print.

FlachyJoe, 2023-02-23:
The issue doesn't came from the fillet as a torus show the same problem
[image]

adrianinsaval, 2023-02-23:
does reducing the deviation parameters help? this is most likely a problem with how opencascade is meshing this shape

luzpaz, 2023-06-10:
bumping for visibility

AjinkyaDahale, 2023-06-11:
@FlachyJoe is the "torus" a revolved quarter-circle?

FlachyJoe, 2023-06-11:
@AjinkyaDahale no, it's a Part_Torus
[image]

gircysm, 2023-06-20:
[image]

Definitely seems to have some relation between radius, view deviation, and view angular deflection, as suggested above. Could just be exacerbating and illustrating some underlying problem in meshing.

For posterity/recap, an exported mesh appears to have the resolution set by the Preferences > Import-Export > Mesh Formats > "Maximum mesh deviation" setting, rather than the view option, and appears to export smooth for the examples at 0.10um.

Manually meshing in the Mesh Design workbench gives interesting results, in that the view mesh seems to have higher resolution than the "Create[d] from Shape" meshes using the same meshing deviation values as the view deviations. Workbench generated meshes seem comparably accurate with angular deviation < 5.0deg vs a view default of 28.5deg, both at 0.1mm deviation.
I don't if one is more "true" than the other, especially if meshing algorithms are different, or if "angular deflection" != "angular deviation". Meshes generated through the workbench don't immediately appear to have the sames bumps / weird normals / whatever, though matching settings was not exact.

(Also not to get sidetracked, but https://tracker.freecad.org/view.php?id=2059 appears to be possibly reverted; view objects don't update automatically, and need the object, or even the feature in the object, to be explicitly recomputed. As it's not used for export, maybe this doesn't matter as much.)

Possibly a quick fix the halve the default view Angular Deflection, and/or update its tool-text to indicate it's not used for export (unless it is by default, and gets overrides with Mesh Design workbench installation) and direct user to that setting? 

[BumpyTests.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11803605/BumpyTests.FCStd.zip)

[code]
Version: 0.20.2. Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
[/code]

AjinkyaDahale, 2023-06-22:
Did some simple tests with a torus. One difference is that despite the artifacts, the STL exported remains clean, as opposed to the fillet.
[image]
[image]

[thin-torus.zip](https://github.com/FreeCAD/FreeCAD/files/11833703/thin-torus.zip) (Extract once for FCStd and STL)

FOREVEREALIZE, 2024-07-11:
I am also having this issue.

> Sidenote: STL is always going to be approximate. Could you try IGES or STEP export?

This is a model I am experiencing the issue with, exported to STEP, and loaded into PrusaSlicer. (also for 3d printing)

[image]

CC: @AjinkyaDahale

AjinkyaDahale, 2024-08-21:
@FOREVEREALIZE did you intend to attach a document?

swdw973, 2024-10-23:
There is a rendering issue that has cropped up. Not sure if it's RC2, the latest windows update, or the new Nvidia studio drivers. These were all updated in the last week. 

This is dealing with the rendering of a rounded edge at the top of a revolve. The rounded top is an arc in the sketch, not a fillet that's been added. When I exported it as an stl, there was no problem with the object. All files are included. Issue did not occur in RC1, but all these updates occurred in between the last time I used this file and opening it today.
FreeCAD render
[image]

STL file in slicer
[image]

Computer configuration
[Uploading Belarc Advisor - Computer Profile.pdf…]()

hyx0329, 2024-10-25:
In my case(FreeCAD 1.0.0 Revision: 38806 (Git), RC2):
- Current preview has defects(very obvious)
- Exported STEP model has defects(much smaller), slightly visible in superslicer
- Exported STL mesh model has no visible defect

The attachment is a simple small cylinder with a fillet.

All three files in one zip(FCStd, STEP, STL):
[freecad-bumpy-fillet.zip](https://github.com/user-attachments/files/17521996/freecad-bumpy-fillet.zip)

inhumantsar, 2025-02-17:
Running into this as well. The issue wasn't apparent until I exported the model to a STEP file and loaded that into Orca Slicer. I ignored it at the time since I was only doing a test print.

The part is a thin-walled (1.6mm) cover. The inner and outer edges both get 5mm fillets. The sawtooth pattern appeared on both. It's hardly visible in FC but is easily noticeable in the slicer.

Later I came back to the model to do a Part Split and noticed that the issue became more pronounced after splitting.

Before

[image]

[image]

After:
[image]

[image]

Happy to provide files and more detail if desired.

inhumantsar, 2025-04-20:
I came back to this today and realised I never tried these suggestions:

> For posterity/recap, an exported mesh appears to have the resolution set by the Preferences > Import-Export > Mesh Formats > "Maximum mesh deviation" setting, rather than the view option, and appears to export smooth for the examples at 0.10um.
> ... 
> Possibly a quick fix to halve the default view Angular Deflection

Maximum angular deflection was set to 6.4 by default. I reduced it to 3.2, then 1.6, then 0.8, restarting the app and reopening the model each time but there was no discernible effect. 

Changing MeshDeviation in Parameter Editor > Mod > Part did not fix the viewport, though it did fix the exported STEP file.

Changing maximum deviation in Preferences > Part/Part Design > Shape View to 0.0001% and hitting Apply fixed the viewport issue and the exported STEP model. I had to save the project before closing the app for the fix to stick. If I didn't, then the issue returned when reopening the project. Worth noting that this worked even though the preference field reverted to 0.01% when clicking outside it, the MeshDeviation parameter showed 0.01 in the UI and in user.cfg.

```xml
<FCFloat Name="MeshDeviation" Value="0.010000000000"/>
```

I tried manually setting the maximum deviation preference and the MeshDeviation parameter back to 0.01, saving the project, then restarting the app but the issue did not reappear. 

I also tried setting deviation back to 0.01, closing/reopening the app, then creating a new project with a 15x15mm cylinder and filleting its edges at 1mm and 5mm. The sawtooth pattern was not visible in either the viewport or the exported STEP file. I've attached the [FCStd and STEP file](https://github.com/user-attachments/files/19825994/FilletTest.zip).

Note that I'm using the AstoCAD fork:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40208 (Git) Conda
Build type: Release
Branch: AstoCAD
Hash: 760cbe471b1cf8d7408e7cbf48f876960012a59c
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/Dark/Fusion
Logical/physical DPI: 96/146.099
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * CAM
  * Draft
  * Fem
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * sheetmetal 0.7.11
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web

FlachyJoe, 2025-06-10:
@PaddleStroke what is your Body.Deviation value? I still have the same display with 0.5

```
OS: Debian GNU/Linux 13 (trixie) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41790 (Git)
Build date: 2025/05/19 19:40:38
Build type: Unknown
Branch: main
Hash: 2e4d7479a76c0bf85698b0aa10d47a1d393a5998
Python 3.13.3, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.16
shiboken 5.15.16, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 8.0.0.rc1-7aa85582ad
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 85/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * lattice2 1.0.0
  * Lithophane
  * Curves 0.6.62
  * fasteners 0.5.38
  * Assembly4 0.50.18
  * Telemetry 1.0.1
  * FreeCAD_SketchArch
  * Road 2025.2.22
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.22
```

FOREVEREALIZE, 2026-02-26:
Still observing this on FreeCAD 1.0.2

[UmbrellaHandle.FCStd.zip](https://github.com/user-attachments/files/25582613/UmbrellaHandle.FCStd.zip)

EnrichedQuanta, 2026-11-16:
I am running into the same issue with fillets on circles... I noticed some say that changing the deviation helped but it did not on mine.
I tried both at 0.1 and the default 0.5mm.

I will corroborate that it does translate over to the Slicer software and therefore to a 3D print at least when exported via STL or STEP. I tend to use STL myself...

**About my installation:**
I just did a fresh install on _Kubuntu 26.04 LTS_.
I installed _FreeCAD 1.1.4_ via FlatPak (Was noticable in the .AppImage install as well on a previous boot)

**My steps to recreate this issue easiest are:** 
1. Sketch a circle of any radius.
2. Close Sketch.
3. Extrude Sketch any distance. Lets do 10mm.
4. Create a fillet on any edge.

**My theory:**
I very much notice when zooming in and the smaller the circle, the more noticeable it is, but its never a true arc. Its always a series of straight lines. It appears to be each of the vertices along the circle that cause the effect, it creates two small triangles to match the smaller top line to the larger bottom line around the circle.

I do think the fact that they aren't making a proper arc and is just a series of lines is fundamental to how CAD software processes, circles but I am not 100% certain on that...
[image]

Tweaking around the fillet thickness, I can still see that it is there on a 9mm radius(though its almost negligable at this point, but further proving that it has to do with how circles are processed in CAD software.)
[image]