FreeCAD/FreeCAD #8556: [Issue] PNG files not scaled / displayed properly.

julia-m-stephenson, 2023-02-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.5
  * fasteners 0.4.27
[/code]
```

### Subproject(s) affected?

None

### Issue description

When opening a PNG file the view seems to be constrained to a letter box. 

[image]

If you zoom in and out using the scroll wheel it displays tiny.

[image]

As a workaround I opened the file in inkscape and saved as SVG. This file was opened as a drawing and displayed correctly.

[image]

Here is the PNG.
[image]

Here is the SVG.
[image]

Thanks. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-02-20:
How are you "opening" the file?  What command? What workbench?

julia-m-stephenson, 2023-02-20:
> How are you "opening" the file? What command? What workbench?

File->Open

WandererFan, 2023-02-21:
I didn't know png was a valid import type.  I don't think this works in v0.21.

You could add the image to a TechDraw Page with [TechDraw_Image](https://wiki.freecad.org/TechDraw_Image).

julia-m-stephenson, 2023-02-21:
It's in the list of supported file formats. 

https://wiki.freecad.org/Import_Export

It almost works just the zoom levels are incorrect. 

This isn't hugely important to me. I'm just raising it as an issue so people know about it. Maybe it's an easy fix or maybe it's something the development team want to drop .

sliptonic, 2023-02-23:
Opening the image isn't terribly helpful since it just opens the image in a new tab and you can't really do anything with it.  To be honest, I don't know why this functionality exists.

If you want to use the image as a reference, go to the image workbench.  import the image to an imageplane.  You can orient the plane in 3D space and scale it as needed.  Then it's useful to draw over.

sliptonic, 2023-02-23:
@julia-m-stephenson if the imageplane does what you want,  I'm going to suggest we change the title of this PR to "Opening an image from the file menu is confusing and not useful"  
Then I'm going to suggest we remove the functionality entirely.

Maybe one of the UI oriented folks like @PaddleStroke  or one of the old timers like @yorikvanhavre knows why this exists and if it should be preserved.

PaddleStroke, 2023-02-23:
I didn't knew FC had a basic image viewer. I don't really see the point as all OS have basic image viewers.
I think basic viewer upon opening png is useless and could be removed. However it could be interesting that opening a PNG file does either of the following : 
A - trigger the import of the image to an image plane to current document.
B - Create a new document with this image in the XY plane.

By the way, the image workbench could maybe be elaborated to enable basic image editing directly in FC.

WandererFan, 2023-02-23:
AFAICT, in v0.21 you can not open a png file using File > Open.

sliptonic, 2023-02-23:
> AFAICT, in v0.21 you can not open a png file using File > Open.

I can.  It's still in the supported formats list.  Opening from the File menu does the same thing as this command:

[image]

> A - trigger the import of the image to an image plane to current document. (but it could be argued that this is the job of 'import').

That's a good idea.  It could automatically align the plane to the current view.

T

> By the way, the image workbench could maybe be elaborated to enable basic image editing directly in FC.

Better tools for scaling and placement but let's kick other kinds of editing down the road.

WandererFan, 2023-02-23:
My bad.  I was using a build without the ImageWB.

sliptonic, 2023-02-23:
FYI, several of the [complaints raised at Fosdem](https://forum.freecad.org/viewtopic.php?t=75768#p658200) were about the image workbench, scaling, and display.

PaddleStroke, 2023-03-17:
[image]
What is the image plane icon supposed to look like?

sliptonic, 2023-03-19:
- [X] Move Image loading to core
- [ ] Implement task panel
- [ ] Remove old Image workbench
- [ ] Revise wiki documentation

adrianinsaval, 2023-03-20:
> What is the image plane icon supposed to look like?

I think that is a generic FeaturePython icon, not specific to image planes

sliptonic, 2023-03-22:
PR 8955 is a significant improvement.  It specifically fixes two issues from the [Fosdem complaint session](https://forum.freecad.org/viewtopic.php?t=75768#p658200)
- Image workbench: once the dimensions are set, changing them uniformly is a pain.
- Images from Image WB show the back side as black: should be transparent, or the image reversed.

The imageplane now has a dedicated task panel that can be used to adjust the scale and maintain aspect ratio.

The PR preserves the ability to open an image in a separate MDI window.  This is deemed a feature since someone may want to look at the image in one window while editing a model in another.

This issue remains open because there is additional work to do.

As of now, there are four different ways to open the image in the MDI window and two ways to open it in an imageplane.

MDI window:

1. file->import 
2. File->Open...
3. Tools->Load 
4. Image WB toolbar button image_open

Image Plane:
1. Drag an image into the 3D window.
2. Image WB toolbar button image_CreateImagePlane

The image workbench should be considered deprecated and will eventually be removed.  When it is, MDI(4) and ImagePlane(2) will go away.

In my opinion file->import should create an imageplane and load the image.
File->open should open the image in a new MDI window
Tools->Load is redundant and can be removed.

Two-point scaling needs a reworked workflow.  AFter that, it can be incorporated into the imageplane task panel and image workbench can be removed.

Did I miss anything?

LHC-Canada, 2023-03-24:
It would be great to have support for older model files that already have images loaded into them

adrianinsaval, 2023-03-25:
Would removing the workbench mean that previous image planes will stop working? A migration script might be a good idea then

adrianinsaval, 2023-04-22:
should this be closed now?

chennes, 2023-04-22:
I think so 👍

closed 2023-04-22 by chennes