FreeCAD/FreeCAD #8557: [Feature Request] GSoC:  FreeCAD-BRLCAD integration

grd, 2023-02-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://wiki.freecad.org/FreeCAD-BRLCAD_integration

### Subproject(s) affected?

Other (specify in description)

### Idea description

Allow to open, edit and save BRL-CAD models in FreeCAD 

https://wiki.freecad.org/index.php?title=Google_Summer_of_Code_2023

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-02-20:
Let's improve the title of this issue.  Can we express it as a problem?  
For example: Problem:  User can't open/modify BRL-CAD models in FreeCAD

grd, 2023-02-20:
I think that it is a problem because it's 350h, but you can better ask the mentors.

utkarshraj779, 2023-02-25:
@grd Hlw I am utkarsh kumar from IIT Kharagpur and i already worked in free cad software and have good ratting in c++ in hacker rank please suggest me topic that should help in contribution process.

luzpaz, 2023-02-26:
@utkarshraj779 are you interested in this ticket or in GSOC in general? If the latter, then please announce yourself on our forum.freecad.org

yorikvanhavre, 2023-02-27:
Corresponding issue on BRL-CAD umbrella repo: https://github.com/opencax/GSoC/issues/14

DiviyamPathak, 2024-03-09:
hello 
I wanted to work on this project for 2024, previous discussions about project tells that it was too long for gsoc 2022, but it still looks interesting to me would any of @yorikvanhavre @drossberg like to mentor.

drossberg, 2024-03-09:
Hi @DiviyamPathak, nice to hear that you are interested in this experiment. The first steps are listed in the wiki's "Details":
* Getting familiar with FreeCAD and BRL-CAD
* Compile the software (for BRL-CAD it's [brlcad](https://github.com/BRL-CAD/brlcad) and [MOOSE](https://github.com/BRL-CAD/MOOSE))
* Learn about the APIs, in case of BRL-CAD you can ask me "How do I *this* in BRL-CAD?"

For further discussions, I recommend to use the usual channels:
* The [FreeCAD forum](https://forum.freecad.org/viewforum.php?f=46), open a new [GSoC 2024] topic there
* The [BRL-CAD Zulip chat](https://brlcad.zulipchat.com)

yorikvanhavre, 2024-03-11:
This is indeed potentially a VERY large project. So the difficulty here, for GSoC, is to define well what you think you can do in the 350h of the project. And indeed as @drossberg says, this involve two pretty big projects, FreeCAD and BRL-CAD so there is a lot to learn on the way :)

I would also suggest you start playing with both applications to understand what's involved.

We originally designed this project to be a "cross-project" project, mentored both by BRL-CAD and FreeCAD mentors. Although both projects are now participating to GSoC under different orgs, I think this is still valid, and a super interesting experience. So IMHO we can still do it together. We will just have to decide if we do it under the FreeCAD or the BRL-CAD banner...

iabnvkr, 2025-04-01:
Dear @yorikvanhavre @drossberg , 

I've made this [post on FreeCAD forum](https://forum.freecad.org/viewtopic.php?t=95991
) regarding working on this project. Presently I'm working on my proposal, very soon, I'll share draft version within 2 days.

Thanks & regards,
Abhinav