FreeCAD/FreeCAD #8558: [Feature Request] GSoC: Upgrade the documentation system

grd, 2023-02-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://wiki.freecad.org/Upgrade_the_documentation_system

### Subproject(s) affected?

None

### Idea description

Rework the documentation system so it supports multiple languages, and allows to switch between offline and online versions. 

https://wiki.freecad.org/Upgrade_the_documentation_system

### Anything else?

https://wiki.freecad.org/index.php?title=Google_Summer_of_Code_2023

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaktusus, 2023-02-25:
Indeed, there is no doubt that we are used to the Crowdin platrform.
An increasing number of translators are becoming fluent in it.
They are becoming familiar with the available mechanisms and functionalities.

gauriimaheshwarii, 2023-02-28:
Hello everyone!
I am Gauri Maheshwari, a second-year student pursuing Mechanical and Automation Engineering at Indira Gandhi Delhi Technical University for Women, India.

I am a technical enthusiast and a regular CAD user. I hope to participate in GSoC'23 with the FreeCAD organization.
I find this project interesting and aligned with my skillset. I have experience with C++ and Python and can code in both languages. I believe I'd be a good fit to work on this project.

Since I am new to this ecosystem, I ask for guidance on how to proceed further and contribute to successfully completing the project.

Thank you!

gauriimaheshwarii, 2023-02-28:
Hello everyone!
I am Gauri Maheshwari, a second-year student pursuing Mechanical and Automation Engineering at Indira Gandhi Delhi Technical University for Women, India.

I am a technical enthusiast and a regular CAD user. I hope to participate in GSoC'23 with the FreeCAD organization.
I find this project interesting and aligned with my skillset. I have experience with C++ and Python and can code in both languages. I believe I'd be a good fit to work on this project.

I am new to the ecosystem, and so I ask for guidance on how to proceed further and contribute to the project and the community.

Thank you!

gauriimaheshwarii, 2023-02-28:
Hello everyone!
I am Gauri Maheshwari, a second-year student pursuing Mechanical and Automation Engineering at Indira Gandhi Delhi Technical University for Women, India.

I am a technical enthusiast and a regular CAD user. I hope to participate in GSoC'23 with the FreeCAD organization.
I find this project interesting and aligned with my skillset, as I do have experience with C++ and Python and can code in both languages.

A strong open-source community, first-hand user experience, and user feedback are all important for improving the software.
Since I am new to the ecosystem, I ask for guidance in proceeding further toward contributing to the project and the FreeCAD community.

Thank you!

yorikvanhavre, 2023-02-28:
I would start by having a look at what we have put together above, and tell what you think of it, what seems easy / not easy, if this sparks any additional idea, etc.

FC-FBXL5, 2023-03-04:
I have browsed through the converted page and still have no clue how translation could work with GitHub.
I'm stll feeling uncomfortable here.
But anyway, here is my maiden post in a GitHub discussion to keep this topic going.

kaktusus, 2023-03-05:
The ability to version FC documentation for individual releases is an undoubted advantage that was not available until now.

utkarshraj779, 2023-03-07:
hey, @yorikvanhavre I am Utkarsh kumar From (IIT KHARAGPUR 3rd year undergraduate student). I think you remember me I want to tell you that last time i worked in API scrapping i mailed you my full document how better can i do in this project? I request you to please once go with my mail.  (I was fully show demo how the process is going)
thankful to you.

utkarshraj779, 2023-03-07:
And I do this best way much time better way with my full efficiency so please go to the document I mailed you and request you 
please read it once carefully.

yorikvanhavre, 2023-03-08:
I updated the issue description to include researching available file-based wiki solutions.

gauriimaheshwarii, 2023-03-08:
I checked out some file-based wiki solutions and found out about [DokuWiki](https://www.dokuwiki.org/dokuwiki) and [BookStack](https://www.bookstackapp.com/).
Both are open source software. Will explore both and update here.

ebrahimraeyat, 2023-03-08:
Hello. I found this useful. I don't know if it is suitable for FreeCAD:

[quarto](https://openscapes.github.io/quarto-website-tutorial/)

yorikvanhavre, 2023-03-08:
It could be. Another part of the problem is to present the online doc nicely. That could be done with such platform

gauriimaheshwarii, 2023-03-12:
> > I checked out some file-based wiki solutions and found out about [DokuWiki](https://www.dokuwiki.org/dokuwiki) and [BookStack](https://www.bookstackapp.com/). Both are open source software. Will explore both and update here.
> 
> Other interesting options:
> 
> * [gollum](https://github.com/gollum/gollum) (interesting: Gollum strives to be [compatible](https://github.com/gollum/gollum/wiki/5.0-release-notes#compatibility-option) with [GitHub](https://docs.github.com/en/communities/documenting-your-project-with-wikis/about-wikis) and [GitLab](https://docs.gitlab.com/ee/user/project/wiki/#create-or-edit-wiki-pages-locally) wikis.)
> * [github wiki](https://docs.github.com/en/communities/documenting-your-project-with-wikis/about-wikis)
> * [gitlab wiki](https://docs.gitlab.com/ee/user/project/wiki/)
> 
> All are git-based, that is, the contents you edit with the wiki interface is also a git repository, so you can also update via normal PR workflow, tie into other tools, etc. Gollum looks really interesting and close to MediaWiki: https://github.com/gollum/gollum/wiki/Screenshots

I tried out [gollum](https://github.com/gollum/gollum) and it does look like a promising alternative to MediaWiki. Since it integrates with Git, it will also allow version control for FreeCAD documentation.

yorikvanhavre, 2023-03-13:
Yes gollum seems to me the most promising, as it is independent from github/gitlab, and has a lot of features, and seems well maintained. However, there is the additional problem that it is an additional piece of software to be installed and maintained, while if we use github/gitlab, this is done for us and there is no additional work.

I couldn't determine how the compatibility with a github wiki works

gollum says

luzpaz, 2023-03-13:
Note: gollum is ruby based

gauriimaheshwarii, 2023-03-18:
> Yes gollum seems to me the most promising, as it is independent from github/gitlab, and has a lot of features, and seems well maintained. However, there is the additional problem that it is an [...]

For compatibility with GitHub wiki, Gollum acts as a front-end to the Git repository, providing a user-friendly wiki interface for editing and viewing pages.

Gollum is set up to use the same Git repository as the GitHub wiki, which allows editing and viewing the same content in both places. So changes made using Gollum will be automatically synced to the GitHub wiki, and vice versa.

Gollum will provide powerful editing and formatting features along with GitHub's collaborative workflow and version control capabilities, but dependency maintenance still remains a concern.

gauriimaheshwarii, 2023-03-18:
> Yes gollum seems to me the most promising, as it is independent from github/gitlab, and has a lot of features, and seems well maintained. However, there is the additional problem that it is an [...]

For compatibility with GitHub wiki, Gollum acts as a front-end to the Git repository, providing a user-friendly wiki interface for editing and viewing pages.

Gollum is set up to use the same Git repository as the GitHub wiki, which allows editing and viewing the same content in both places. So changes made using Gollum will be automatically synced to the GitHub wiki, and vice versa.

Gollum will provide powerful editing and formatting features along with GitHub's collaborative workflow and version control capabilities, but dependency and maintenance still remain a concern.

luzpaz, 2023-03-19:
IMHO, we should look for projects that run on python.

yorikvanhavre, 2023-03-23:
Another possibly interesting candidate: http://jekyllthemes.org/themes/git-wiki-for-documentation/

gauriimaheshwarii, 2023-03-28:
Hello everyone!
This is the [initial draft](https://docs.google.com/document/d/1T6jhWznzS5uG4VIFYp08QMCuGv4OISIbebxBXgYzhiw/edit?usp=sharing) of my GSoC proposal for this project.

The comment access is given to everyone. Your reviews, suggestions, and additional changes are appreciated :)

gauriimaheshwarii, 2023-03-28:
Hello everyone!
This is the [initial draft](https://docs.google.com/document/d/1T6jhWznzS5uG4VIFYp08QMCuGv4OISIbebxBXgYzhiw/edit?usp=sharing) of my GSoC proposal for this project.

The comment access is given to everyone. Your reviews, suggestions, and additional changes are highly appreciated :)

yorikvanhavre, 2023-03-29:
Very good! I left some comments there.

luzpaz, 2023-04-02:
@gauriimaheshwarii can you set it so access without a gmail account is possible ?

gauriimaheshwarii, 2023-04-03:
> @gauriimaheshwarii can you set it so access without a gmail account is possible ?

@luzpaz fixed it.

gauriimaheshwarii, 2023-05-04:
Hello everyone,
I have been selected as a GSoC contributor for this project. 
I would like to thank @yorikvanhavre, @kaktusus, @luzpaz, @FC-FBXL5, @ebrahimraeyat, and @utkarshraj779 for being a part of this discussion. Your inputs were all really helpful.🙌
A huge thanks to @yorikvanhavre for helping me out with the proposal review.✨

Looking forward to working with the FreeCAD and open source community and learning something new each day :)

yorikvanhavre, 2023-05-05:
We are all really happy to have you on board @gauriimaheshwarii ! We liked a lot your approach on this complex issue and we think you can really achieve something there. Looking forward to it! @kkremitzki will be your primary mentor, and we're still sorting out who will co-mentor which project 😅 

Be sure to read https://forum.freecad.org/viewtopic.php?p=680044#p680044 , i have put some rough guidelines for the next weeks.

Erudition, 2024-09-08:
An update to the original issue:
Weblate [now supports markdown](https://github.com/WeblateOrg/weblate/issues/3106).
There should also be more bullets on the advantages of Weblate... to me, the biggest one is support for [Project Fluent](https://projectfluent.org/), as the new Fluent format enables more natural and correct translations than any format before it.

What will happen with that GSoC project?
I followed the forum thread and saw the netlify page - it looks decent and has search, but the [source code](https://github.com/FreeCAD/FreeCAD-documentation-docusaurus) hasn't moved since last year and I see that the project's [conclusion](https://github.com/gauriimaheshwarii/gsoc-2023) has been written.

Since FreeCAD is already tied up in Github, I'd like to give my +1 on the idea of just using Github's wiki feature. Not only is it hosted and ready to go, but anyone with a Github account can make an edit (-> PR) without knowing Git or Markdown.

If a custom web design is really desired, we could move the docusaurus project into the main FreeCAD codebase. Github Pages can be configured in one click to serve html from `/docs`, so you'd already have a website on FreeCAD.github.io (custom domain possible) which is free and CDN-hosted. Or, a Github Action can be set up to rebuild the docs when changes are made.

I see keeping the docs in the same repo as a huge boon for easily keeping them up to date with the changes made to FreeCAD over time. A commit that adds a feature can also update the documentation for it in one go. The docs can then be browsed based on an exact version of the software.

Just my 2cents

yorikvanhavre, 2024-09-09:
Apparently crodin now supports markdown too and very nicely. We could start doing a test with https://freecad.github.io/DevelopersHandbook/

Reqrefusion, 2024-09-09:
https://github.com/FreeCAD/FreeCAD-Homepage/issues/190 Here I advocate the idea of ​​improving the existing one with a better alternative.

eduardtibet, 2024-12-28:
Have no idea, if the issue in Stale status actually converts into something like "Closed" after some amount of time. But anyway, just want to add my 5 cents as a Technical Writer.

While reading the proposal I couldn't understand why Markdown as a source language was chosen. The point is it doesn't have extended semantic capabilities (well... to be honest, it doesn't have them at all). Moreover due to some Python/C++ libraries are going to be documented as mentioned here: _"A simpler, file-based system would allow much more automation, scripting and integrations. Python/C++ API documentation can be built on-the-fly and  integrated"_, more Python-centric tool/toolchain seems to be a better choice. So I have no idea, why [reStructuredText](https://www.sphinx-doc.org/en/master/usage/restructuredtext/index.html) (as a source) + [Sphinx](https://www.sphinx-doc.org/en/master/) (as an offline SSG) or [Read the docs](https://about.readthedocs.com/pricing/#/community) (as a cloud publishing service) that is free for open source project was placed out of scope of the proposal/analysis. Any reason to stick to the markup language (Markdown) that even doesn't contain "Note" or "Warning" blocks within it's content model? Moreover so called markdown flavors make the transition to other toolchain(-s) or publishing provider more complex resulting in loosing flavor-specific features on move.

Also there is an [asciidoc ](https://asciidoc.org/) - the same grade lightweight markup language with the same (and somewhere even more complex) functions within [comparison table](https://asciidoc.org/#compare)

yorikvanhavre, 2025-01-07:
I think this github action should avoid issues tagged with "GSoC", as they are typically reused over years... Can it be done @maxwxyz ?

maxwxyz, 2025-01-07:
@yorikvanhavre done, I have assigned the no-auto-close to the GSoC issues. The action will not close feature requests in general so this was the only issue without the Feature tag.

Reqrefusion, 2025-01-07:
> Have no idea, if the issue in Stale status actually converts into something like "Closed" after some amount of time. But anyway, just want to add my 5 cents as a Technical Writer.
>
> While reading the [...]

It's hard not to agree with you about Markdown, @eduardtibet . It suffers from many basic features, and efforts to add these features are in my opinion inadequate. But it is quite understandable why he wanted to switch to markdown. Because most people are more comfortable with it. Personally, I prefer using html instead of markdown :D. As someone who has worked on this and is working on it, I am one of those who think a transition is ridiculous. The existing documents are already written in wikitext format. And wikitext is a more popular format worldwide than markdown, I think switching to less known formats is a negative move as there are thousands of documents with their translations. In my opinion, such a move would be unnecessary and would negatively affect those working on documentation. In this regard, I designed a hybrid system that is fully synchronized with the wiki. It has been operating at full capacity without any problems for months. And I think this is the most reasonable solution. You can check out the Wikitext repository [here](https://github.com/Reqrefusion/FreeCAD-Documentation-Project) and the html repository [here](https://github.com/Reqrefusion/FreeCAD-Documentation-html). All documents are available in all languages. In addition, relevant changes are instantly reflected on the wiki.

drLite35, 2025-02-04:
I have gone through this project and also have gone through all the related issues mentioned here and proposal made by @Reqrefusion as well, I would love to contribute to this project as this seems somewhat of a enormous task and aligns to my interest as well.
@yorikvanhavre @kkremitzki and @Reqrefusion I would like to first start by knowing the current status of all the work that has been done/ is in the pipeline cause its very tough to track what all have been achieved and what is yet to be.
All the help is welcome I would really like to start with some contribution.
P.S Already doing one of doc related project here https://github.com/FreeCAD/FreeCAD/issues/12547
Thanks!!.

Reqrefusion, 2025-02-05:
> I have gone through this project and also have gone through all the related issues mentioned here and proposal made by [@Reqrefusion](https://github.com/Reqrefusion) as well, I would love to contribute to this project as this seems somewhat of a enormous task and aligns to my interest as well. [@yorikvanhavre](https://github.com/yorikvanhavre) [@kkremitzki](https://github.com/kkremitzki) and [@Reqrefusion](https://github.com/Reqrefusion) I would like to first start by knowing the current status of all the work that has been done/ is in the pipeline cause its very tough to track what all have been achieved and what is yet to be. All the help is welcome I would really like to start with some contribution. as would need to understand completely the status what all is needed in general if this idea is into gsoc'25. P.S Already doing one of doc related project here [#12547](https://github.com/FreeCAD/FreeCAD/issues/12547) Thanks!!.

Hello, I'm glad you are interested in the project. I have written a [report](https://github.com/FreeCAD/FPA/blob/main/reports/%2315%20Documention/1st%20Report%20Dated%20December%203%2C%202024.wikitext) about progress. If you have a question that attached to your head, you can ask me from anywhere by tagging me. Even though the name is the same, we're actually dealing with different things. Wiki, which is created by writing, wants more different things from what you work on.

Raporda belirtiğim projenin git kopyası için aslında aklımda sadece iki şey kaldı: wikiden eş zamanlı veri aktarımı ki bunu ancak bazı yetkilere sahip olduğumda yapabilirim ve kişisel token yerine app kullanımı ki bunuda bir önceki şey tamamlandıktan sonra yapmak istiyorum. Bunlar tamamladığında projeye FreeCAD altına taşıyacağım ve umarım artık FreeCAD serverlarını kullanacağım. Tabi bunlar benim aklımdakiler sizin aklınızda başka şeyler varsa ekleyebilirisizin.