FreeCAD/FreeCAD #8559: [Feature Request] GSoC: GSoC FEM Solver Z88

grd, 2023-02-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://wiki.freecad.org/GSoC_FEM_Solver_Z88

### Subproject(s) affected?

FEM

### Idea description

Extend the capabilities of FEM solver Z88OS in FreeCAD FEM workbench. 

### Anything else?

https://wiki.freecad.org/index.php?title=Google_Summer_of_Code_2023

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-02-20:
Improve title by expressing as a problem that can be solved.
For example:  Problem: It is not possible to use many Z88 capabilities in FreeCAD

rahulrauki, 2023-02-24:
Hi @grd I'm interested in this idea for GSoC 2023. I'm to introduce myself in the forum to get started right?

utkarshraj779, 2023-02-27:
Hlw @grd For this project I want to apply my skill to whatever I should learn now and I want help from you please provide me with some information that should give me a best idea about how the project going to work so I could start learning.
 I am greatly thankful to you.

yorikvanhavre, 2023-02-28:
@rahulrauki yes, I saw you did it already!
@utkarshraj779 it's you guys who need to decide how you want to do this project. What we provide here is just a preliminar idea, to describe the problem we are trying to solve, and how we imagine a solution could be. But to make a good proposal, you'll need to investigate, understand the problem, and think of how you will propose to implement a solution.
Of course we are here to help! But it is fundamentally YOU who define how the project is going to work ;)

utkarshraj779, 2023-02-28:
@yorikvanhavre  I installed (z88OS 14 )version for windows and i want to see the code of free code after that i should be good for me for reading(x,y,z) dimension positon design code. should it be fine in windows or i need to install it in linux.( I already mention me in your form (utkarsh kumar)). can you suggest me where should i got last year project done in free code.

yorikvanhavre, 2023-02-28:
The FreeCAD code is easy to find, it's right here on github. You should look at how the FEM workbench works in FreeCAD. Check maybe the source code (modules are under src/Mod), but also the FEM section on https://wiki.freecad.org and the FEM section of the forum.

@donovaly and @berndhahnebach , any thought over this GSOC idea?

donovaly, 2023-03-09:
I am a bit skeptic about Z88 in general. Z88Aurora is not OpenSource and the documentation is poor.

I extended support for Z88 already a bit last summer but stopped since Z88 can only perform mechanical analyses. The main feedback up to now was that FreeCAD's FEM misses multiphysics (acoustics in combination with heat etc.). Therefore I focused on Elmer. CalculiX can also perform much more than the thermomechanical analyses we support at the moment, but the problem is in my opinion the poor documentation and low user base for non-thermomechanical analyses.

Concerning the connection to the VTK post processing, this is now resolved in master. One gets now automatically a result pipeline so that one can change constraints recalculate and see the results as one defined the pipeline.

Therefore I don't see much room for a GSoC project regarding Z88. Personally I think the FEM workbench would benefit from more multiphysics. We have now support for electromagnetics, but other basic things like acoustics are still missing.
Thanks to the recent Elmer extension project, I think this could be a relatively easy task. One can copy the code for the electromagnetics equations and modify it for acoustics and elastics and other physical equations.
The challenge is hereby to catch the cases of multiphysics. So when you e.g. perform an analysis of a fluid that is heated and the fluid flows in an elastic pipe that is squeezed while the fluid is flowing, one must solve the different equation in a certain order. Ideally this should be done by FreeCAD because otherwise one would have to dive deep into the docs of Elmer for every case.

So for GSoC, I would either extend our Elmer support as easy project or extend CalculiX as medium task.

luzpaz, 2023-03-09:
Also, Z88 team hasn't seemed to publish any updates open source updates to their solver in years.

1202anthonychen1202, 2023-03-29:
Hello @grd I'm new to open source and I am interested in doing a project with this idea for GSoC 2023. Should I introduce myself in the IRC or the forum to get started?

grd, 2023-03-29:
> Hello @grd I'm new to open source and I am interested in doing a project with this idea for GSoC 2023. Should I introduce myself in the IRC or the forum to get started?

You are asking this to the wrong guy. I only put this text into Github. But yes, you should introduce yourself and write about your ideas in the forum.