FreeCAD/FreeCAD #8560: [Feature Request] GSoC: Advanced FreeCAD test system

grd, 2023-02-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://wiki.freecad.org/Advanced_FreeCAD_test_system

### Subproject(s) affected?

None

### Idea description

Create a framework for result file based comparisons and workbench specific comparators. Also creation of test cases should be simplified by a GUI or wizard. 

### Anything else?

https://wiki.freecad.org/index.php?title=Google_Summer_of_Code_2023

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-02-20:
Need to improve title by expressing it as a problem that can be solved.
Are we talking about end-user tests or unit tests?  Is the problem that we need a tool for creating them or a documented framework for how they should be created?

Siddharth-lal-13, 2023-03-28:
Hello sir, I'm Siddharth Lal, a CS Major student based in India. I specialize in Python (along with that I have knowledge of C and C++) and I wish to work on this project and contribute to FreeCAD as much as possible, as of now I want to work on this issue as a participant via GSoC.  @grd please guide me through the application steps because I believe I can solve this marvellously within the given time frame.

Siddharth-lal-13, 2023-03-29:
> @Siddharth-lal-13 <https://github.com/Siddharth-lal-13> you should first
> read our docs at https://wiki.freecad.org/Google_Summer_of_Code_2023 then
> maybe introduce yourself on the FreeCAD forum on the GSoC thread at
> https://forum.freecad.org/viewtopic.php?t=75828&start=70
>
> Lots of things have changed in FreeCAD since this idea came up, we now
> have a solid testing infrastructure, both for python code and C++ code. The
> basics are at https://wiki.freecad.org/Testing (C++ testing is new and
> not yet documented, @chennes <https://github.com/chennes> can help
> orienting you)
>
> However, there is still a lot to be done in FreeCAD regarding unit tests:
> build more in-depth geometry testing, identify areas of FreeCAD that lack
> unit tests, etc.
>
> The first thing you might want to do, if interested in tackling the C++
> part, is try to compile FreeCAD (instructions on the wiki), and study the
> current unit testing systems.
>
> You could work solely in Python too, and concentrate on Python unit tests.
> Then you don't need to compile FreeCAD. You should still have a look at how
> FreeCAD works, and build a list of areas that lack proper unit testing.
>
> This is a bit of a tedious work, but it would be highly beneficial and
> welcome. This research work should be part of your GSoC project.
>
> However, together with your application, you should demonstrate that you
> have studied your subject, that you have looked at FreeCAD, understood more
> or less how units testing works, and that you have a good understanding of
> what you'll need to do during your GSoC time.

Thank you so much @yorikvanhavre. And my apologies for any inconvenience. I
will follow up with the things that you have mentioned.

maxwxyz, 2024-04-15:
@FEA-eng FYI

FEA-eng, 2024-04-15:
@maxwxyz I wouldn't mark it as FEM-related because it's much more general and seems to be more inclined towards Part/Part Design with no mention of FEM.

drLite35, 2025-03-01:
[image]

Hi @yorikvanhavre @grd @maxwxyz,

I have reviewed all the Python-related tests and exploring the FreeCAD codebase the ones including the test modules available at the top of each directory, such as (https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/Test).
It’s clear that there is an extensive testing infrastructure in place, but from my understanding that, given the complexity of the project, we can do a lot of work..

I wanted to ask about the long-term expectations for this project. Where would you suggest I start contributing more effectively as I continue exploring the codebase?
I am particularly interested in enhancing our unit tests and taking them to the next level.

Looking forward to your guidance!

Thanks!

yorikvanhavre, 2025-03-11:
@drLite35 the official test master now is @chennes . He might be able to give you a bit more indications. Also, we have a new C++ test framework in place, which can give hints as how things could evolve on the Python tests part.

Basically, what's inside the project description stays valid: Writing tests is still a lot of work, and consequently people don't write enough tests. How could we remedy that situation.

chennes, 2025-03-11:
I'm happy to assist with any questions you have about the test frameworks, or the tests themselves. To familiarize yourself with the testing process, you could consider picking some part of Base or App, choose a single class that is wrapped in Python, and take a careful look at the current tests. You'll find that there are two main classes of changes that are needed: one, the tests test too much at once, and two, the there are LOTS of bits of code that aren't being tested. This is almost universally true, so you really can pick nearly anything.