FreeCAD/FreeCAD #8561: [Feature Request] GSoC: Unified Measurement Facility

grd, 2023-02-20:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://wiki.freecad.org/Unified_Measurement_Facility

### Subproject(s) affected?

Core

### Idea description

 Create a generalized measurement tool that combines functionality from workbench-specific ones 

### Anything else?

https://wiki.freecad.org/index.php?title=Google_Summer_of_Code_2023

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-02-20:
Let's improve the title by stating it as a problem to be solved:
Example:  Problem: existing measurement tools are inconsistent and incomplete.

adrianinsaval, 2023-02-23:
would be a good idea to describe desired interface, measurement modes and such.

I think we want
- edge length
- face area
- shell/solid volume
- distance measurements
  - from face
  - from edge
  - from centerpoint for circles and ellipses and maybe from center of mass of arbitrary faces and middle point of arbitrary edges
  - from point
  - radius for circles and ellipses
- angle measurements
  - from faces
  - from straight edges
  - arcs

as for interface I would like to have at least two measurement commands: "Measure" and "Quick measure"
the first activates a task dialog with advanced options to choose which types of measurements to make, to choose to save those measurements as objects in the document and optionally with selection filters.
quick mesure would not open a task dialog (allowing to measure while doing something else), will try to auto detect what type of measurement is desired based on selection (optionally allowing switching between them with a keyboard shortcut, similar to how polyline tool works in sketcher) and the measurements will be temporary, disappearing as soon as you deactivate the tool.

luzpaz, 2023-02-26:
Of course we should include @easyw who's the author of the [Manipulator WB](https://github.com/easyw/Manipulator) in this discussion.

utkarshraj779, 2023-02-27:
Hlw i am utkarsh kumar from (IIT KHARAGPUR)
I already worked in free cad software https://youtu.be/9pkoUEM093U this was my last project in it and i want to contribute in this project please allow me for that one.

adrianinsaval, 2023-02-27:
Btw your video has nothing about FreeCAD as far as I can see, but I don't think previous experience developing on FreeCAD is a prerequisite for a student anyways

utkarshraj779, 2023-02-27:
yes, i am a 3rd year BTech student in (IIT KHARAGPUR) and want to play a role as a contributor. if you help me in getting ideas  material on this project i start working on it.

adrianinsaval, 2023-02-27:
I would love for you to work on this project, check the description here, my comment with desired measurements modes (it's ok if you only manage to do some of those). Check manipulator wb and assembly 4 measurement tools for inspiration but we are looking for a solution well integrated into the FreeCAD core.

utkarshraj779, 2023-02-27:
thanks i started working in this.

adrianinsaval, 2023-02-27:
I'm not really one of the organizers... sorry if I gave you the wrong impression, if you have doubts feel free to ask here or in the forum, I'm sure there will be people willing to help

DanMiel, 2023-03-29:
IMO the different measurements should take place with different actions. Perhaps when moving the curser, minimum dimensions can be shown in the status bar when an entity is hovered over, for a line the length, a vertex the x, y, z, hover over a circle or arc the radius, diameter, and center. When a dimension is required between two features then activate a program such as Quick Measure which gives measurements according to selections made. This way programs which need two selections, such as A2plus, will not be slowed down while a measurement program is running in the background.
Dan

adrianinsaval, 2023-03-29:
measurement on hover seems overkill to me.

DanMiel, 2023-03-29:
Personally I don’t remember having the measure on hover on any software I used. I thought that’s what users were asking for in this post [https://forum.freecad.org/viewtopic.php?t=77054](url) and what this issue was about. If this issue is not for having continuous input, then, of the measuring tools I have used the Quick Measure add-in is my vote of how a measure tool should act.

hlorus, 2023-05-06:
This task is now part of my GSoC project. Check out the forum topic where alot of the communication will happen.

[GSoC 2023: Unified Measurement Facility](https://forum.freecad.org/viewtopic.php?t=78147)

hlorus, 2023-08-21:
> What about doing a single measure tool, and depending on what you click you get different information. Kind of like the constraint contextual tool for sketcher.

Absolutely, that's the approach i took. Check out the forum thread for the current progress: https://forum.freecad.org/viewtopic.php?t=78147

sliptonic, 2024-04-22:
will be fixed by #9750

TheKenWiley, 2024-11-13:
I am new to FreeCad.  I'm cad/cam challenged even though I've been trying to learn since the early 70's!
To my point:  I have a sketch of a tube, that is supposed to be 4mm OD and 2 mm ID and about 100mm long.  When I import this into FreeCad I have no way of figuring out how to measure anything.  I get an OD reading of something I don't understand!  I can't find where the base of scale is!  What am I missing?  Is this the right place to look or ask?  Where is the forum for FreeCad?

adrianinsaval, 2024-11-13:
> Is this the right place to look or ask? 

not quite, this is where we discuss bugs in the software and technical things about the codebase, for usage info you can look at the https://wiki.freecad.org or ask for help in the forum

> Where is the forum for FreeCad?

It is at https://forum.freecad.org, I would recommend specifically to post in the help subforum: https://forum.freecad.org/viewforum.php?f=3&sid=e706c33306786912195c8dab3a1abd05

Jo2PEX, 2026-01-22:
Bonjour 
je bricole sur FreeCad OS: macOS 15.6.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-themes 2025.6.4
  * OpenTheme 2025.5.20
  * boltsfc 2022.11.5
  * fasteners 0.5.38
  * Help 1.0.3
et je me suis aperçu que une esquisse d un cercle de Ø 45 mm suivi dune protrusion de 68 mm suivi d un évidement pour obtenir un tube d'épaisseur de 2 mm .
 Retourne dans Mesure:  Ø extérieur 24,5 Ø intérieur 22,5 protrusion 68 mm les mesures sur les Rayon son fausse 
sa ne me gaine pas plus que ça mai je signale il y a surement d' autre qui l' on remarqué 
BJN