FreeCAD/FreeCAD #8562: [Issue] 'pi' and 'e' used with units not recognized after file reopening in Spreadsheet

mathiaslec, 2023-02-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * fcgear 1.0.0
  * OpticsWorkbench 1.0.9
  * fasteners 0.4.54
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Issue description

Hi all,
It looks like the number Pi ('pi') and Euler's number ('e') cannot be used with units right after.
It works in a first time, but after reopening the file, these constants and their units are merge and therefore not recognized.

How to reproduce :

1. Open a Spreadsheet
2. Type something in a cell like : '=pi rad' (you should see the results 180°);
3. Save the file, close it, then reopen it.
4. In the cell, the results is red and shows '=pirad'. 

Thank you for the help.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2023-02-22:
@mathiaslec could you please try with latest dev version. IIRC there has been some fixes in this area already. Thx

xtemp09, 2023-02-23:
I can reproduce this bug in the latest stable build and the latest weekly build.

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31917 (Git)
Build type: Release
Branch: master
Hash: 4639283b459f9ff9093e289908cc003db8745b82
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

mathiaslec, 2023-02-23:
> @mathiaslec could you please try with latest dev version. IIRC there has been some fixes in this area already. Thx

I still have the same problem with the latest dev version.
```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31998 (Git) Snap 623
Build type: Release
Branch: master
Hash: 57b365f18f5e4e167bfa631d82fabdb2fc07e148
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

0penBrain, 2023-02-23:
@xtemp09 @mathiaslec Thanks for testing.

0penBrain, 2023-02-23:
OK, it's worse than description, but probably it will find easier to find the bug:
1. New document, add a Part/Cube (or any shape actually)
2. Set an expression (with Property Pane) to Base/Placement/Angle
3. Enter as expression 'pi rad'
4. Error is immediately raised at validation with below message
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'RuntimeError'>: Property 'pirad' not found in 'pirad'
```
Then another error when recomputing, but due to above one.

0penBrain, 2023-02-23:
May someone test #8609 to be sure it solves and doesn't create regression ?

xtemp09, 2023-02-24:
How can I download the build to test?

0penBrain, 2023-02-24:
> How can I download the build to test?

Weekly builds are available here : https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Unfortunately the latest one doesn't have the fix. You'll have to wait, or build by yourself :)

xtemp09, 2023-02-24:
So inconvenient. I hoped I could have download [somewhere from here](https://github.com/FreeCAD/FreeCAD/actions/runs/4254732696).

0penBrain, 2023-02-24:
> So inconvenient.

Feel free to improve. :)

> I hoped I could have download [somewhere from here](https://github.com/FreeCAD/FreeCAD/actions/runs/4254732696).

That would be crazy ATM to have images for all PRs. :)

xtemp09, 2023-02-24:
Actually, it's not crazy. The files are compiled and still occupy some space on the servers. It would make no difference if a user download it till it's expired.

I remember reporting bugs and downloading the files of PR, though I don't remember what program that was. Perhaps, [qBittorrent](https://github.com/qbittorrent/qBittorrent/pull/18596/checks).

0penBrain, 2023-02-24:
> Actually, it's not crazy. The files are compiled and still occupy some space on the servers. It would make no difference if a user downloaded it till it's expired.

It's absolutely wrong. The runner owning the files is just destroyed at the end of the workflow.

> I remember reporting bugs and downloading the files of PR, though I don't remember what program that was. Perhaps, [qBittorrent](https://github.com/qbittorrent/qBittorrent/pull/18596/checks).

qBitTorrent is of a definitely smaller size, and only provides for Windows. ;)

Anyway, we know it would be good any user could test PRs, but this isn't as simple.

xtemp09, 2023-02-24:
> qBitTorrent is of a definitely smaller size, and only provides for Windows. ;)

qBittorrent is released for different platforms, Linux, Windows, masOS. Perhaps, the artifacts are available only for Windows, but I'm fine with it.

adrianinsaval, 2023-02-24:
It's it really that crazy though? The installed package could be uploaded as artifact by the CI, with proper compression it can be just 60mb. We could have a docker image with the proper environment for testing these.

adrianinsaval, 2023-02-24:
Btw if you need testers for a PR we have a system for generating snaps for them, just ask kunda1 (luzpaz here) to make one.

closed 2023-02-24 by wwmayer

adrianinsaval, 2023-02-24:
> But this only adresses Linux users. :)
yes, but it's certainly better than nothing at all. If we get windows CI running we can also upload the build as artifact, perhaps without the libpack stuff copied over to save storage and bandwith for testers.

> There are such solutions, but that's not so simple. Just installing Docker on Windows is already a mess for "normal" user. Then user have to setup a Xserver that will connect to the Docker container. :)

for windows installing on WSL would likely be easier, where with wslg you can run gui apps without effort. For linux users, those who have the same version of ubuntu can run it on their own system. For the rest, we can make snaps or let us setup docker or distrobox, for me it would be easier to setup the environment once, update every once in a while and just download a compiled package for testing, rather than having to compile each PR I want to test.

PR #8609 (2023-02-24): [BugFix] Expression: always add space char before unit when converting to string
Fixes #8562
commit 09c1b9ce (2023-02-24): [BugFix] Expression: always add space char before unit when converting to string, fixes #8562
commit 3497ca45 (2023-02-24): [BugFix] Expression: always add space char before unit when converting to string, fixes #8562