FreeCAD/FreeCAD #8575: [Problem] FreeCAD doesn't remember which workbench was being used last when loading a document

sliptonic, 2023-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31731 (Git)
Build type: Unknown
Branch: master
Hash: c5c3077f8fe99efafc6112b3c5d27bc3e08813f0
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

The issue affects workflow across all workbenches. 

The current system automatically loads and activates a single workbench on startup.  The default workbench is 'start' but the user can select a different workbench.  This behavior is somewhat limiting because during the normal lifecycle of a project a user my spend significant time in different workbenches and restart FreeCAD many times.  Moreover, the user may be working on many documents that are in different stages of their lifecycle.  

Thus the user may be working on project #1 and spending significant time in Part Design.  Then load document #2 in which he is creating a Path Job.  The result is that the user almost always has to switch workbenches when switching documents.  A single default isn't sufficient.

What's needed is for the system to remember which workbench was active for any given document and reload that workbench when the document is loaded.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-02-21:
Possible related forum topic: https://forum.freecad.org/viewtopic.php?t=43326

sliptonic, 2023-02-23:
> a property should be added to the document object `lastWorkbench` and updated whenever a workbench is activated, the question is if this should be applied on all open documents or just the active document?

That was my first thought too.  @chennes said that the topic came up in a forum discussion and people suggested that it shouldn't be in the document but in the recent file list.  I couldn't find that discussion.

If it is stored in the document, it will need to fail gracefully.  Documents will be shared with people who don't have addon workbenches installed.

When a workbench switch happens, it should only be recorded to the active document.  

A different question is whether the active workbench should be switched when the active document changes.  If I have two documents open and was last active in Part in document1 but I'm currently active in Path in document2, what should happen when I switch to document1?

Jee-Bee, 2023-03-03:
I'm not sure if i see anything in **only** loading latest WB. If you want to propose something i would say go for something as `Re-open last session`. 

The reason is as follows:
If you end with a single file containing a single object that made with a single WB there is a big change that reloading the same WB works out.
What to do with a single file that contain multiple objects from lest say 2 WB's (PD and surface for example)? Is the latest the best?
If a user works one multiple files at the same time and multiple WB (For example PD, AssemblyX, TD, Surface or Sheet metal) the change that the right one is chosen is becoming smaller.

If everything from the last session can opened on startup / at least at once then it is useful to load one or multiple WB's

An addition what i think of now. look at the latest feature in the tree and open the corresponding WB.

johnmichaeldnichols, 2023-09-26:
Is anyone currently working on this? If not I would like to attempt to make a fix. Is there a due process for submitting my suggested fix for the issue?

adrianinsaval, 2023-09-26:
https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md

Feel free to give it a try and ask any question if you need help with the process.

I tend to agree with what sliptonic mentioned above about storing this as the user's data rather than on the document itself as I initially suggested. Think of how pdf readers usually remember in what page you left off last time but this isn't stored on the document.

yorikvanhavre, 2023-09-26:
Also, I think one must always be able to disable this. Ex. I'm an architect, I work with others (civil engineers, etc) I always want to open files and find myself in Arch, regardless of what the last user was using,

FlachyJoe, 2023-09-26:
@yorikvanhavre what about keeping a trace of the last editor in the file? So if same, open last workbench else open the defined default.

yorikvanhavre, 2023-09-26:
Definitely keeping track of the last workbench used is okay, and a good first step. Now, after opening a file, I think FreeCAD should:

1. Look for a pref setting saying if it should adopt file-defined workbenches or not
2. If no, do nothing (same as now)
3. If yes, look for a file-defined workbench name
4. If found (the WB could be an addon and could be not present on this machine), switch to it

One still needs to test this workflow because for ex. the start workbench might attempt to switch workbench after file load, so one has to check everything happens in the right order

@johnmichaeldnichols Note that this could be entirely done in a python module, using a document observer to insert the given data on file save, and read it/handle it on file restore. I think it's best to do that in FreeCAD core and not in an addon module, but it could be done in one of the built-in python modules such as Tux, if that suits you better.

adrianinsaval, 2023-09-26:
> Also, I think one must always be able to disable this. Ex. I'm an architect, I work with others (civil engineers, etc) I always want to open files and find myself in Arch, regardless of what the last user was using,

O yes, this should have a preference checkbox.

luzpaz, 2023-09-27:
Anyone want to step-up and implement this nice QoL feature ?

chennes, 2023-09-28:
I'm not sure we've yet achieved consensus on what this feature looks like, I see two different approaches discussed above and I'm not certain we've arrived at a single one.

The two approaches are either a) to store the last used workbench in the file itself, or b) to store it as part of the "recent files" list (or something along those lines). The preference among attendees at last year's FreeCAD day was for it to be stored on a per-user basis in their recent files list, so that a different user reading the same file is not affected at all by what workbench some other user was using. Is that the general consensus here as well? In particular, @yorikvanhavre it sounds to me like you're still envisioning a piece of information stored in the actual file, rather than purely in user data. Is that right?

luzpaz, 2023-09-28:
Note, the outcome of this ticket could inform #10817

johnmichaeldnichols, 2023-09-28:
I was thinking something simple such as only remembering the last known workbench the user was on for now. Implementing it on the core FreeCAD, and having it in user settings; not a per file basis. How does that sound? More features such as a preference setting for starting workbench on 'new' files, etc could be added later and I will code the base-line feature as such to allow easy expansion later.

BazookaJoe26, 2026-08-15:
Hi! Is anyone currently working on this issue?

I’d be happy to take it on as my first FreeCAD issue, if that works for everyone. I have successfully built FreeCAD locally and created a fork, so I’m ready to get started.

From the previous discussion, I understand that there are two possible approaches:

1. Store the last active workbench in the FreeCAD document itself. When the document is opened, FreeCAD could optionally switch to that workbench, controlled by a user preference.

2. Store the information only in the user’s local preferences, without modifying the document. As a first step, FreeCAD could simply remember and restore the user’s last active workbench globally. This could later be extended to remember it per file.

Which approach is currently preferred for this issue? @maxwxyz

maxwxyz, 2026-08-15:
@FreeCAD/design-working-group might want to weigh in

obelisk79, 2026-08-15:
I think option 1 would be better from a user-experience standpoint, and is a solution I think is quite good at face-value. I would be a bit more discerning of a PR.

Also, be extremely careful to ensure any changes to what goes into a FCStd file doesn't negatively impact forward/backward compatibility. That's an easy way to get either instantly rejected or take a straightforward solution and make it take months to get merged.

BazookaJoe26, 2026-08-16:
Hi @maxwxyz 

i saw the discussion in the discord channel, but it doesn't seem like a clear answer was found? 😅