FreeCAD/FreeCAD #8596: After dimension delete is not posible to add ne dimensions

librucha, 2023-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

After delete dimension (created for edge of Rectangle) is not possible to add other dimensions, Text or Label

It looks as some pointer is still set to deleted dimension.

### Anything else?

00:26:25  Running the Python command 'Draft_Dimension' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftguitools/gui_dimensions.py", line 98, in Activated
    super(Dimension, self).Activated(name="Dimension")
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftguitools/gui_base_original.py", line 287, in Activated
    super(Creator, self).Activated(name, noplanesetup)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftguitools/gui_base_original.py", line 117, in Activated
    App.activeDraftCommand.finish()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftguitools/gui_edit.py", line 359, in finish
    self.deformat_objects_after_editing(self.edited_objects)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftguitools/gui_edit.py", line 837, in deformat_objects_after_editing
    obj_gui_tools = self.get_obj_gui_tools(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Draft/draftguitools/gui_edit.py", line 785, in get_obj_gui_tools
    obj_gui_tools = obj.obj_gui_tools

Cannot access attribute 'obj_gui_tools' of deleted object

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-26:
@librucha can you create an example .FCStd file, zip it, and drag/drop in to this github ticket for us to easily reproduce ?

yorikvanhavre, 2023-02-27:
I can't confirm this, for me the following works normally:
* Create a rectangle
* Create a dimension
* Delete that dimension
* Create another dimension
* 
I suspect something else has been done inbetween, such as closing the dialog, or something like that. @librucha could you provide an exact step-by-step procedure that makes the bug happen again?

librucha, 2023-02-28:
Hello everyone.

I will try to repeat the problem, but quit and start of FreeCad solved it in my project file (which is not shareable). Give me few days plese.

Thanks

luzpaz, 2023-03-05:
@librucha any progress?

Roy-043, 2023-03-11:
I can reproduce in V0.20.2:
You have to delete the dimension while editing it with Draft_Edit and then close the task panel.

In the current V0.21 this is no longer possible. The dimension being edited is placed into edit mode and therefore cannot be deleted (See #7970).

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32198 (Git)
Build type: Release
Branch: master
Hash: f51b2156f35399cab38eef1e957a59ad5a11de66
Python 3.8.16, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
```

closed 2023-03-11 by Roy-043