FreeCAD/FreeCAD #8610: [Issue] GLTF export only available in Gui module, not in headless

DanielOrchard, 2023-02-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=645933&hilit=gltf#p645933

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

File formats

### Issue description

GLTF Exporter only exists as part of the FreeCADGui module. This makes it impossible to use in exporting in a headless python-only approach. 

All other modules export via the Import.export(), except when doing GLTF, which routes through ImportGui.export()

I know GLTF goes through a different pipeline so I don't know if this is a required limitation, but it would be great to know if there are any workarounds or if it might change in future updates.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-26:
@wwmayer does #8649 close this ticket ?

wwmayer, 2023-02-26:
Yes.

closed 2023-02-26 by wwmayer

lfertig, 2023-03-29:
Hello together,

im not quite sure this is completely solved yet: I'm running a windows dev installer from a couple of days ago and also tried a conga build with rapidjson enabled as posted in some comments online. The following line exports a simple testfile properly to .gltf when being run from the python console inside FreeCAD (either of the two versions)

`Import.export([obj for obj in doc.Objects if hasattr(obj, "Shape")], "test.gltf")`

When run headless from a python script, though, it produces a rudimentary .gltf file with an empty .bin alongside it. Going through the freecadcmd console i get the following error messages which leads me to assume this has to do with internal render meshes that are generated by the gui version for display?

```

[FreeCAD Console mode <Use Ctrl-Z plus Return to exit.>]
>>> from FreeCAD import Import
>>> doc = FreeCAD.open('test.fcstd')
Importing project files......
Postprocessing......                            (100 %)
>>> Import.export([obj for obj in doc.Objects if hasattr(obj, 'Shape')], 'test.gltf')
RWGltf_CafWriter skipped node 'Bottom' without triangulation data
RWGltf_CafWriter skipped node 'Top' without triangulation data
RWGltf_CafWriter skipped node 'Side L' without triangulation data
RWGltf_CafWriter skipped node 'Side R' without triangulation data
RWGltf_CafWriter skipped node 'Back' without triangulation data
>>> from FreeCAD import MeshPart
>>> mesh = doc.addObject('Mesh::Feature','meshed object')
>>> mesh.Mesh = MeshPart.meshFromShape(Shape=doc.Objects[1].Shape, LinearDeflection=0.01)
>>> Import.export([mesh], 'test.gltf')
<Import> ImportOCAF2.cpp(1193): test#meshed_object has null shape
```

I'm not deep enough into FreeCAD yet to further investigate, but let me know if i can be of any help.

regards,
Lukas

wwmayer, 2023-03-29:
In the GUI version the view provider is responsible to tessellate a shape in order to display something in the 3D view. When running FreeCAD in headless mode there is no such instance and it's up to the user to explicitly call the function tessellate() of the given shape.

So, add this to your script
```
shapes = [obj for obj in doc.Objects if hasattr(obj, 'Shape')
for shape in shapes:
    shape.tessellate(0.1)

Import.export(shapes, 'test.gltf')
```
Passing a mesh object has no effect because the OCCT algorithms expects a shape and from there it extracts the mesh data.

morganrallen, 2023-06-14:
Looks like Alpine also needs RapidJSON include in the opencascade package, I've created an issue there and will make a pull request too.