FreeCAD/FreeCAD #8625: [Issue] Python3.10/Pysides6/gentoo -> crash at boot

alxchk, 2023-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
AboutImage=freecadabout
AppDataSkipVendor=true
AppHomePath=/usr/lib64/freecad/
AppIcon=freecad
AppTempPath=/tmp/
BinPath=/usr/lib64/freecad/bin/
BuildRepositoryURL=Unknown
BuildRevision=Unknown
BuildRevisionDate=Unknown
BuildVersionMajor=0
BuildVersionMinor=20
BuildVersionPoint=2
Console=0
CopyrightInfo=© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Debug=0
DocPath=/usr/lib64/freecad/doc/
ExeName=FreeCAD
ExeVendor=FreeCAD
ExeVersion=0.20.2
LoggingConsole=1
MaintainerUrl=http://www.freecad.org/wiki/Main_Page
PythonSearchPath=/usr/bin/../lib/python310.zip:/usr/bin/../lib/python3.10:/usr/bin/../lib/python3.10/lib-dynload
RunMode=Gui
SplashAlignment=Bottom|Left
SplashInfoColor=#c8c8c8
SplashInfoPosition=15.210
SplashScreen=freecadsplash
SplashTextColor=#ffffff
StartWorkbench=StartWorkbench
UserAppData=/home/user/.local/share/FreeCAD/
UserCachePath=/home/user/.cache/FreeCAD/Cache/
UserConfigPath=/home/user/.config/FreeCAD/
UserHomePath=/home/user
UserMacroPath=/home/user/.local/share/FreeCAD/Macro/```
```

### Subproject(s) affected?

None

### Issue description

Fastest way to reproduce

```/tmp > freecad -c
[FreeCAD Console mode <Use Ctrl-D (i.e. EOF) to exit.>]
>>> import PySide6
Fatal Python error: init_enum: could not load enum pickling helper function
Python runtime state: initialized

Current thread 0x00007f0f96e26ac0 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/usr/lib/python3.10/site-packages/shiboken6/__init__.py", line 27 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/usr/lib/python3.10/site-packages/PySide6/__init__.py", line 64 in _setupQtDirectories
  File "/usr/lib/python3.10/site-packages/PySide6/__init__.py", line 124 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<stdin>", line 1 in <module>

Extension modules: FreeCAD, shiboken6.Shiboken (total: 2)
zsh: IOT instruction (core dumped)  freecad -c```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-02-24:
Please provide the info from FreeCAD menu Help -> About FreeCAD (use the copy button).

I think your FreeCAD version is probably compiled with Qt5, so if you try to import PySide6 in it you probably get a conflict.

alxchk, 2023-02-24:
[code]
OS: Gentoo Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Release
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

Yes, looks like it. However it's not clear why it's try to search pysides6 then:
```
During initialization the error "No module named 'PySide6'" occurred in /usr/lib64/freecad/Mod/AddonManager/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /usr/lib64/freecad/Mod/Draft/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /usr/lib64/freecad/Mod/Arch/InitGui.py
Please look into the log file for further information
During initialization the error "No module named 'PySide6'" occurred in /usr/lib64/freecad/Mod/Tux/InitGui.py
Please look into the log file for further information
Unknown command 'Std_AddonMgr'
```

Also to make it run I removed pysides6 completely

yorikvanhavre, 2023-02-24:
That´s strange! there is no mention of PySide6 anywhere in the source code of those modules...
Could you try this in a terminal:
```bash
cd /usr/lib64/freecad/Mod/Draft/
grep -r "PySide6" .
```
also please check the log file in ~/.local/share/FreeCAD/FreeCAD.log ? (you might need to start FreeCAD with `FreeCAD -l`)

alxchk, 2023-02-24:
```tachi /usr/lib64/freecad/Mod/Draft # grep -r "PySide6" .
grep: ./__pycache__/Draft_rc.cpython-310.opt-2.pyc: binary file matches
grep: ./__pycache__/Draft_rc.cpython-310.pyc: binary file matches
grep: ./__pycache__/Draft_rc.cpython-310.opt-1.pyc: binary file matches
./Draft_rc.py:from PySide6 import QtCore```

[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/10824697/FreeCAD.log)

chennes, 2023-02-24:
All of those modules will only be importing "PySide" which is itself a wrapper designed to provide whichever version of PySide FreeCAD was compiled with (in all cases right now, PySide2). So it appears there is actually something wrong with that wrapper.

chennes, 2023-02-27:
PySide.py

yorikvanhavre, 2023-02-28:
Indeed, all the errors in the log come from files that have `import PySide`.

adrianinsaval, 2023-03-28:
can you post the options you used during compilation? FreeCAD souldn't pickup pyside6 when using qt5

adrianinsaval, 2023-05-11:
@alxchk is this still a problem?

adrianinsaval, 2023-06-14:
No response, not actionable. I'll assume it was something wrong on OP's system and close. Feel free to reopen if you can still reproduce and can provide further information on how you arrived at this state.

closed 2023-06-14 by adrianinsaval