FreeCAD/FreeCAD #8640: [Issue] PartDesign Loft+Pad Resulting Shape is not a solid

racer-coder, 2023-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76365

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal
```

### Subproject(s) affected?

PartDesign

### Issue description

Creating a body with a pad and a loft, where the pad overlaps the loft, and the pad has the exact same outline as one of the faces of the loft, can result in "Resulting shape is not a solid."  If the pad is reversed (so it goes away from the loft) then it works.

Duplicating in Part by unioning two bodies, one with the pad, and the other with the loft, results in "<Part> ViewProviderExt.cpp(1270): Cannot compute Inventor representation for the shape of Unnamed#Fusion: Bnd_Box is void"

Possible duplicate of #8394, I do not segment fault but I get the same error message.  There is no description in 8394 of the object being created though.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

racer-coder, 2023-02-24:
Forgot to include the actual FreeCAD file, so one does not have to get it from the forum.
[loftbug.zip](https://github.com/FreeCAD/FreeCAD/files/10829166/loftbug.zip)

grubertechnic, 2023-07-15:
When i open the file, everything is fine. Also i see no issue at all.

Please pinpoint your issue and/or give a step by step instruction to replicate the issue.

racer-coder, 2023-07-15:
1. Open attached file
2. For the PumpPad object, change it so Reversed=true
3. Note that now PumpPad has an error saying "Pad: Resulting shape is not a solid."  This IMO is not true as PumpPad should overlap LoftFromPumpToBracket.

grubertechnic, 2023-07-22:
Thanks, yes i can confirm that.

That is pretty sure an OCCT issue. Also a pocket (or cut) is invalid in that case, but the geometry check said OK, which is wrong, volume, mass and so on is positive.

Note: that is no duplicate of #8394, that is different, maybe there is an other duplicate, maybe not, hard to screen all tickets so fast.

@Roy-043 can you add the the _3rd Party: OCC_ label?

maxwxyz, 2024-07-16:
There is no error in the current weekly but the reverse operation is still not possible.
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.41
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.22

```

ElementW, 2026-08-25:
The *error* no longer appears since toponaming fixes that happened in 2024, but the Reversed=true pad operation still results in no geometry as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328.