FreeCAD/FreeCAD #8657: [Problem] PR Template Too Wordy

chennes, 2023-02-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Problem description

The current [PR template](https://github.com/FreeCAD/FreeCAD/blob/master/.github/PULL_REQUEST_TEMPLATE.md) looks like this:

---

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [ ]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to update the Wiki with the features added or changed once this PR is merged.  
**Note**: If you don't have wiki access, then please mention your contribution on the [1.0 Changelog Forum Thread](https://forum.freecad.org/viewtopic.php?f=10&t=69438).

---

The vast majority of Pull Requests do not need to update the wiki: Maintainers and others have been quite diligent about reminding authors of things that DO need an update to do so manually, and it is unlikely that anyone is actually being reminded by the statement. The more we can shorten this bit of text up, the more likely it is to be read, and the less cluttered it will make pull requests where the requestor does not delete it. I think in an ideal world, a PR would simply contain the line:

- [X]  This Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-02-27:
Yes, before it was much more like, just a series of 4 or 5 checkboxes that basically said you had checked everything you need to check before doing the PR.
I'm always doubting if the wiki line should be there or not... It's a good reminder to have...

luzpaz, 2023-02-27:
We can also hide content in a [collapsed div](https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab)

chennes, 2023-02-27:
I'd agree about the wiki line being a good reminder if I believed anyone was reading it and _being_ reminded 😄 . My feeling is that it's lost, and that for cases where we really do need to update the wiki Maintainers (especially @donovaly) have been good about reminding contributors to update the wiki. The thing I really, really want new contributors to see is the link to CONTRIBUTING.md -- as far as I am concerned, everything else just distracts. If it was more obvious how to use the checkbox thing I'd get rid of the first sentence too.

yorikvanhavre, 2023-02-28:
What about a:
* [ ] I checked if I need to update the wiki

WandererFan, 2023-03-01:
I am a big believer in checklists:

[] Your Pull Request meets the requirements outlined in section 5 of CONTRIBUTING.md for a Valid PR

- change to -

[] Maintainer quick fix -or-
[] Cosmetic fix -or-
[] Solves Issue xxxx

[] PR Passes Test Framework
[] Reviewed by yyyyy
[] Each commit compiles cleanly & passes tests
[] No new dependencies
[] wiki updated (if applicable)
[] New tests created (if applicable)
[] ...

chennes, 2023-03-01:
I'd be OK with elucidating what the CONTRIBUTING.md says, if that's what you're going for here. The "radio button" thing at the top is a bit subverted by GitHub's little counter thing, though. Is the idea that you'd delete the ones that don't apply? 

The first "Passes test" is handled by the CI, and I'm not sure about the necessity of the "reviewed by" (since GitHub is going to show that information automatically).

WandererFan, 2023-03-03:
I was thinking an 'X' if we can't do real check marks for the "type" of PR.

If there is more than one thing from CONTRIBUTING that we want confirmed, then we should itemize them.

chennes, 2023-03-03:
To my read, the things from CONTRIBUTING.md that are "check-off-able" are:
* This a minimal and accurate answer to exactly one identified and agreed-on problem
* Each commit compiles cleanly
* The PR does not include non-trivial code from other projects
* Each commit message succinctly explains what the commit achieves

The stuff about passing tests is covered by CI, I don't see any reason to manually enter a checkmark. The code formatting will eventually be covered by something like pre-commit hooks, or automatically-applied formatting of some other variety. There's nothing in the current CONTRIBUTING.md about the wiki, or about _creating_ unit tests.

...and I still circle back to the fact that in most cases (it seems to me) this template text is being wholly ignored. It's being left unmodified in the PR submission, with the checkbox untouched. My hope is that by making the text shorter, we might be able to persuade someone into reading it (OK, probably just wishful thinking).  I suppose it's possible to automate some kind post-submission hook that forces submitters to check those boxes, but that's starting to feel pretty heavy-handed again.

adrianinsaval, 2023-05-31:
Can't this be more similar to the issue template?
[image]
there we can add text that doesn't end up in the actual PR, we can there explain the contribution rules and remind about updating the wiki.
we can also put checkboxes that will then actually show up in the PR and add a description field for everything else

chennes, 2023-06-04:
All that stuff ends up in the issue, too (just formatted differently). I'm not a fan of any of this template stuff: I just don't believe anyone is reading it. Let's stop wasting anyone's time with it: if someone *doesn't* follow CONTRIBUTING.md when submitting a PR, a Maintainer can tell them so directly.