FreeCAD/FreeCAD #8675: [Problem] Sketch axis size should follow zoom level

FlachyJoe, 2023-02-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.15162 (Git shallow)
Build type: Unknown
Branch: master
Hash: c437b79adabb6a91e8fb6dc46729cb720ba16a5f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * sheetmetal 0.2.49
  * fc_nodes
  * A2plus 0.4.56
  * CurvedShapes 1.0.4
  * Curves 0.6.6
  * mooc-workbench
  * Render 2022.1.0
  * Plot 2022.4.17
  * lattice2 1.0.0
  * fcpd-workbench 0.2.0
  * Silk
  * Wood-Frame
  * Assembly4 0.11.12
  * fasteners 0.3.41
  * BIM 2021.12.0
  * Nodes 0.1.36
  * MeshRemodel 1.8919.0
  * ExtremeProLight 1.4.3
  * fcgear 1.0.0
  * SelectorToolbar
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Sketch axis follow geometries dimension. When starting a new sketch and zooming out axis can become invisible and auto-constraint can't be create on them.

Axis size should follow the zoom rather the geometries.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-10:
@FlachyJoe was this ticket impacted by 8473 ?

FlachyJoe, 2023-06-11:
@luzpaz no.

PR #17312 (2024-11-25): [Feature] Sketcher infinite axes
Squashed and rebased #12319

Fix #8675
closed 2024-11-25 by yorikvanhavre