FreeCAD/FreeCAD #8676: [Feature Request] Ratio constraint in Sketcher

edwilliams16, 2023-02-26:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?style=1&t=69147

### Subproject(s) affected?

Sketcher

### Idea description

Be able to specify the ratio between two line segments (or other lengths...)  in Sketcher.  This would be a natural extension of equality between lengths and could simplify construction.  In the forum thread @abdullahtahiriyo volunteered to be assigned this.
I won't repeat the forum discussion here.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

edwilliams16, 2024-05-09:
Perhaps this should assigned to someone else?

FlachyJoe, 2024-05-12:
I think it was spoken in another issue. IMHO Expression do the job without more work.
The GUI improvement I imagine is if Expression Editor could refer to a constraint by a click (or a segment length with autocreate a driven constraint on it).

AjinkyaDahale, 2024-05-14:
Agree with @FlachyJoe. Only thing I might add is that the selecting anything while an expression is on may not always be possible. For example in GNOME, in some cases the expression window prevents the main window from being clicked.

Either way, this becomes now a design issue.

edwilliams16, 2024-05-14:
> IMHO Expression do the job without more work. 

How do you do that?   
Suppose I want to make one line's length, say, 5 times that of another, leaving the actual lengths to be determined by other constraints in the sketch.   
If I name the first line's length _length1_  as a reference constraint and set the second line's length to _.Constraints.length1 * 5_, this gives the error **reference constraint from this sketch cannot be used cannot be used in this expression**.  
  I have an elaborate work-around using separate construction geometry I have shown on the forum, but I don't see how to do it with expressions.  
 It seems to me the equality constraint could be modified to put a multiplier on one of the lengths - equality being the special case where that multiplier is one.

AjinkyaDahale, 2024-05-14:
> Suppose I want to make one line's length, say, 5 times that of another, leaving the actual lengths to be determined by other constraints in the sketch. If I name the first line's length _length1_ as [...]

I can see how this might cause a problem, and unfortunately it requires some fundamental changes in the solver. Still, the expressions engine is the best idea. I can imagine more complex relations to be requested (arc length equals line length), and adding them for a relatively small portion of use cases (this could be a wrong assumption, but not sure if there's a way to know it) may only clutter things.

> I have an elaborate work-around using separate construction geometry I have shown on the forum, but I don't see how to do it with expressions.

I would like to see that forum post.

> It seems to me the equality constraint could be modified to put a multiplier on one of the lengths - equality being the special case where that multiplier is one.

It would be a patchwork solution, and obsolete once expressions can support reference constraints.

edwilliams16, 2024-05-14:
See [https://forum.freecad.org/viewtopic.php?p=599512#p599512](url)  I think there's a potential problem with self-referential expressions in the sketcher.

Here's a simpler example of my construction than in the thread.  You first set the two sides of the construction triangle in the correct ratio - say height 10, base 20 for 1:2. Then create a reference angle constraint. Delete the side length constraints and make the angle constraint driving.  The sides are made equal to the corresponding sides of the model. 
 If you know trig, you can just set the angle to atan(side_ratio) = atan(1/2). This forces the construction triangle sides to remain in the 1:2 ratio as its size varies.
[setratio.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/15312982/setratio.FCStd.zip)

FlachyJoe, 2024-05-14:
You are indeed right about the expressions.
We can remove the limitation to use the driven constraints but the sketch has to be recomputed in order to update the expression.
Try to comment these lines and refer the constraint by index (i.e. _.Constraints[12]_):
https://github.com/FreeCAD/FreeCAD/blob/cf57d0adc766db26c797b606746a052a8994ad97/src/Mod/Sketcher/App/SketchObject.cpp#L8684-L8698

edwilliams16, 2024-05-14:
I'll give that a try when I get compiling working again.  That just removes the self-referential prohibition.  It doesn't address the convergence issue I brought up if you alternate between the sketcher solver and constraint evaluation. It seems hit or miss whether it would work in any given case, as it depends on which solution the solver comes up with when not fully constrained.

FlachyJoe, 2024-05-14:
For integer ratio you can use construction segments and equalities.

[image]

AjinkyaDahale, 2024-05-14:
I agree with @edwilliams16. Adding support for expressions with this is effectively custom constraints.

edwilliams16, 2024-05-14:
>  Adding support for expressions with this is effectively custom constraints.

Is that under consideration?  I'm only guessing how that might work.  I assume you'd have to add in a custom constraint term into the objective function you are minimizing to solve the system?

AjinkyaDahale, 2024-05-14:
Notice that I said "effectively". The expression engine is operating outside the solver, and as such doesn't know anything about the constraint. This does pose a challenge since now we don't havr information on the gradients, just on errors.

FlachyJoe, 2024-05-15:
A new ratio constraint is quite easy to implement in the solver from the ConstraintEqualLineLength and the ratio parameter will be as expression-able as other constraint value.
I can write such but it needs also GUI choices I won't assume.

AjinkyaDahale, 2024-05-15:
> A new ratio constraint is quite easy to implement...

That's true, but consider in the future someone wants to make a length that's, say, a geometric (or harmonic) mean of 2 or more lengths.

FlachyJoe, 2024-05-15:
@AjinkyaDahale it will be another issue :smile: 
This current issue is fixable but it will introduce an exception in the KISS design of solver i.e. a constraint which isn't fully necessary because workaround exists.

FYI driven constraints are usable in expression as hidden reference a recompute is requested to update them.
[image]

AjinkyaDahale, 2024-05-15:
> @AjinkyaDahale it will be another issue 😄 This current issue is fixable but it will introduce an exception in the KISS design of solver i.e. a constraint which isn't fully necessary because workaround exists.

For sure. I guess the question right now is _what form that issue takes_.

> FYI driven constraints are usable in expression as hidden reference a recompute is requested to update them. 

Is this without removing the check? I expect this could get complicated in terms of iterations if there is some circular relation between the reference and the expression, which may or may not be (uniquely) solvable.

avaziman, 2024-09-06:
any updates here?

luzpaz, 2024-09-06:
Probably not, since we're trying to get 1.0 out the door.