FreeCAD/FreeCAD #8677: upgrade from 0.20.1 to 0.20.2 requires older python version

ebo, 2023-02-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20.1 and 0.20.2 segfault on boot.  I am not able to pull up the version information because it dies as it is starting up.
```

### Subproject(s) affected?

None

### Issue description

I recently upgraded anaconda, and it broke my freecad-0.20.1 install.  When I tried to specifically install the newer 0.20.2, it "package freecad-0.20.2-py38h63dbee4_0 requires python >=3.8,<3.9.0a ...".  I eventually removed both environments and installed clean.  Both 0.20.1 and 0.20.2 are now broken at a clean install.  I installed 0.20.0, and it works out of the box, but interestingly it installed python-3.10.8 with 0.20.0, but I thought that it required a python-3.9.

My apologizes for not taking the time to figure out what is broken with the install packages, but this sould be easily replicated by running and of the following:

mamba  create -n freecad_0.20.2 freecad==0.20.2

mamba  create -n freecad_0.20.1 freecad==0.20.1

and the one that works:

mamba  create -n freecad_0.20.0 freecad==0.20.0

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-26:
Maybe should be posted to https://github.com/FreeCAD/FreeCAD_Conda ?  
CC @looooo

ebo, 2023-02-26:
I can cross post to FreeCAD_Conda, but for the freecad folks, I thought they needed to know that there is a compatibility issue.  I will make another post with the package list.

ebo, 2023-02-26:
# packages in environment at /home/ebo/mambaforge/envs/freecad_0.20.1:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
coin3d                    4.0.0             hc543bf6_1011    conda-forge
curl                      7.88.1               hdc1c0ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
expat                     2.5.0                h27087fc_0    conda-forge
ffmpeg                    4.4.2           gpl_h8dda1f0_112    conda-forge
fftw                      3.3.10          nompi_hf0379b8_106    conda-forge
flann                     1.9.1             he05ef13_1011    conda-forge
fltk                      1.3.8                h83e168f_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freecad                   0.20.1          py310h8dc7af3_3    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gdk-pixbuf                2.42.8               hff1cb4f_1    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
gl2ps                     1.4.2                h0708190_0    conda-forge
glew                      2.1.0                h9c3ff4c_2    conda-forge
glib                      2.74.1               h6239696_1    conda-forge
glib-tools                2.74.1               h6239696_1    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gmsh                      4.11.0               hc719622_0    conda-forge
gnutls                    3.7.8                hf3e180e_0    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
graphviz                  7.0.5                h2e5815a_0    conda-forge
gst-plugins-base          1.21.3               h4243ec0_1    conda-forge
gstreamer                 1.21.3               h25f0c4b_1    conda-forge
gstreamer-orc             0.4.33               h166bdaf_0    conda-forge
gtk2                      2.24.33              h90689f9_2    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
harfbuzz                  6.0.0                h8e241bc_0    conda-forge
hdf4                      4.2.15               h9772cbc_5    conda-forge
hdf5                      1.12.2          nompi_h4df4325_101    conda-forge
icu                       70.1                 h27087fc_0    conda-forge
imath                     3.1.6                h6239696_1    conda-forge
jack                      1.9.22               h11f4161_0    conda-forge
jpeg                      9e                   h0b41bf4_3    conda-forge
jsoncpp                   1.9.5                h4bd325d_1    conda-forge
jxrlib                    1.1                  h7f98852_2    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.20.1               h81ceb04_0    conda-forge
lame                      3.100             h166bdaf_1003    conda-forge
lcms2                     2.14                 h6ed2654_0    conda-forge
ld_impl_linux-64          2.40                 h41732ed_0    conda-forge
lerc                      4.0.0                h27087fc_0    conda-forge
libaec                    1.0.6                hcb278e6_1    conda-forge
libblas                   3.9.0           16_linux64_openblas    conda-forge
libcap                    2.66                 ha37c62d_0    conda-forge
[... 116 lines cut ...]
xcb-util-keysyms          0.4.0                h516909a_0    conda-forge
xcb-util-renderutil       0.3.9                h166bdaf_0    conda-forge
xcb-util-wm               0.4.1                h516909a_0    conda-forge
xerces-c                  3.2.4                h55805fa_1    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
[... cut ...]
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h0b41bf4_1003    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge

luzpaz, 2023-04-11:
@ebo did you cross-post this to FreeCAD_Conda ?

adrianinsaval, 2023-07-25:
if this is still a problem report to https://github.com/conda-forge/freecad-feedstock instead, this is not managed here

closed 2023-07-25 by adrianinsaval