FreeCAD/FreeCAD #8684: [Issue] Adding in sketcher preference the hidden parameter "RadiusDiameterConstraintDisplayAngleRandomness"

69Piero, 2023-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=76236

### Version

0.21 (Development)

### Full version info

```shell
Adding in sketcher preference the hidden parameter "RadiusDiameterConstraintDisplayAngleRandomness" so the user can prevent overlapping diametral dimensions
```

### Subproject(s) affected?

Sketcher

### Issue description

currently in sketcher diametral/radial dimensions overlap every time making a real mess especially in complex sketch. There is an hidden parameter named "RadiusDiameterConstraintDisplayAngleRandomness" that can prevent it. 

if this parameter is shown in sketcher preference, the user is free to decide if having or not overlapping dimensions in sketch view 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-27:
This is a feature request and therefore the feature request template should of been used. I'll clean it up now, but please next time please follow the our submission workflow

adrianinsaval, 2023-02-28:
I don't know why we still have different templates given that the new contribution process states that bugs and feature request should be treated the same and be described as a problem in need of a solution.

luzpaz, 2023-02-28:
Sorry, my bad.

maxwxyz, 2024-03-30:
@PaddleStroke FYI

pierreporte, 2024-08-30:
@maxwxyz UI/UX not CAD feature